raspberrypi/linux #3038: 4.14.123 RPI AX88179 Drive causes crash

suyuan168, 2019-07-01:
I am using Openwrt and my device is RPI2
I used two AX88179s as LAN or WAN
But after a while, when a network card has no data transmission, it will crash.
The log is as follows
Normal use for a while:
[ 5208.757324] blest is not available
[ 5210.176208] blest is not available
[ 5213.882767] blest is not available
[ 5219.823918] blest is not available
[ 9157.107129] ------------[ cut here ]------------
[ 9157.113027] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:320 dev_watchdog+0x16c/0x268
[ 9157.123672] NETDEV WATCHDOG: eth1 (ax88179_178a): transmit queue 0 timed out
[ 9157.132056] Modules linked in: rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib qcserial option ipw brcmfmac usb_wwan usb_serial_simple ti_usb_3410_5052 sierra rtlwifi rtl8xxxu rt2x00usb rt2x00lib pppoe ppp_async pl2303 oti6858 nf_nat_pptp nf_conntrack_pptp mt7601u mos7720 mmc_spi mct_u232 mac80211 keyspan iptable_nat ipt_REJECT ipt_MASQUERADE garmin_gps ftdi_sio ebtable_nat ebtable_filter ebtable_broute cypress_m8 cp210x ch341 cfg80211 cdc_mbim belkin_sa bas_gigaset ark3116 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_owner xt_ndpi xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iface xt_hl xt_helper xt_hashlimit xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_condition xt_comment xt_addrtype xt_TRACE xt_TPROXY xt_TCPMSS
[ 9157.217624] xt_REDIRECT xt_LOG xt_IPMARK xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xt_ACCOUNT visor usbserial usbhid usb_gigaset ts_fsm ts_bm sr9700 sierra_net ser_gigaset rtl8150 rndis_host r8712u r8152 qmi_wwan pppox ppp_generic pegasus of_mmc_spi nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda mcs7830 macvlan kaweth kalmia iptable_raw iptable_mangle
[ 9157.305959] iptable_filter ipt_ECN ipheth ip6table_raw ip_tables huawei_cdc_ncm hso hid_generic gigaset ezusb ethoc ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 dm9601 crc7 crc_itu_t crc_ccitt compat_xtables compat cdc_wdm cdc_subset cdc_ncm cdc_ether cdc_eem cdc_acm brcmutil ax88179_178a asix arptable_filter arpt_mangle arp_tables sch_cake sch_teql sch_sfq sch_red sch_prio sch_pie sch_multiq sch_mqprio sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp cls_flower act_vlan act_simple act_police act_pedit act_ipt act_gact act_csum sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred snd_bcm2835 hid evdev i2c_gpio i2c_algo_bit i2c_dev i2c_core
[ 9157.394882] spi_ks8995 xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 ip6t_NPT ip6t_MASQUERADE nf_nat_masquerade_ipv6 nf_nat nf_conntrack nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 isdn_bsdcomp dss1_divert isdn slhc ifb dm9000 sit sctp libcrc32c tunnel4 ip_tunnel veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm_dmaengine snd_pcm snd_timer snd_mixer_oss snd_hwdep snd_compress snd soundcore mISDN_dsp l1oip mISDN_core
[ 9157.483827] nls_utf8 sha256_generic seqiv jitterentropy_rng drbg md5 hmac ghash_generic gf128mul gcm ctr nls_iso8859_1 nls_cp437 xhci_plat_hcd xhci_hcd ehci_platform ehci_hcd gpio_button_hotplug ptp broadcom bcm_phy_lib pps_core
[ 9157.509826] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.14.123 #0
[ 9157.517784] Hardware name: BCM2835
[ 9157.523009] [<8010f78c>] (unwind_backtrace) from [<8010af58>] (show_stack+0x10/0x14)
[ 9157.534357] [<8010af58>] (show_stack) from [<805d21b0>] (dump_stack+0x98/0xb8)
[ 9157.543422] [<805d21b0>] (dump_stack) from [<80117fe8>] (__warn+0xe4/0x100)
[ 9157.552187] [<80117fe8>] (__warn) from [<8011803c>] (warn_slowpath_fmt+0x38/0x48)
[ 9157.563156] [<8011803c>] (warn_slowpath_fmt) from [<804dfdec>] (dev_watchdog+0x16c/0x268)
[ 9157.574830] [<804dfdec>] (dev_watchdog) from [<8016947c>] (call_timer_fn.constprop.6+0x28/0x94)
[ 9157.587029] [<8016947c>] (call_timer_fn.constprop.6) from [<801697c4>] (run_timer_softirq+0x1c0/0x44c)
[ 9157.599846] [<801697c4>] (run_timer_softirq) from [<80101570>] (__do_softirq+0xe0/0x240)
[ 9157.611422] [<80101570>] (__do_softirq) from [<8011d01c>] (irq_exit+0xd4/0x138)
[ 9157.620523] [<8011d01c>] (irq_exit) from [<80159080>] (__handle_domain_irq+0x9c/0xac)
[ 9157.631807] [<80159080>] (__handle_domain_irq) from [<8010bb0c>] (__irq_svc+0x6c/0x90)
[ 9157.643182] [<8010bb0c>] (__irq_svc) from [<801085d4>] (arch_cpu_idle+0x34/0x38)
[ 9157.654112] [<801085d4>] (arch_cpu_idle) from [<8014e0e4>] (do_idle+0xe0/0x1e0)
[ 9157.663238] [<8014e0e4>] (do_idle) from [<8014e440>] (cpu_startup_entry+0x18/0x1c)
[ 9157.674354] [<8014e440>] (cpu_startup_entry) from [<80800c20>] (start_kernel+0x358/0x364)
[ 9157.686206] ---[ end trace ac9d67b942685282 ]---

P33M, 2019-07-02:
In the case where a network queue stall has happened, what is the output if you do the commands in the linked comment?
https://github.com/raspberrypi/firmware/issues/582#issuecomment-205768216

suyuan168, 2019-07-02:
I created the file usb-net-asix-ax88179.conf in /etc/modules.d as follows:
Options ax88179_178a rtw_power_mgnt=0 rtw_enusbss=0
I disabled automatic sleep of the device. It seems that it has not collapsed yet.
I will try to cancel this file. Let the problem reappear.

suyuan168, 2019-07-02:
Thank you for your help

lemieuxm, 2020-10-02:
Just a quick note for others having a similar problem.  When I added the options @suyuan168 suggested, I found this in my boot logs: 

```
[    7.675318] ax88179_178a: unknown parameter 'rtw_power_mgnt' ignored
[    7.675335] ax88179_178a: unknown parameter 'rtw_enusbss' ignored
```

I'm still looking for a solution to the random "NETDEV WATCHDOG" issues.   In case it helps: 

```
pi@rfw1:~ $ uname -a
Linux rfw1 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux
```

```
Oct  2 21:58:33 rfw1 kernel: [11411.035321] ------------[ cut here ]------------
Oct  2 21:58:33 rfw1 kernel: [11411.035377] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:448 dev_watchdog+0x310/0x314
Oct  2 21:58:33 rfw1 kernel: [11411.035393] NETDEV WATCHDOG: eth1 (ax88179_178a): transmit queue 0 timed out
Oct  2 21:58:33 rfw1 kernel: [11411.035405] Modules linked in: nfnetlink_queue nfnetlink_log nft_chain_nat xt_MASQUERADE xt_nat nf_nat nf_log_ipv4 nf_log_common nft_limit nft_counter xt_LOG xt_limit xt_
state xt_conntrack nf_conntrack nf_defrag_ipv4 nf_defrag_ipv6 xt_tcpudp ipt_REJECT nf_reject_ipv4 nft_compat nf_tables nfnetlink cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc br
cmfmac brcmutil ax88179_178a sha256_generic libsha256 cfg80211 raspberrypi_hwmon rfkill bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) snd_bcm2835(C) videobuf2_dma_co
ntig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common snd_pcm snd_timer snd vc_sm_cma(C) videodev mc fixed uio_pdrv_genirq uio ip_tables x_tables
Oct  2 21:58:33 rfw1 kernel: [11411.035662] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        5.4.51-v7+ #1333
Oct  2 21:58:33 rfw1 kernel: [11411.035667] Hardware name: BCM2835
Oct  2 21:58:33 rfw1 kernel: [11411.035671] Backtrace: 
Oct  2 21:58:33 rfw1 kernel: [11411.035689] [<8010d480>] (dump_backtrace) from [<8010d778>] (show_stack+0x20/0x24)
Oct  2 21:58:33 rfw1 kernel: [11411.035700]  r6:80d00000 r5:00000000 r4:80d95cb8 r3:70470b37
Oct  2 21:58:33 rfw1 kernel: [11411.035714] [<8010d758>] (show_stack) from [<808cdbe4>] (dump_stack+0xe0/0x124)
Oct  2 21:58:33 rfw1 kernel: [11411.035729] [<808cdb04>] (dump_stack) from [<8012002c>] (__warn+0xec/0x104)
Oct  2 21:58:33 rfw1 kernel: [11411.035740]  r8:000001c0 r7:00000009 r6:80b0f6a4 r5:00000000 r4:80d01cec r3:70470b37
Oct  2 21:58:33 rfw1 kernel: [11411.035752] [<8011ff40>] (__warn) from [<801200c8>] (warn_slowpath_fmt+0x84/0xc0)
Oct  2 21:58:33 rfw1 kernel: [11411.035764]  r9:80b0f6a4 r8:000001c0 r7:807db124 r6:00000009 r5:80b0f6bc r4:80d04f48
Oct  2 21:58:33 rfw1 kernel: [11411.035778] [<80120048>] (warn_slowpath_fmt) from [<807db124>] (dev_watchdog+0x310/0x314)
Oct  2 21:58:33 rfw1 kernel: [11411.035790]  r9:0010f440 r8:b3d4a800 r7:00000000 r6:80d03d00 r5:b3d4aaa8 r4:00000000
Oct  2 21:58:33 rfw1 kernel: [11411.035806] [<807dae14>] (dev_watchdog) from [<8019be6c>] (call_timer_fn+0x40/0x180)
Oct  2 21:58:33 rfw1 kernel: [11411.035816]  r8:807dae14 r7:00000100 r6:80d00000 r5:b3d4aaa8 r4:b6b25440
Oct  2 21:58:33 rfw1 kernel: [11411.035830] [<8019be2c>] (call_timer_fn) from [<8019cfbc>] (run_timer_softirq+0x288/0x654)
Oct  2 21:58:33 rfw1 kernel: [11411.035841]  r9:00000000 r8:b3d4aaa8 r7:80d00000 r6:0010f440 r5:80d01dc4 r4:b6b25440
Oct  2 21:58:33 rfw1 kernel: [11411.035855] [<8019cd34>] (run_timer_softirq) from [<8010249c>] (__do_softirq+0x1a4/0x418)
Oct  2 21:58:33 rfw1 kernel: [11411.035866]  r10:00000004 r9:00000082 r8:b645e000 r7:00000100 r6:80d00000 r5:00000001
Oct  2 21:58:33 rfw1 kernel: [11411.035871]  r4:80d03084
Oct  2 21:58:33 rfw1 kernel: [11411.035885] [<801022f8>] (__do_softirq) from [<80126130>] (irq_exit+0x100/0x110)
Oct  2 21:58:33 rfw1 kernel: [11411.035896]  r10:b73ff900 r9:80d00000 r8:b645e000 r7:00000001 r6:00000000 r5:00000000
Oct  2 21:58:33 rfw1 kernel: [11411.035901]  r4:80ca42e4
Oct  2 21:58:33 rfw1 kernel: [11411.035915] [<80126030>] (irq_exit) from [<80181200>] (__handle_domain_irq+0x70/0xc4)
Oct  2 21:58:33 rfw1 kernel: [11411.035928] [<80181190>] (__handle_domain_irq) from [<80102228>] (bcm2836_arm_irqchip_handle_irq+0x60/0xa4)
Oct  2 21:58:33 rfw1 kernel: [11411.035939]  r8:80d9b18d r7:80d01f0c r6:ffffffff r5:60000013 r4:00000000 r3:80d01ed8
Oct  2 21:58:33 rfw1 kernel: [11411.035951] [<801021c8>] (bcm2836_arm_irqchip_handle_irq) from [<80101a3c>] (__irq_svc+0x5c/0x7c)
Oct  2 21:58:33 rfw1 kernel: [11411.035957] Exception stack(0x80d01ed8 to 0x80d01f20)
Oct  2 21:58:33 rfw1 kernel: [11411.035970] 1ec0:                                                       80109b94 00000000
Oct  2 21:58:33 rfw1 kernel: [11411.035982] 1ee0: 60000093 8011a520 80d04f68 80d00000 80d04fb0 00000001 80d9b18d 80c69a40
Oct  2 21:58:33 rfw1 kernel: [11411.035993] 1f00: b73ff900 80d01f34 80d0535c 80d01f28 00000000 80109b98 60000013 ffffffff
Oct  2 21:58:33 rfw1 kernel: [11411.035999]  r4:80109b98 r3:70470b37
Oct  2 21:58:33 rfw1 kernel: [11411.036017] [<80109b64>] (arch_cpu_idle) from [<808eea44>] (default_idle_call+0x34/0x48)
Oct  2 21:58:33 rfw1 kernel: [11411.036030] [<808eea10>] (default_idle_call) from [<80153bc0>] (do_idle+0xec/0x170)
Oct  2 21:58:33 rfw1 kernel: [11411.036040] [<80153ad4>] (do_idle) from [<80153f20>] (cpu_startup_entry+0x28/0x2c)
Oct  2 21:58:33 rfw1 kernel: [11411.036052]  r8:00000000 r7:80c69a40 r6:80dac5c0 r5:80dac618 r4:000000ce r3:80d01f50
Oct  2 21:58:33 rfw1 kernel: [11411.036063] [<80153ef8>] (cpu_startup_entry) from [<808e7e68>] (rest_init+0xbc/0xc0)
[... cut ...] orking a few times each day.  A quick reboot fixes it, but kind of annoying to have to keep doing it.  

```
root@rfw1:~# ./bin/test_ax88179.sh 
... (several times)  
root@rfw1:~# ./bin/test_ax88179.sh 
root@rfw1:~# cat test.txt  | uniq
Reg@0x000418 = 0x0000000c
Reg@0x000418 = 0x00000018
Reg@0x000418 = 0x0000000a
Reg@0x000418 = 0x00000050
Reg@0x000418 = 0x00000084
Reg@0x000418 = 0x00000014
Reg@0x000418 = 0x000000c0
Reg@0x000418 = 0x00000012
Reg@0x000418 = 0x00000048
Reg@0x000418 = 0x000000a0
```