raspberrypi/linux #3042: CONFIG_USB_CONFIGFS missing from bcmrpi_defconfig

ali1234, 2019-07-02:
This configuration is included in the new Pi 4 kernel but not the Pi Zero kernel.

It is needed for the new composite gadget modes defined through configfs. Without it we can only use the legacy drivers like g_serial and g_webcam which can only work one at a time.

pelwell, 2019-07-02:
Do you know the impact on core kernel size of enabling CONFIG_USB_CONFIGFS?

ali1234, 2019-07-02:
I don't know off-hand but I could test it. Will take about half an hour to build the kernel twice.

ali1234, 2019-07-02:
Although since it is a module the impact on core kernel size is 0...

ali1234, 2019-07-02:
The module itself is 48kB and there is no change to the (compressed) kernel size (it actually got smaller between tests).

pelwell, 2019-07-02:
Any objections, @popcornmix?

popcornmix, 2019-07-02:
Seems fine.

pelwell, 2019-07-02:
Pushed to rpi-4.19.y.

ali1234, 2019-07-02:
You have to say "y" to the drivers that should be available as well (not shown until you set CONFIG_USB_CONFIGFS to y or m). These aren't modules themselves - it just tells configfs to expose them. They do select the modules they require though. bcm2711_defconfig has everything selected except audio class 1.0 legacy.

pelwell, 2019-07-02:
Also pushed.

6by9, 2019-07-02:
So these are now in bcmrpi_defconfig and bcm2711_defconfig.
Sufficient gain in having them in bcm2709_defconfig for Pi3A+?

ali1234, 2019-07-02:
Good point. CM3 can theoretically use them too.

JamesH65, 2019-07-30:
@ali1234 Looks like this can be closed?

ali1234, 2019-07-30:
There is just the outstanding question of whether you want to add the same options in the bcm2709_defconfig. This would be of use for 3A+ and CM3 users.