raspberrypi/linux #3048: Space constraints on the /boot partition

ThomasKaiser, 2019-07-04:
I noticed Raspbian increased partition size of the FAT partition holding ThreadX and kernel/DT to 256 MB.

I'm currently maintaining an OS image for RPi 2 to 4 that uses a FAT partition which is just 64 MB in size:

    /dev/mmcblk0p1   63M   40M   23M  63% /boot

The contents as follows:

    root@raspberrypi:/# ls -al /boot/
    total 39533
    drwxr-xr-x  4 root root    3584 Jan  1  1970 .
    drwxr-xr-x 24 root root    4096 May 31  2018 ..
    -rwxr-xr-x  1 root root   23501 Jul  1 13:51 bcm2708-rpi-b.dtb
    -rwxr-xr-x  1 root root   23760 Jul  1 13:51 bcm2708-rpi-b-plus.dtb
    -rwxr-xr-x  1 root root   23278 Jul  1 13:51 bcm2708-rpi-cm.dtb
    -rwxr-xr-x  1 root root   23226 Jul  1 13:51 bcm2708-rpi-zero.dtb
    -rwxr-xr-x  1 root root   23962 Jul  1 13:51 bcm2708-rpi-zero-w.dtb
    -rwxr-xr-x  1 root root   24848 Jul  1 13:51 bcm2709-rpi-2-b.dtb
    -rwxr-xr-x  1 root root   26018 Jul  1 13:51 bcm2710-rpi-3-b.dtb
    -rwxr-xr-x  1 root root   26637 Jul  1 13:51 bcm2710-rpi-3-b-plus.dtb
    -rwxr-xr-x  1 root root   24816 Jul  1 13:51 bcm2710-rpi-cm3.dtb
    -rwxr-xr-x  1 root root   39502 Jul  1 13:51 bcm2711-rpi-4-b.dtb
    -rwxr-xr-x  1 root root   52296 Jul  1 13:51 bootcode.bin
    -rwxr-xr-x  1 root root     143 Jun 26 22:18 cmdline.txt
    -rwxr-xr-x  1 root root    1750 Jul  2 13:12 config.txt
    -rwxr-xr-x  1 root root   18693 Jul  1 13:51 COPYING.linux
    -rwxr-xr-x  1 root root      41 Nov  5  2016 .firmware_revision
    -rwxr-xr-x  1 root root    3033 Jul  1 13:50 fixup4cd.dat
    -rwxr-xr-x  1 root root    6069 Jul  1 13:50 fixup4.dat
    -rwxr-xr-x  1 root root    9145 Jul  1 13:51 fixup4db.dat
    -rwxr-xr-x  1 root root    9145 Jul  1 13:51 fixup4x.dat
    -rwxr-xr-x  1 root root    2643 Jul  1 13:51 fixup_cd.dat
    -rwxr-xr-x  1 root root    6724 Jul  1 13:50 fixup.dat
    -rwxr-xr-x  1 root root    9800 Jul  1 13:51 fixup_db.dat
    -rwxr-xr-x  1 root root    9800 Jul  1 13:51 fixup_x.dat
    drwxr-xr-x  2 root root     512 Jul  1 15:44 .fseventsd
    -rwxr-xr-x  1 root root     145 Jun 20 16:59 issue.txt
    -rwxr-xr-x  1 root root 5305768 Jul  1 13:51 kernel7.img
    -rwxr-xr-x  1 root root 5608912 Jul  1 13:51 kernel7l.img
    -rwxr-xr-x  1 root root 5025568 Jul  1 13:51 kernel.img
    -rwxr-xr-x  1 root root    1494 Jul  1 13:51 LICENCE.broadcom
    drwxr-xr-x  2 root root   14848 Jul  1 13:51 overlays
    -rwxr-xr-x  1 root root  762400 Jul  1 13:50 start4cd.elf
    -rwxr-xr-x  1 root root 4715912 Jul  1 13:50 start4db.elf
    -rwxr-xr-x  1 root root 2758660 Jul  1 13:50 start4.elf
    -rwxr-xr-x  1 root root 3672072 Jul  1 13:50 start4x.elf
    -rwxr-xr-x  1 root root  685508 Jul  1 13:50 start_cd.elf
    -rwxr-xr-x  1 root root 4854056 Jul  1 13:50 start_db.elf
    -rwxr-xr-x  1 root root 2878148 Jul  1 13:50 start.elf
    -rwxr-xr-x  1 root root 3791528 Jul  1 13:50 start_x.elf

This [OpenMediaVault](https://www.openmediavault.org) image is based on Debian Stretch (`armhf` architecture) and we only pull in the following packages from archive.raspberrypi.org: `libraspberrypi-bin libraspberrypi0 raspberrypi-bootloader raspberrypi-kernel` to be up to date with kernel and ThreadX fixes.

I'm concerned about the space on the `/boot` partition especially since a 64-bit kernel is announced. Are you taking into account that there are other 'consumers' of the kernel/ThreadX packages you build or do you only care about Raspbian (which seems to prepare for growing contents on the FAT partition)?

Strit, 2019-07-04:
In Manjaro we use a 100MB partition for our 64-bit kernels.

Why can't you just increase the partition in your build tools?

ThomasKaiser, 2019-07-04:
> Why can't you just increase the partition in your build tools?

How should this solve problems with already existing installations? Based on count of downloads maybe 200,000 installations are affected.

pelwell, 2019-07-04:
None of the files in the current release are gratuitous - all are necessary for some use cases and for now. We may get to the point where we can revert to a common firmware for all Pis, i.e. remove the start4*.elf and fixup4*.dat, but I'm not going to make any promises to that end. The obvious way to keep a distribution smaller is to omit some of the firmware variants - `db` being the obvious candidate - but we aren't going to limit the capabilities of the firmware just to keep it within 56/64MB, so it would be wise to think about a migration strategy.

ThomasKaiser, 2019-07-04:
> The obvious way to keep a distribution smaller is to omit some of the firmware variants - db being the obvious candidate

I've to admit that I'm not familiar with any firmware details but I believe it's about deleting `start4db.elf` and `start_db.elf`? While something like this would be possible I assume it doesn't really help since when you're upgrading the `raspberrypi-bootloader` package on archive.raspberrypi.org and the new version will be installed it might end up with 'filesystem full' during package install anyway...

About a migration strategy I've no idea how an automated unattended partition resize operation on 200,000 user installations should work.

pelwell, 2019-07-04:
There are other ways to upgrade the firmware - all the individual files are available on GitHub in various locations - see `rpi-update` for an example of one way of doing it, or you could host your own subset package.

pelwell, 2019-07-04:
Another way would be to continue using the Raspbian package, but for /boot to be a directory in your ext4 partition that, with the FAT partition mounted elsewhere and updated after the common firmware update.

ThomasKaiser, 2019-07-04:
Thank you for this brilliant suggestion. Something like this as a preparation step followed by an rsync from `/boot` to `/firmware` excluding at least `kernel.img`, `start4db.elf` and `start_db.elf` should do the job:

    umount /boot
    mkdir -m775 /firmware
    sed -i -e 's|/boot|/firmware|' /etc/fstab
    mount /firmware
    rsync -a /firmware/* /boot/

Anyway: I take your comments here as a confirmation that installations using those upstream packages from archive.raspberrypi.org with a FAT partition just 64MB in size are about to be bricked sometimes in the future...

closed 2019-07-04 by ThomasKaiser

pelwell, 2019-07-04:
Don't forget you can omit fixup4db.dat and fixup_db.dat as well, even though they aren't very large. Also, if you are worried about unattended upgrades you could differentiate between new images with >64MB FAT and old with <=64MB, and only support Pi4 on a new image. That would allow you to strip out all of the Pi4 support files.

>  I take your comments here as a confirmation that installations using those upstream packages from archive.raspberrypi.org with a FAT partition just 64MB in size are about to be bricked sometimes in the future...

That's an unnecessarily pessimistic view, but it would be wise to think ahead to the Pi12 launch...