raspberrypi/linux #3051: Recent change "Model Divider in CCF" broke VC4 on Model B

kirgene, 2019-07-05:
Commit [i2c: bcm2835: Model Divider in CCF](https://github.com/raspberrypi/linux/commit/2f8d963db206ce596f9a9e951ec425e9c3e1b4d9) broke VC4 on Raspberry Pi Model B.

After applying above commit TV screen becomes blank and the following errors appear in dmesg:
```
kernel: [drm:vc4_hdmi_bind] *ERROR* Failed to get pixel clock
kernel: vc4-drm soc:gpu: failed to bind 20902000.hdmi (ops vc4_hdmi_ops): -517
kernel: vc4-drm soc:gpu: master bind failed: -517
```

popcornmix, 2019-07-05:
Are you using a Pi 1 model B?
Are you using ``dtoverlay=vc4-kms-v3d``? Any other config.txt settings?

lategoodbye, 2019-07-05:
There is a pull-request #3018 to fix this issue.

pelwell, 2019-07-05:
Now merged.

kirgene, 2019-07-06:
Great, it works now, thanks!

closed 2019-07-06 by kirgene