raspberrypi/linux #3055: mmap()-ing /dev/gpiomem should return MAP_FAILED if accessing unsupported offset

hzeller, 2019-07-07:
**Describe the bug**
When `mmap()`-ing a region from `/dev/gpiomem` that is _not_ in the range of supported GPIO, mmap() is not returning `MAP_FAILED`, but instead returns a memory region which is not the requested one (it actually seems to be fixed to the gpio register offset).

This is dangerous, as a caller might assume that the requested offset is supported and proceed to manipulate registers different from what they intended to change.

**To reproduce**
```c
/* gcc gpiommap.c ; ./a.out */
#include <fcntl.h>
#include <stdint.h>
#include <stdio.h>
#include <sys/mman.h>
#include <sys/stat.h>
#include <sys/types.h>

// Choose proper range. Note: with the bug present, it will not actually make a difference.
#define BCM2708_PERI_BASE        0x20000000  // Pi 1
#define BCM2709_PERI_BASE        0x3F000000  // Pi 2, 3
#define BCM2711_PERI_BASE        0xFE000000  // Pi 4

#define PERI_BASE BCM2709_PERI_BASE

int main() {
	int fd = open("/dev/gpiomem", O_RDWR|O_SYNC);
	if (fd < 0) { perror("open(/dev/gpiomem)"); return 1; }

	const off_t not_at_all_gpio_register_offset = 8 * 0x200000; // something bogus
	uint32_t *result =
		(uint32_t*) mmap(NULL,   // no specific target address.
				 4096,   // Map length
				 PROT_READ|PROT_WRITE,
				 MAP_SHARED,
				 fd,
				 PERI_BASE + not_at_all_gpio_register_offset);
	if (result != MAP_FAILED) {
		fprintf(stderr, "mmap() with offset 0x%08x + 0x%08x should've "
			"failed, but succeeded; p=%p\n",
			PERI_BASE, not_at_all_gpio_register_offset, result);
		return 1;
	} else {
		fprintf(stderr, "Properly rejected mmap() request\n");
		return 0;
	}
}
```

**Expected behaviour**
`mmap()`-ing an offset outside the supported range should return `MAP_FAILED`.

**Actual behaviour**
`mmap()` silently succeeds, returning a memory range that the caller did not intent to map.

**System**
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
    Any Pi. Here tested on a Pi3 and Pi4, but the implementation of /dev/gpiomem is fundamentally broken.

* Which OS and version (`cat /etc/rpi-issue`)?
 Tested on
  Raspberry Pi reference 2017-08-16
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, aaaf0aea13f1e4f0375c31c87d7389299228598d, stage2
 
   Raspberry Pi reference 2019-06-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 150e25c4f8123a4c9c63e8dca1b4737fa6c1135c, stage2

* Which firmware version (`vcgencmd version`)?
   Not relevant, but
    version a3d7660e6749e75e2c4ce4d377846abd3b3be283
    version 407b1da8fa3d1a7108cb1d250f5064a3420d2b7d
* Which kernel version (`uname -a`)?
    Linux nope3 4.9.41-v7+ #1023 SMP Tue Aug 8 16:00:15 BST 2017 armv7l GNU/Linux
    Linux pi4 4.19.50-v7l+ #895 SMP Thu Jun 20 16:03:42 BST 2019 armv7l GNU/Linux

**Additional context**
The implementation seems to be intetionally broken, as it _does not range check_ the requested offset to return an error, but instead just returns a mmap of its own choosing, without verifying that this was the same as requested:

https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/char/broadcom/bcm2835-gpiomem.c#L106

**What needs to be done**
Call parameter  _must_ be validated. If it is not exactly the same offset as gpiomem would return regardless, mmap() should fail instead of succeeding silently, returning a non-requested memory region.

I could imagine a reason why this is behaving like this: maybe the intention was to have ranges work independent of the pi-specific base address of the periphery (`0x20000000`, `0x3F000000` or `0xFE000000`) ?

So if the intention was to be idempotent regardless of the provided peripheral base address, then there is still a way to fix this: at the very least remove the peripheral base and mask off the offset with `0xFFFFFF` to just get the offset the user intents to map and verify that the remainder is _exactly_ `0x200000`, which is the GPIO base address. Otherwise return failure.
(This will provide ways to add more supported ranges later, if there are other ranges that should be accessible from user-space)

pelwell, 2019-07-08:
/dev/gpiomem is behaving as intended.

closed 2019-07-08 by pelwell

hzeller, 2019-07-08:
But what is then intention ? Failing the user silently by returning a non-requested range instead of `MAP_FAILED` can't possibly be it.

pelwell, 2019-07-08:
That is precisely the intention:
https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/char/broadcom/bcm2835-gpiomem.c#L106
This is a very useful driver that we have no intention of upstreaming or changing for anybody else.

hzeller, 2019-07-08:
I agree that this device is useful in the way it allows non-root users to access GPIO functionality. And my suggestion will not take that away, it will just make sure that it is not broken.

Yes, I have read that comment by the original author, but this looks like a 'don't care I want to get this finished quickly' argument than a consideration of what actually should be done.

It is like saying we want to restrict access to a particular file, but when that is attempted to open a different file not return `EPERM`  but instead open the other file instead with no indication that this happened. This is surprising at best, but typically just plain dangerous. If more of these things creep into the kernel we end up with a very unusable system where things just behave randomly as application developers will not get the necessary feedback for their error handling.

What _should_ be done is to at least test that the last 24 bits in the requested offset are exactly `0x200000`. It is simple, it is the right way to do, and it will not break anyone who is using `/dev/gpiomem` properly (i.e. use as /dev/mem but with a restricted range).
But it _will_ allow developers to see what is available and not let them confused.

Ferroin, 2019-07-08:
> What should be done is to at least test that the last 24 bits in the requested offset are exactly 0x200000. It is simple, it is the right way to do, and it will not break anyone who is using /dev/gpiomem properly (i.e. use as /dev/mem but with a restricted range).

It's debatable whether that's 'correct' usage or not.  The device file itself is supposed to only map the GPIO memory region, so it's just as logical for offset 0 in the file to correspond to offset zero within the GPIO memory region instead of physical address 0x00000000.  If that is the intended behavior of the driver (and I suspect it has been since day one), then the fact that trying to use it like you are (map the actual address of the GPIO memory region in physical memory) works is just there for compatibility.

hzeller, 2019-07-08:
If that was the intention of the driver, then it would check if the requested offset is 0x0, but it doesn't.
As I showed above, it does not do _any_ range checking that would express its intent.
This is dangerous.

Ferroin, 2019-07-08:
> If that was the intention of the driver, then it would check if the requested offset is 0x0, but it doesn't.
As I showed above, it does not do any range checking that would express its intent.

No, it would not necessarily need to have that check if that is the intent.   The fixed mapping behavior is pretty obviously a compatibility hack so that software that was written to poke at the relevant GPIO MMIO region via `/dev/mem` will work without modification using `/dev/gpiomem`.

> This is dangerous.

Actually, barring the case of doing already risky things with the GPIO pins, it's safer right now than if there was range checking.  The fact that the user-supplied location and range are ignored mean that those can't ever be used to trick the driver into mapping memory it shouldn't.

mzandrew, 2020-06-18:
I'd just like to add that because of the existing behavior, I spent a couple hours trying to guess why I couldn't set the drive strength by mmap'ing /dev/gpiomem.  I eventually figured it out (need to open /dev/mem and mmap for the drive strength page of registers [offset 0x100000 instead of 0x200000], which must be done sudo), but some others haven't yet.

I found a couple examples online of current code in various projects where they open /dev/gpiomem, mmap twice (once for gpio_port and once for gpio_pads) and blindly set the wrong register because (presumably) when the code originally said open /dev/mem, it worked, and they were asked to support sudo-less mode and changed it to /dev/gpiomem.  It doesn't *obviously* fail (but the drive strength is NOT set right, even when run with sudo).  BTW, I am confirming with an oscilloscope to look at the ringing to tell whether my setting of the drive strength is working.

So +1 for hzeller's feature request.

hzeller, 2020-06-18:
It is not even a feature request, but a not-broken request :)