raspberrypi/linux #3061: dtoverlay=i2c-sensor,lm75 Not working in latest Kernels

Trotter73, 2019-07-08:
**Describe the bug**
config.txt Parameter dtoverlay=i2c-sensor,lm75  stopped working on recent Kernal updates, possibly after 4.19.42+ Update

**To reproduce**
Add config.txt Parameter dtoverlay=i2c-sensor,lm75, reboot

**Expected behaviour**

dmesg|grep lm75

You should see something like this if everything is working correctly...

[ 6.738872] lm75 1-004f: hwmon0: sensor 'lm75'

**Actual behaviour**

sudo i2cdetect -y 1, address is not "UU"'d, sensor is not detected.

**System**

System Information
------------------

Raspberry Pi 3 Model B Plus Rev 1.3 
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"

Raspberry Pi reference 2019-04-08
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, e910bf55ac92e27c6e77043e8995bcb5e0a2d742, stage5

Linux raspberrypi3bplus 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l GNU/Linux
Hardware	: BCM2835
Revision	: a020d3
Serial		: 000000006f08a419
Throttled flag  : throttled=0x80000
Camera          : supported=0 detected=0

Videocore information
---------------------

Mar 27 2019 15:45:53 
Copyright (c) 2012 Broadcom
version 2e98b31d18547962e564bdf88e57b3df7085c29b (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem            1K-blocks      Used Available Use% Mounted on
/dev/root             115271168   3611232 106905188   4% /
devtmpfs                 342316         0    342316   0% /dev
tmpfs                    473900         0    473900   0% /dev/shm
tmpfs                    473900     12312    461588   3% /run
tmpfs                      5120         4      5116   1% /run/lock
tmpfs                    473900         0    473900   0% /sys/fs/cgroup
/dev/sda1                 43234     22683     20551  53% /boot
//192.168.1.6/Dropbox 731942908 467316336 264626572  64% /home/pi/Dropbox
tmpfs                     94780         0     94780   0% /run/user/1000

Filename				Type		Size	Used	Priority
/var/swap                               file		102396	0	-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: 1.20190219
[... 442 lines cut ...]
[    9.512781] brcmfmac: power management disabled
[    9.876870] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   15.030718] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   15.530675] Bluetooth: Core ver 2.22
[   15.530777] NET: Registered protocol family 31
[   15.530785] Bluetooth: HCI device and connection manager initialized
[   15.530808] Bluetooth: HCI socket layer initialized
[   15.530821] Bluetooth: L2CAP socket layer initialized
[   15.530862] Bluetooth: SCO socket layer initialized
[   15.545374] Bluetooth: HCI UART driver ver 2.3
[   15.545390] Bluetooth: HCI UART protocol H4 registered
[   15.545493] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   15.545719] Bluetooth: HCI UART protocol Broadcom registered
[   15.745578] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   15.745590] Bluetooth: BNEP filters: protocol multicast
[   15.745615] Bluetooth: BNEP socket layer initialized
[   15.906478] Bluetooth: RFCOMM TTY layer initialized
[   15.906521] Bluetooth: RFCOMM socket layer initialized
[   15.906576] Bluetooth: RFCOMM ver 1.11
[   24.278579] FS-Cache: Netfs 'cifs' registered for caching
[   24.279422] Key type cifs.spnego registered
[   24.279441] Key type cifs.idmap registered
[   24.281439] No dialect specified on mount. Default has changed to a more secure dialect, SMB2.1 or later (e.g. SMB3), from CIFS (SMB1). To use the less secure SMB1 dialect to access old servers which do not support SMB3 (or SMB2.1) specify vers=1.0 on mount.
[   24.943384] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k FS
[   28.593244] fuse init (API version 7.27)

pelwell, 2019-07-08:
Please download a new version of /usr/lib/raspberrypi-sys-mods/i2cprobe from here: https://raw.githubusercontent.com/pelwell/raspberrypi-sys-mods/is3061/usr/lib/raspberrypi-sys-mods/i2cprobe

I think it will fix the problem for you by providing support for the old way of loading I2C modules.

Trotter73, 2019-07-08:
Yup that's fixed it, perfect !! Thanks a million for such a speedy response !! Shall I close the issue  ?

pelwell, 2019-07-08:
With your permission I'll make sure it gets closed once the fix makes its way into the raspberrypi-sys-mods Raspbian package.

Trotter73, 2019-07-08:
No problem, close when you see fit !! :)

closed 2020-06-01 by pelwell

AloneWarrior11011, 2025-04-10:
> With your permission I'll make sure it gets closed once the fix makes its way into the raspberrypi-sys-mods Raspbian package.

```
root@raspberrypi3:~# cat /etc/os-release
ID=openbmc-phosphor
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.18.0-dev"
VERSION_ID=2.18.0-dev-696-g5fc9038b35-dirty
VERSION_CODENAME="scarthgap"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.18.0-dev"
CPE_NAME="cpe:/o:openembedded:openbmc-phosphor:2.18.0-dev-696-g5fc9038b35-dirty"
BUILD_ID="20250325043000"
OPENBMC_TARGET_MACHINE="raspberrypi3"
EXTENDED_VERSION="2.18.0-dev-696-g5fc9038b35-dirty"
```

I have specify dtoverlay=i2c-sensor,lm75 
and this not working in my case too. 
```
root@raspberrypi3:~# dmesg | grep lm75
root@raspberrypi3:~# lsmod | grep lm75
```
shows nothing....that means driver not loaded. 
do help me with this context.

AloneWarrior11011, 2025-04-10:
> > With your permission I'll make sure it gets closed once the fix makes its way into the raspberrypi-sys-mods Raspbian package.
> 
> ```
> root@raspberrypi3:~# cat /etc/os-release
> ID=openbmc-phosphor
> NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
> VERSION="2.18.0-dev"
> VERSION_ID=2.18.0-dev-696-g5fc9038b35-dirty
> VERSION_CODENAME="scarthgap"
> PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.18.0-dev"
> CPE_NAME="cpe:/o:openembedded:openbmc-phosphor:2.18.0-dev-696-g5fc9038b35-dirty"
> BUILD_ID="20250325043000"
> OPENBMC_TARGET_MACHINE="raspberrypi3"
> EXTENDED_VERSION="2.18.0-dev-696-g5fc9038b35-dirty"
> ```
> 
> I have specify dtoverlay=i2c-sensor,lm75 and this not working in my case too.
> 
> ```
> root@raspberrypi3:~# dmesg | grep lm75
> root@raspberrypi3:~# lsmod | grep lm75
> ```
> 
> shows nothing....that means driver not loaded. do help me with this context.
> 
> does is that the reason device not created under sysfs on rpi3.

6by9, 2025-04-10:
This issue was closed in 2019. Any issue you're encountering now is very unlikely to be directly related.

Raspberry Pi have nothing to do with Phosphor OpenBMC.
https://github.com/openbmc/openbmc implies it is a Yocto based system of some form, with bugs tracked via https://github.com/openbmc/openbmc/issues, or using the openbmc tag on Stack Overflow.

Being Yocto based means that it is exceedingly unlikely to make use of the raspberrypi-sys-mods package from Raspberry Pi OS/Raspbian.