raspberrypi/linux #3062: Kernel 4.19.50+ dtoverlay=i2c-gpio, bus 3 and 4 may be mislabeled

gamerl3, 2019-07-08:
**Is this the right place for my bug report?**
This relates to entries in config.txt. It could be either kernel or firmware. I don't know where the dividing line is.

**Describe the bug**
When I add two dtoverlay=i2c-gpio lines to config.txt for bus=3 and bus=4, the resulting /dev names are not consistent. If the bus=3 line precedes the bus=4 line in config.txt, then the /dev names are reversed: the one I entered as bus=3 has the name i2c-4, and vice versa. However, if the bus=4 line is first in config.txt, then they are named the same as the bus number.

**To reproduce**
Enter two dtoverlay=i2c-gpio lines in config.txt, one with bus=3, the other with bus=4, and the respective data and clock pins. Then reboot. Read/write the attached devices. If you entered 3 before 4, the devices will be swapped. If you entered 4 before 3, then the device name will be the same as the bus number.

**Expected behaviour**
My expectation is that the device I enter as bus=3 will be /dev/i2c-3, and device I enter as bus=4 will be /dev/i2c-4. It should not matter which is entered first in config.txt.

**Actual behaviour**
It appears that the bus=x part of the dtoverlay=i2c-gpio entry has no meaning. No matter what you enter in bus=x, the last dtoverlay line will be /dev/i2c-3, the one before that will be /dev/i2c-4. I presume, if you also entered a 3rd dtoverlay=i2c-gpio line, the first one in config.txt would be /dev/i2c-5, but I did not test this.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi1B+

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-06-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 150e25c4f8123a4c9c63e8dca1b4737fa6c1135c, stage4

* Which firmware version (`vcgencmd version`)?
Jun 20 2019 16:12:41
Copyright (c) 2012 Broadcom
version a59fb7a74180be0111dbc5c18a37ec6df86f14a3 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux rasppi4 4.19.50+ #896 Thu Jun 20 16:09:52 BST 2019 armv6l GNU/Linux

**Logs**
n/a

**Additional context**
n/a

gamerl3, 2019-07-08:
from config.txt:
# add a software driven i2c bus, dat12=p32, clk16=p36 (add higher num bus first)
dtoverlay=i2c-gpio,bus=4,i2c_gpio_sda=12,i2c_gpio_scl=16,i2c_gpio_delay_us=8
# add a software driven i2c bus, dat23=p16, clk24=p18, /boot/overlays/README
dtoverlay=i2c-gpio,bus=3,i2c_gpio_sda=23,i2c_gpio_scl=24,i2c_gpio_delay_us=8

from dmesg:
[    6.168076] i2c /dev entries driver
[    7.745656] systemd[1]: Started File System Check Daemon to report status.
[    7.963924] systemd[1]: Started Apply Kernel Variables.
[    8.443904] systemd[1]: Started File System Check on Root Device.
[    8.505474] systemd[1]: Starting Remount Root and Kernel File Systems...
[    8.814920] systemd[1]: Started Journal Service.
[    9.581655] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[   10.215281] systemd-journald[88]: Received request to flush runtime journal from PID 1
[   12.597605] gpio-23 (i2c@3): enforced open drain please flag it properly in DT/ACPI DSDT/board f$
[   12.597723] gpio-24 (i2c@3): enforced open drain please flag it properly in DT/ACPI DSDT/board f$
[   12.599189] i2c-gpio i2c@3: using lines 23 (SDA) and 24 (SCL)
[   12.599668] gpio-12 (i2c@4): enforced open drain please flag it properly in DT/ACPI DSDT/board f$
[   12.599769] gpio-16 (i2c@4): enforced open drain please flag it properly in DT/ACPI DSDT/board f$
[   12.601535] i2c-gpio i2c@4: using lines 12 (SDA) and 16 (SCL)

pelwell, 2019-07-09:
It looks like the bus numbering functionality has been broken for a long time - it's cause by the refusal of the firmware to create a "reg" property as the result of an overlay parameter if one doesn't already exist. Fortunately there is a sensible default value that can be used by the overlay that causes the auto-generated number to be used unless overridden by the "bus" parameter.

A fixed overlay will be in the next firmware release, but until then you can download the updated version from here: https://drive.google.com/file/d/1dTwQ3sz_q7oQyu2ytNrEF61m27beu_pz/view?usp=sharing
After downloading the file, copy it to /boot/overlays (you'll need to "sudo cp i2c-gpio.dtbo /boot").

gamerl3, 2019-07-09:
Thanks. That was quick! (And thanks for the edit of my first comment.)

pelwell, 2019-07-09:
Let me know when you've had chance to try it so we can close the issue.

gamerl3, 2019-07-09:
The test worked. The issue is fixed. Thank you.

closed 2019-07-09 by gamerl3