raspberrypi/linux #3067: Pi 3+ buster pxe nfsboot - kernel hangs / "Fatal exception in interrupt" panics

maxnet, 2019-07-12:
**Describe the bug**

PXE/nfs booting Buster on a Pi 3+ does not quite work

Does boot the kernel and starts to execute the first systemd units, but then the system hangs some seconds after on random moments (text cursor stops blinking, no activity, numlock does not respond)

Happens both with the kernel version you ship by default in your June 2019 Buster image, and 4.19.57 you get after running "apt-get upgrade"

In most cases kernels hangs without outputing any message, but seen the following kernel panic as well.

```
         Starting Helper to synchronize boot up for ifupdown...
[  OK  ] Mounted /tmp.
[  OK  ] Mounted /media.
[  OK  ] Mounted /var/lib/sudo.
[  OK  ] Mounted /var/lib/dhcp.
[  OK  ] Mounted /var/tmp.
[  OK  ] Mounted /home.
[  OK  ] Mounted /var/log.
[  OK  ] Mounted /var/cache.
[  OK  ] Mounted /var/spool.
[  OK  ] Started udev Kernel Device Manager.
[  OK  ] Mounted /var/lib/systemd/rfkill.
[  OK  ] Mounted /var/lib/lightdm.
[  OK  ] Mounted /var/lib/systemd/coredump.
[  OK  ] Started Helper to synchronize boot up for ifupdown.
         Starting Flush Journal to Persistent Storage...
[  OK  ] Reached target Local File Systems.
         Starting Preprocess NFS configuration...
         Starting Set console font and keymap...
         Starting Raise network interfaces...
[  OK  ] Started Preprocess NFS configuration.
[  OK  ] Reached target NFS client services.
[  OK  ] Reached target Remote File Systems (Pre).
[  OK  ] Reached target Remote File Systems.
[   14.481793] Unable to handle kernel paging request at virtual address f74193c8
[   14.500886] pgd = 30af461a
[   14.515245] [f74193c8] *pgd=00000000
[   14.530421] Internal error: Oops: 80000005 [#1] SMP ARM
[   14.547260] Modules linked in: bcm2835_v4l2(C+) v4l2_mem2mem snd_timer v4l2_common bcm2835_mmal_vchiq(C) snd videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media vc_sm_cma(C) uio_pdrv_genirq uio fixed i2c_dev ip_tables x_tables ipv6
[   14.597249] CPU: 0 PID: 9 Comm: ksoftirqd/0 Tainted: G         C        4.19.57-v7+ #1244
[   14.617972] Hardware name: BCM2835
[   14.633734] PC is at 0xf74193c8
[   14.649171] LR is at rcu_process_callbacks+0x2cc/0x674
[   14.666604] pc : [<f74193c8>]    lr : [<8018e0b8>]    psr: a0000013
[   14.685285] sp : b9d3de58  ip : b9d3de58  fp : b9d3decc
[   14.702969] r10: b9d3c018  r9 : 0000000a  r8 : 80d10c80
[   14.720594] r7 : b9d3de8c  r6 : ba352660  r5 : 80d10780  r4 : ba352640
[   14.739565] r3 : f74193c8  r2 : f74193c8  r1 : b9d3de08  r0 : b7419654
[   14.758469] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   14.778024] Control: 10c5383d  Table: 376fc06a  DAC: 00000055
[   14.796173] Process ksoftirqd/0 (pid: 9, stack limit = 0x1631a8d3)
[   14.814825] Stack: (0xb9d3de58 to 0xb9d3e000)
[   14.831548] de40:                                                       00000000 80d04d48
[   14.852230] de60: ba351e40 00000001 80d10780 80d97784 80d04d48 80d04d70 80d8ed74 80c903f0
[   14.872765] de80: 00000001 80d04e10 80102374 f74193c8 b8350000 fffffffa ffffffff a271c001
[   14.893156] dea0: b9d3decc 80d030a4 00000009 b9d3c010 00000100 80d0be48 00000024 00000200
[   14.913429] dec0: b9d3df34 b9d3ded0 80102410 8018ddf8 802018e8 80202aec 00000000 04208040
[   14.933588] dee0: 80d03d00 ffff907a 00000009 80c94a00 80d8f278 80d04d70 80c8c308 80d03080
[   14.953722] df00: b9d3ded0 0000000a b9d3df34 b9d3c000 b9ccb380 00000000 00000001 80d0be48
[   14.973701] df20: 80d04d48 00000000 b9d3df44 b9d3df38 8012664c 8010228c b9d3df7c b9d3df48
[   14.993483] df40: 80146b1c 80126614 b9d40000 a271c001 b9d3df7c b9ccb580 b9ccb540 00000000
[   15.013156] df60: b9ccb380 801469e4 b9ccb59c b9d0ddd4 b9d3dfac b9d3df80 8014251c 801469f0
[   15.032816] df80: ffffffff b9ccb540 801423e4 00000000 00000000 00000000 00000000 00000000
[   15.052490] dfa0: 00000000 b9d3dfb0 801010ac 801423f0 00000000 00000000 00000000 00000000
[   15.072096] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   15.091679] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[   15.111014] [<8018e0b8>] (rcu_process_callbacks) from [<80102410>] (__do_softirq+0x190/0x3f0)
[   15.130616] [<80102410>] (__do_softirq) from [<8012664c>] (run_ksoftirqd+0x44/0x54)
[   15.149185] [<8012664c>] (run_ksoftirqd) from [<80146b1c>] (smpboot_thread_fn+0x138/0x1d0)
[... 34 lines cut ...]
[   15.747679] [<80109a88>] (arch_cpu_idle) from [<80831eb4>] (default_idle_call+0x34/0x48)
[   15.765310] [<80831eb4>] (default_idle_call) from [<8015265c>] (do_idle+0xec/0x17c)
[   15.782510] [<8015265c>] (do_idle) from [<801529ac>] (cpu_startup_entry+0x28/0x2c)
[   15.799636] [<801529ac>] (cpu_startup_entry) from [<8010fd54>] (secondary_start_kernel+0x134/0x140)
[   15.818384] [<8010fd54>] (secondary_start_kernel) from [<0010270c>] (0x10270c)
[   15.835311] CPU2: stopping
[   15.847637] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G      D  C        4.19.57-v7+ #1244
[   15.865445] Hardware name: BCM2835
[   15.878460] [<80111f38>] (unwind_backtrace) from [<8010d4b0>] (show_stack+0x20/0x24)
[   15.896045] [<8010d4b0>] (show_stack) from [<80814ea0>] (dump_stack+0xd4/0x118)
[   15.913200] [<80814ea0>] (dump_stack) from [<80110350>] (handle_IPI+0x388/0x3a8)
[   15.930420] [<80110350>] (handle_IPI) from [<801021f4>] (bcm2836_arm_irqchip_handle_irq+0xa0/0xa4)
[   15.949258] [<801021f4>] (bcm2836_arm_irqchip_handle_irq) from [<801019bc>] (__irq_svc+0x5c/0x7c)
[   15.967977] Exception stack(0xb9d4ff38 to 0xb9d4ff80)
[   15.982797] ff20:                                                       80109a84 00000000
[   16.000859] ff40: 40000093 40000093 80d04d70 b9d4e000 80d04db8 00000004 80d8ed3e 410fd034
[   16.018940] ff60: 00000000 b9d4ff94 80d0517c b9d4ff88 00000000 80109a88 40000013 ffffffff
[   16.037026] [<801019bc>] (__irq_svc) from [<80109a88>] (arch_cpu_idle+0x34/0x4c)
[   16.054360] [<80109a88>] (arch_cpu_idle) from [<80831eb4>] (default_idle_call+0x34/0x48)
[   16.072432] [<80831eb4>] (default_idle_call) from [<8015265c>] (do_idle+0xec/0x17c)
[   16.090080] [<8015265c>] (do_idle) from [<801529ac>] (cpu_startup_entry+0x28/0x2c)
[   16.107627] [<801529ac>] (cpu_startup_entry) from [<8010fd54>] (secondary_start_kernel+0x134/0x140)
[   16.126830] [<8010fd54>] (secondary_start_kernel) from [<0010270c>] (0x10270c)
[   16.144261] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---
```

idontcarewhatmydamnusernameis, 2019-07-12:
Confirmed. It's broken. How can I revert to previous version?

maxnet, 2019-07-12:
Some more panics, which seems to be different each time.
But all happen pretty soon after boot.

```
[   14.212069] Unable to handle kernel paging request at virtual address 726f7774
[   14.231232] pgd = f02b8ead
[   14.245697] [726f7774] *pgd=00000000
[   14.261056] Internal error: Oops: 5 [#1] SMP ARM
[   14.277303] Modules linked in: bcm2835_v4l2(C+) bcm2835_codec(C+) snd_timer v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig v4l2_common snd videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common vc_sm_cma(C) videodev media uio_pdrv_genirq fixed uio i2c_dev ip_tables x_tables ipv6
[   14.329028] CPU: 0 PID: 187 Comm: setupcon Tainted: G         C        4.19.57-v7+ #1244
[   14.349746] Hardware name: BCM2835
[   14.365579] PC is at __kmalloc_track_caller+0xac/0x288
[   14.383203] LR is at __kmalloc_track_caller+0x60/0x288
[   14.400693] pc : [<80297b6c>]    lr : [<80297b20>]    psr: 20000113
[   14.419399] sp : b6e9d4e0  ip : b6e9d4e0  fp : b6e9d524
[   14.437112] r10: ffffffff  r9 : b9c01b40  r8 : 806e63fc
[   14.454771] r7 : 00490220  r6 : b9c01b40  r5 : 00000180  r4 : 726f7774
[   14.473775] r3 : 00000000  r2 : 80c97aa0  r1 : 396bd000  r0 : b6e9d4e0
[   14.492706] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   14.512303] Control: 10c5383d  Table: 36ed006a  DAC: 00000055
[   14.530456] Process setupcon (pid: 187, stack limit = 0xbd2c0117)
[   14.549055] Stack: (0xb6e9d4e0 to 0xb6e9e000)
[   14.565815] d4e0: b815d480 80d04d48 b9d16000 00001cb8 b6e9d50c b6e9d500 80126b88 b6e9d563
[   14.586555] d500: 00480020 80d04d48 806e63fc 00000180 00480020 ffffffff b6e9d54c b6e9d528
[   14.607118] d520: 806e6320 80297acc 8060763c b815d3c0 00000092 80d04d48 8060763c b9e42240
[   14.627511] d540: b6e9d594 b6e9d550 806e63fc 806e62f8 b6e9d563 b6e9d560 802018e8 00000000
[   14.647742] d560: 20000113 6950c633 b9d16558 00000092 00000001 b9d16558 00000001 b9d16500
[   14.667857] d580: b9d16564 b9d16558 b6e9d5dc b6e9d598 8060763c 806e6378 80d04d48 b9d16538
[   14.687894] d5a0: ba349298 00000000 00000006 6950c633 b6e9d5dc b9d16568 b9d1656c 00000000
[   14.707793] d5c0: ba349298 00000000 00000006 00000040 b6e9d604 b6e9d5e0 80126cf4 8060749c
[   14.727524] d5e0: 80d03098 00000006 b6e9c018 00000101 80d04d48 00000018 b6e9d614 b6e9d608
[   14.747198] d600: 80126dd4 80126c9c b6e9d67c b6e9d618 80102410 80126db8 b6e9d69c b6e9d628
[   14.766851] d620: 00000000 00400040 80d03d00 ffff905f 0000000a 80c94a00 80d8f278 80d04d70
[   14.786507] d640: 80c8c308 80d03080 b6e9d618 00000007 b815d480 60000193 b815d480 b969e400
[   14.806177] d660: 0000008a 80d04d48 00000010 00000000 b6e9d694 b6e9d680 801266d0 8010228c
[   14.825923] d680: b6e9c000 00000200 b6e9d6ac b6e9d698 801267fc 80126668 6950c633 00000000
[   14.845529] d6a0: b6e9d6e4 b6e9d6b0 8075d7f8 80126734 8db2a8c0 6950c633 0000000c b815d480
[   14.864988] d6c0: 80d04d48 b97c49c0 00000000 80d7fc40 000005dc b9d16000 b6e9d71c b6e9d6e8
[   14.884304] d6e0: 8075f4a4 8075d5d4 00000200 80d04d48 0000048a b815d480 80d04d48 0000048a
[   14.903562] d700: 00000000 80d7fc40 b97c49c0 b9d16000 b6e9d76c b6e9d720 80760114 8075f36c
[   14.922822] d720: 8075f658 8075d544 807609c4 8070e998 b9f46700 00000000 00000000 6950c633
[   14.941989] d740: 8075e61c b815d480 b97c49c0 80d7fc40 00000068 00000000 b6e9d868 00000060
[   14.961057] d760: b6e9d78c b6e9d770 8075f78c 8076004c 00003ae7 b815d480 80d7fc40 b80be434
[   14.980123] d780: b6e9d7a4 b6e9d790 80760a8c 8075f750 b815d480 b97c49c0 b6e9d7cc b6e9d7a8
[   14.999212] d7a0: 8078e638 80760a74 b97c49c0 b815d480 b6e9d868 80d04d48 b6e9d840 00000000
[   15.018292] d7c0: b6e9d914 b6e9d7d0 80791388 8078e518 00000068 00000008 b6e9d818 b6e9d814
[   15.037381] d7e0: b6e9d840 00004040 80d7fc40 8db2a8c0 00000000 00000000 00000108 8075f0c8
[   15.056488] d800: 00000000 00000000 b97c4c20 00000000 805b2158 00000000 00000000 00000000
[   15.075604] d820: 00000000 00000000 8db2a8c0 00000000 00000000 ffff0000 00000000 00000000
[   15.094710] d840: 00000000 00000000 00000000 000005dc 00000068 00000000 ffff0000 00000000
[   15.113754] d860: 00000000 00000000 00000002 00000001 00000000 00110000 00000000 00000040
[   15.132743] d880: 00000000 00000000 00000000 801c4cac e2b2a8c0 8db2a8c0 3d030108 80154f84
[   15.151744] d8a0: 00000001 00000000 000001d9 00000000 00000000 00000000 8081f0d4 ba351ec0
[   15.170359] d8c0: b9422d80 80d04d48 00000000 4f1765da 00000003 00000000 b6e9d94c b6e9d8e8
[   15.188606] d8e0: 80154f84 6950c633 b9422d80 b97c49c0 b9ab0000 00000000 6950c633 00000000
[   15.206876] d900: b6e9da38 00000000 b6e9d934 b6e9d918 8079c1c0 80790e74 b6e9d94c 6950c633
[   15.225220] d920: b6e9d97c b9ab0000 b6e9d944 b6e9d938 806ddab8 8079c184 b6e9d964 b6e9d948
[   15.243527] d940: 806ddbf4 806ddaa0 00000060 00000402 80d04d48 b97d5004 b6e9d9d4 b6e9d968
[   15.261857] d960: 807e78fc 806ddbc0 00000060 ba351e40 b6e9d9c4 b97d5004 00000060 b97b100c
[   15.280188] d980: 00000010 00000003 00000000 00000000 b6e9d97c 00000000 00000000 00000000
[   15.298259] d9a0: 00000000 00004040 00000000 6950c633 b6eb9e00 00000001 00000060 b970b804
[   15.316069] d9c0: b9ab0000 b9f46e00 b6e9da24 b6e9d9d8 807e799c 807e785c 00000000 00000000
[   15.333752] d9e0: b6e9da24 b6e9d9f0 8019b964 806a7b48 45ed5e2c 00000003 b9b89958 b970b800
[   15.351427] da00: 80d04d48 b97b1000 b970b804 b9f46e00 00000000 00000000 b6e9da64 b6e9da28
[   15.369060] da20: 807e7e54 807e7924 00000000 00000001 b6e9da38 b6e9da40 00000000 6950c633
[... cut ...]
[   17.660640] [<80831eb4>] (default_idle_call) from [<8015265c>] (do_idle+0xec/0x17c)
[   17.678727] [<8015265c>] (do_idle) from [<801529ac>] (cpu_startup_entry+0x28/0x2c)
[   17.696727] [<801529ac>] (cpu_startup_entry) from [<8010fd54>] (secondary_start_kernel+0x134/0x140)
[   17.716281] [<8010fd54>] (secondary_start_kernel) from [<0010270c>] (0x10270c)
[   17.733992] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---
```

Issue seems to be present with the 4.19.x kernels.
4.14.x works fine with same root file system.

maxnet, 2019-07-12:
>Confirmed. It's broken. How can I revert to previous version?

If using piserver and "apt-get upgrade" gave you a 4.19.x kernel that bricked your system, use the following to downgrade to 4.14.x:

- In Piserver GUI go to tab "software", select operating system, open chroot shell

```
cd /tmp
wget https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20190401-1_armhf.deb
dpkg -i r*kernel*.deb
```

pelwell, 2019-07-12:
This looks like memory corruption - something is probably writing off the end of a buffer, or continuing to write to it after our has been freed. @maxnet There are some kernel heap validation features of varying performance impact that can be enabled if you are prepared to build your own kernel.

Correction: "slub" debugging is a runtime option - try adding "slub_debug" to cmdline.txt.

maxnet, 2019-07-12:
@pelwell

slub_debug did not give anything useful.

But did notice that if for testing purposes I boot without modules (rename the /lib/modules folder), it does boot to X correctly.
So after that I have been playing a bit with blacklisting modules one by one. And the problem module seems to be bcm2835_v4l2

maxnet, 2019-07-13:
Two things I noticed.

- With the default 3 seconds timeout the MMAL PORT_PARAMETER_GET command to query camera information times out (reply does arrive after all some seconds later)

```
[   16.692332] bcm2835_mmal_vchiq: >>> sync message type:COMPONENT_CREATE(4) length:160
[   16.709582] <<h: 00000000: 6c616d6d 00000004 00000000 00000001  mmal............
[   16.726864] <<h: 00000010: 00000000 b7cb18e0                    ........
[   16.743603] <<p: 00000000: 00000000 656d6163 695f6172 006f666e  ....camera_info.
[   16.760915] <<p: 00000010: 00000000 00000000 00000000 00000000  ................
[   16.778346] <<p: 00000020: 00000000 00000000 00000000 00000000  ................
[   16.795789] <<p: 00000030: 00000000 00000000 00000000 00000000  ................
[   16.813190] <<p: 00000040: 00000000 00000000 00000000 00000000  ................
[   16.830287] <<p: 00000050: 00000000 00000000 00000000 00000000  ................
[   16.847153] <<p: 00000060: 00000000 00000000 00000000 00000000  ................
[   16.864036] <<p: 00000070: 00000000 00000000 00000000 00000000  ................
[   16.880486] <<p: 00000080: 00000000 80159be8                    ........
[   16.901818] bcm2835_mmal_vchiq: <<< reply message type:COMPONENT_CREATE(4) length:44
[   16.918532] <<h: 00000000: 6c616d6d 00000004 3f2d6f04 00000001  mmal.....o-?....
[   16.934842] <<h: 00000010: 00000000 b7cb18e0                    ........
[   16.934862] <<p: 00000000: 00000000 00000040 00000000 00000000  ....@...........
[   16.966806] <<p: 00000010: 00000000                             ....
[   16.982145] bcm2835_mmal_vchiq: >>> sync message type:PORT_INFO_GET(8) length:36
[   16.997751] <<h: 00000000: 6c616d6d 00000008 00004003 00000001  mmal.....@......
[   17.013277] <<h: 00000010: 00000000 00000004                    ........
[   17.028356] <<p: 00000000: 00000040 00000001 00000000           @...........
[   17.028510] bcm2835_mmal_vchiq: <<< reply message type:PORT_INFO_GET(8) length:316
[   17.028531] <<h: 00000000: 6c616d6d 00000008 3f2d6f04 00000001  mmal.....o-?....
[   17.028544] <<h: 00000010: 00000000 00000004                    ........
[   17.028568] <<p: 00000000: 00000000 00000040 00000001 00000000  ....@...........
[   17.028586] <<p: 00000010: 00000000 00000000 3f4edae0 3f4edb80  ..........N?..N?
[   17.028603] <<p: 00000020: 00000001 00000000 00000000 3f4edbe0  ..............N?
[   17.028629] <<p: 00000030: 00000004 00000060 00000000 00000000  ....`...........
[   17.028646] <<p: 00000040: 00000000 00000004 00000060 3f4ed860  ........`...`.N?
[   17.028663] <<p: 00000050: 3f2db12c 00000000 00000001 00000000  ,.-?............
[   17.028678] <<p: 00000060: 00000000 3f4edc00 00000000 00000000  ......N?........
[   17.028694] <<p: 00000070: 00000000 00000000 00000000 00000000  ................
[   17.028710] <<p: 00000080: 00000000 00000000 00000000 00000000  ................
[   17.028732] <<p: 00000090: 00000000 00000000 00000000 00000000  ................
[   17.028758] <<p: 000000a0: 00000000 00000000 00000000 00000000  ................
[   17.028775] <<p: 000000b0: 00000000 00000000 00000000 00000000  ................
[   17.028791] <<p: 000000c0: 00000000 00000000 00000000 00000000  ................
[   17.028808] <<p: 000000d0: 00000000 00000000 00000000 00000000  ................
[   17.028824] <<p: 000000e0: 00000000 00000000 00000000 00000000  ................
[   17.028840] <<p: 000000f0: 00000000 00000000 00000000 00000000  ................
[   17.028857] <<p: 00000100: 00000000 00000000 00000000 00000000  ................
[   17.028877] <<p: 00000110: 00000000 00000000 00000000 00000000  ................
[   17.028887] <<p: 00000120: 00000000                             ....
[   17.034900] bcm2835_mmal_vchiq: >>> sync message type:PORT_PARAMETER_GET(15) length:40
[   17.378035] <<h: 00000000: 6c616d6d 0000000f 00000000 00000002  mmal............
[   17.378054] <<h: 00000010: 00000000 00000000                    ........
[   17.405463] <<p: 00000000: 00000040 00000000 0001001c 00000098  @...............
[   20.479224] bcm2835_mmal_vchiq: timed out waiting for sync completion
[   20.492217] Failed to get camera info
[   20.503996] bcm2835_mmal_vchiq: >>> sync message type:COMPONENT_DESTROY(5) length:28
[   20.518230] <<h: 00000000: 6c616d6d 00000005 ffffe000 00000002  mmal............
[   20.532081] <<h: 00000010: 00000000 80faf030                    ....0...
[   20.545115] <<p: 00000000: 00000040                             @...
[   23.411269] bcm2835_mmal_vchiq: <<< reply message type:PORT_PARAMETER_GET(15) length:180
[   23.425794] <<h: 00000000: 6c616d6d 0000000f 3f2d6f04 00000002  mmal.....o-?....
[   23.439509] <<h: 00000010: 00000000 00000000                    ........
[   23.452390] <<p: 00000000: 00000000 0001001c 00000098 00000000  ................
[   23.465959] <<p: 00000010: 00000002 00000000 00000000 00000000  ................
[   23.479494] <<p: 00000020: 00000000 00000000 00000000 00000000  ................
[   23.492986] <<p: 00000030: 00000000 00000000 00000000 00000000  ................
[   23.506385] <<p: 00000040: 00000000 00000000 00000000 00000000  ................
[   23.519777] <<p: 00000050: 00000000 00000000 00000000 00000000  ................
[   23.533017] <<p: 00000060: 00000000 00000000 00000000 00000000  ................
[   23.546100] <<p: 00000070: 00000000 00000000 00000000 00000000  ................
[... cut ...]
```

If I raise SYNC_MSG_TIMEOUT to say 10, it does not timeout.
But in both cases the system hangs (or panics) some seconds afterwards.

- If I alter the kernel code so that the PORT_PARAMETER_GET command is not sent at all (but the other commands are), the system does not hang.
Like there is something wrong with the firmware side implementation of that command.
Does firmware also have the potential to corrupt memory or otherwise get the system to hang some seconds later if there is a bug there?

kpfleming, 2019-07-15:
I noticed a number of errors with the bcm2835_v4l2 module as well when trying to boot the 4.19.57 kernel, and had to roll back to 4.19.42 to get a working system. I didn't have kernel crashes, but I'm also using nfsboot and the lan78xx driver failed to keep the link working after the bcm2835_v4l2 module failure messages (could be unrelated of course).

6by9, 2019-07-15:
We did have a memory corruption issue with bcm2835_v4l2 and PORT_PARAMETER_GET previously. Life got a little tortuous as upstream applied a different patch to "fix" it, so it's possible it's still partially broken.
I'll have a look.

I'm bemused that we get a timeout at 3 seconds - the VPU should be doing very little at that point. It is probing for the connected sensor, but that should take less time than that.

maxnet, 2019-07-15:
>I'm bemused that we get a timeout at 3 seconds - the VPU should be doing very little at that point.

That also only happens when the Pi is being booted using a PXE network boot BTW.
If I put same kernel on SD card, even when having root file system still on NFS, the command does complete fast, and system boots properly.

```
[   17.087453] bcm2835_mmal_vchiq: >>> sync message type:PORT_PARAMETER_GET(15) length:40
[   17.087465] <<h: 00000000: 6c616d6d 0000000f 00000000 00000002  mmal............
[   17.087473] <<h: 00000010: 00000000 00000000                    ........
[   17.087483] <<p: 00000000: 00000040 00000000 0001001c 00000098  @...............
[   17.104555] bcm2835_mmal_vchiq: <<< reply message type:PORT_PARAMETER_GET(15) length:180
[   17.104586] <<h: 00000000: 6c616d6d 0000000f 3f2d6ec4 00000002  mmal.....n-?....
[   17.104604] <<h: 00000010: 00000000 00000000                    ........
[   17.104621] <<p: 00000000: 00000000 0001001c 00000098 00000000  ................
[   17.104639] <<p: 00000010: 00000002 00000000 00000000 00000000  ................
[   17.104656] <<p: 00000020: 00000000 00000000 00000000 00000000  ................
[   17.104673] <<p: 00000030: 00000000 00000000 00000000 00000000  ................
[   17.104688] <<p: 00000040: 00000000 00000000 00000000 00000000  ................
[   17.104706] <<p: 00000050: 00000000 00000000 00000000 00000000  ................
[   17.104725] <<p: 00000060: 00000000 00000000 00000000 00000000  ................
[   17.104742] <<p: 00000070: 00000000 00000000 00000000 00000000  ................
[   17.104761] <<p: 00000080: 00000000 00000000 00000000 00000000  ................
[   17.104776] <<p: 00000090: 00000000 00000001 00000001           ............
```

kpfleming, 2019-07-15:
Just a note: Pis do not support PXE, at all. They can boot off of a network (I have four that do), but they don't use PXE and it's really unfortunate that the community has adopted this terminology. PXE is a specific protocol and configuration, it's not just another name for 'network boot' :-)

maxnet, 2019-07-15:
>Just a note: Pis do not support PXE, at all. 

If a device spits out DHCP packets that say "PXEClient" I call it a PXE boot.
Regardless of the level of protocol conformity.

kpfleming, 2019-07-15:
Indeed, a decent part of the blame falls on the RPi team which decided to use PXEClient as an identifier in the DHCP packets for a non-PXE device. In my network I don't test for it, and don't include it in responses back to the devices (and I don't even check for the magic vendor identifier either) and it all works fine.

kpfleming, 2019-09-02:
I have resolved my booting problems by blacklisting the`bcm2835_codec` and `bcm2835_v4l2` modules in `/etc/modprobe.d/blacklist.conf`. I have no need for any VPU support in my applications and with these modules blacklisted my boards boot properly when using kernel DHCP/tftp-boot/nfsroot.

lurch, 2019-09-02:
@kpfleming That's kinda what https://github.com/raspberrypi/piserver/issues/75#issuecomment-516123335 also says.

gorky, 2019-09-03:
In my case, the PIs were hanging on boot w/o displaying anything in regards to a kernel panic, halt or anything else.  Blacklisting these modules resolved the issue.  PI2 & PI3, several different versions.

gorky, 2019-09-03:
I wonder if [https://github.com/raspberrypi/linux/issues/2528](https://github.com/raspberrypi/linux/issues/2528) has anything to do with this?

Flole998, 2019-09-05:
I'm also affected, thought it had something to do with the bootcode at first but apparently it doesn't.

FeepingCreature, 2019-09-07:
Huh. Blacklisting those modules fixed it for me, too. Let me throw in some Google keywords to help others: "raise network interfaces", "Network Time Synchronization" (it sometimes froze with that task displayed), Raspberry Pi 3 B nfsboot NFS boot freeze hang broken.

esticle, 2019-09-09:
Thanks google, this thread took me a while to find! Brand new Buster image on RPI3 modelB using "PXE" boot / nfsroot fails, uname -a claims '4.19.57-v7+ #1244'. Blacklisting those two modules works for me as well.

Mine got stuck in startup at "creating volatile files and directories" or when systemd starts "udev kernel device manager". I twice had a kernel panic, but this has not happened after I hooked up a serial console to try and capture the output.

Thanks @kpfleming for the temporary workaround.

rfeague, 2019-09-10:
+1 --- blacklisting bcm2835_codec and bcm2835_v4l2 fixed network boot for me.  Huge kudos to @maxnet, @kpfleming, and anyone else who contributed here.  This bug cost me a dozen hours or more, stopped dead in my tracks.  The problem manifests itself in a variety of ways, most commonly hanging in Network Time Synchronization, but sometimes "raise network interfaces" as @esticle and @FeepingCreature note.  I've attached the two files I created to blacklist these modules.  Place them in /etc/modprobe.d/ and reboot.

[blacklist.zip](https://github.com/raspberrypi/linux/files/3593660/blacklist.zip)

gorky, 2019-09-10:
I am wondering what it would take to get a contributor or someone from the @raspberrypi group to have a look at this?

linuxkidd, 2019-09-10:
Another +1 on this issue, and the same workaround ( blacklisted bcm2835_v4l2 ).
Kernel: `4.19.71-v7+ #1262`

XECDesign, 2019-10-01:
This is currently holding up piserver images. Should I apply the blacklist workaround or is a fix in the works?

6by9, 2019-10-01:
I've done a quick test on the old PiServer image I had around but using the latest rpi-update kernel and firmware, and it's booting fine, with or without camera.
Firmware is 30th Sept, 6820edeeee4ef
Kernel is 30th Sept, 4.19.75-v7+

If you have a more recent image that I can test with, then I'm happy to do so.

6by9, 2019-10-01:
Ah, I do have ```start_x=1``` in config.txt. Without that I do see boot stalling.
It'll probably be VLL loading again - I thought we'd solved that ages ago.

I'll try to get a debugger working on it and see what is going on.

6by9, 2019-10-01:
Firmware patch created that seems to work for me.
It needs to ripple through the firmware release process before it makes its way into the wild.

peter-chineham, 2020-01-05:
The depressing thing is that I am getting exactly this problem with 2 x brand new RPi4 netbooting from another RPi4 (following two official guides closely), but another two work fine. On the two which work, I got a kernel panic as described after pressing enter to continue, these two now work no problem and reboot fine. The other two just hang exactly as in the original description... The workarounds in the git error report don't work, not surprisingly, so I am now going to try booting from an SD card with just /boot on it which directs to the original nfs folder and see if that works. I suspect I have configured something wrong which the panic corrected, but I can't see anything obvious.

lurch, 2020-01-06:
@peter-chineham Netbooting on RPi4 is still in beta, so it may be more appropriate to report your problem at https://github.com/raspberrypi/rpi-eeprom/issues ?

maxnet, 2020-01-06:
>The workarounds in the git error report don't work

May be a different problem then.
Have you captured the kernel panics through serial console, and do they look different each time (which can indicate memory corruption like in this issue), or are they always the same?

You may also want to try if it also happens when you PXE boot your Pi through piserver. As that uses the same known good rootfs for all Pi.
When you are using different writable rootfs for each individual Pi, there are also other things that could go wrong.

alkisg, 2020-01-19:
While adding [Raspberry support to the new LTSP](https://github.com/ltsp/ltsp/issues/68#issuecomment-576013033), I too bumped into this issue, and it took me many days to find the appropriate bug reports and workarounds.

I believe it's the same as #2335 which resurfaced, so I initially [commented there](https://github.com/raspberrypi/linux/issues/2335#issuecomment-576011598); then I found this one.

Fortunately with the workaround everything is fine, LTSP can now netboot rpi2, rpi3b+ and rpi4 with no issues other than camera support due to the blacklisting.