raspberrypi/linux #3071: RPi: arm64: CONFIG_BRCMSTB_THERMAL missing in bcm2711_defconfig

aswild, 2019-07-13:
**Describe the bug**
The arm64 bcm2711_defconfig doesn't enable CONFIG_BRCMSTB_THERMAL like it's arm counterpart does. The RPi4 device tree includes a thermal driver with compatible string `brcm,avs-tmon-bcm2838` which is matched by the brcmstb thermal driver, rather than bcm2835.

**To reproduce**
`cat /sys/class/thermal/thermal_zone0/temp`

**Expected behaviour**
Returns a temperature value

**Actual behaviour**
Reading fails with EINVAL
`cat: /sys/class/thermal/thermal_zone0/temp: Invalid argument`

**System**
Raspberry Pi 4 with locally-compiled arm64 kernel.

Enabling CONFIG_BRCMSTB_THERMAL in my config (and disabling CONFIG_BCM2835_THERMAL since it's unneeded) gets the thermal driver working as expected.

lategoodbye, 2019-07-14:
Maybe related, you also need to make sure CONFIG_HW_RANDOM_IPROC_RNG200 is enabled, because this driver is used instead of CONFIG_HW_RANDOM_BCM2835.

aswild, 2019-07-14:
That's true, as I found out the other day. I think the rng200 driver got enabled by default when I did `make bcm2711_defconfig`. I'll double check, but probably wouldn't hurt to explicitly enable it somewhere if possible

Jc2k, 2019-07-16:
If it helps the maintainers, I want to chip in I had the same "Invalid argument" error and can confirm that enabling CONFIG_BRCMSTB_THERMAL fixes it for me too.

aswild, 2019-07-16:
@lategoodbye I just double checked: CONFIGHW_RANDOM_IPROC_RNG200 is needed for the Pi4, but its Kconfig uses `default HW_RANDOM` so it's already enabled by bcm2711_defconfig, which sets `CONFIG_HW_RANDOM=y`

Since it looks like bcm2711_defconfig is generated from `make savedefconfig`, there's no way to explicitly enable HW_RANDOM_IPROC_RNG200 there without changing its Kconfig default.

CONFIG_BRCMSTB_THERMAL doesn't have that sort of Kconfig default, so it should be in the defconfig.

lategoodbye, 2019-07-17:
Sorry about this and thanks for double checking.

closed 2019-07-17 by pelwell

aswild, 2019-07-17:
Happy to help, thanks for accepting my PR!

pelwell, 2019-07-17:
We're always happy to receive PRs, particularly for the arm64 world where we aren't actively developing.