raspberrypi/linux #3072: Undefined refererence to devm_mfd_add_devices building arm64

dflogeras, 2019-07-13:
Using a working config from raspberrypi-kernel_1.20190517-1, when I attempt to build a 64bit kernel, it fails when linking:

AR built-in.a
LD vmlinux.o
MODPOST vmlinux.o
LD vmlinux
aarch64-unknown-linux-gnu-ld: drivers/mfd/bcm2835-pm.o: in function bcm2835_pm_probe': bcm2835-pm.c:(.text+0xa4): undefined reference to devm_mfd_add_devices'
aarch64-unknown-linux-gnu-ld: bcm2835-pm.c:(.text+0x124): undefined reference to `devm_mfd_add_devices'

It looks like a dependency was missed maybe and nothing is building mfd-core.c. I haven't got anything under Multifunction Devices checked in my config.

lategoodbye, 2019-07-14:
Maybe the tree is missing this upstream patch:
https://lkml.org/lkml/2019/3/8/778

dflogeras, 2019-07-14:
I can confirm that adds the missing object and now my kernel links, thanks for that!

pelwell, 2019-07-14:
Cherry-picked.

dflogeras, 2019-07-14:
That answers the 'should I make a PR' question.  Thanks for merging!

closed 2019-07-14 by dflogeras