raspberrypi/linux #3077: espeak fails after gstreamer update to  gstreamer1.0-alsa:armhf

rbees, 2019-07-14:
pi@raspberrypi:~ $ espeak --version
eSpeak text-to-speech: 1.48.03  04.Mar.14  Data at: /usr/lib/arm-linux-gnueabihf/espeak-data

Espeak has not been updated since it was installed 12/21/18 from the raspian repos. But Gstreamer has and said update was applied 5/27/19.  At that time espeak stoped working.

pi@raspberrypi:~ $ espeak -s 65 "This is a test line."
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Expression 'paInvalidSampleRate' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2048
Expression 'PaAlsaStreamComponent_InitialConfigure( &self->playback, outParams, self->primeBuffers, hwParamsPlayback, &realSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2722
Expression 'PaAlsaStream_Configure( stream, inputParameters, outputParameters, sampleRate, framesPerBuffer, &inputLatency, &outputLatency, &hostBufferSizeMode )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2843
wave_open_sound > Pa_OpenStream : err=-9997 (Invalid sample rate)
Expression 'paInvalidSampleRate' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2048
Expression 'PaAlsaStreamComponent_InitialConfigure( &self->playback, outParams, self->primeBuffers, hwParamsPlayback, &realSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2722
.
.
.
Expression 'PaAlsaStream_Configure( stream, inputParameters, outputParameters, sampleRate, framesPerBuffer, &inputLatency, &outputLatency, &hostBufferSizeMode )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2843
pi@raspberrypi:~ $ 

pi@raspberrypi:/var/log/apt $ zgrep 'alsa' *
history.log.2.gz:Upgrade: gstreamer1.0-alsa:armhf (1.10.4-1, 1.10.4-1+deb9u1), ruby2.3:armhf (2.3.3-1+deb9u3+rpi1, 2.3.3-1+deb9u6+rpi1), lxplug-ptbatt:armhf (0.4, 0.5), raspi-config:armhf (20190219, 20190423), libgs9:armhf (9.26a~dfsg-0+deb9u1, 9.26a~dfsg-0+deb9u2), libpng-dev:armhf (1.6.28-1, 1.6.28-1+deb9u1), libjs-jquery:armhf (3.1.1-2, 3.1.1-2+deb9u1), gstreamer1.0-plugins-base:armhf (1.10.4-1, 1.10.4-1+deb9u1), lxplug-ejecter:armhf (0.3, 0.4), python-cryptography:armhf (1.7.1-3+b2, 1.7.1-3+deb9u1), python3-cryptography:armhf (1.7.1-3+b2, 1.7.1-3+deb9u1), libruby2.3:armhf (2.3.3-1+deb9u3+rpi1, 2.3.3-1+deb9u6+rpi1), libgstreamer-plugins-base1.0-0:armhf (1.10.4-1, 1.10.4-1+deb9u1), gstreamer1.0-x:armhf (1.10.4-1, 1.10.4-1+deb9u1), ghostscript:armhf (9.26a~dfsg-0+deb9u1, 9.26a~dfsg-0+deb9u2), lxplug-network:armhf (0.14, 0.15), unzip:armhf (6.0-21, 6.0-21+deb9u1), libgs9-common:armhf (9.26a~dfsg-0+deb9u1, 9.26a~dfsg-0+deb9u2), libaudiofile1:armhf (0.3.6-4, 0.3.6-4+deb9u1), libmariadbclient18:armhf (10.1.37-0+deb9u1, 10.1.38-0+deb9u1), rsync:armhf (3.1.2-1+deb9u1, 3.1.2-1+deb9u2), dns-root-data:armhf (2017072601~deb9u1, 2019031302~deb9u1), libpng-tools:armhf (1.6.28-1, 1.6.28-1+deb9u1), java-common:armhf (0.58, 0.58+deb9u1), base-files:armhf (9.9+rpi1+deb9u8, 9.9+rpi1+deb9u9), libpng16-16:armhf (1.6.28-1, 1.6.28-1+deb9u1)
term.log.2.gz:Preparing to unpack .../08-gstreamer1.0-alsa_1.10.4-1+deb9u1_armhf.deb ...
term.log.2.gz:Unpacking gstreamer1.0-alsa:armhf (1.10.4-1+deb9u1) over (1.10.4-1) ...
term.log.2.gz:Setting up gstreamer1.0-alsa:armhf (1.10.4-1+deb9u1) ...

cpoisson, 2019-07-18:
Hi @rbees ,

Looks like an issue related to snipsco/snips-issues#148

We found out on our side that since the upgrade from 4.14 to kernel 4.19, portaudio is not able anymore to play a wav sampled at 22050 when the playback interface is set to a rate of 16000 Hz.

espeak use portaudio and generate an audio file sampled at 22050 Hz

rpipa, 2020-03-21:
I have a similar problem which depends on the kernel version I use.

Starting with kernel version 4.19.37, the following error message appears with the latest version of PortAudio:

`Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924`

1. why does the access work until kernel 4.19.36 and not 4.19.37?

2. how can this bug be fixed?

linhz0hz, 2020-07-16:
Any update on this issue or workaround? I am hitting the same issue today. If I use alsa to play audio directly (aplay and pyalsaaudio) it works fine. But anything that uses PortAudio gave very similar error.
I hit the issue using pyaudio and sounddevice in python. But to makesure the issue is related to PortAudio, I followed these instructions [https://app.assembla.com/wiki/show/portaudio/Platforms_RaspberryPi](url) and run bin/paex_sine. This is the error message I get:
` 
portaudio/portaudio/bin/paex_sine
PortAudio Test: output sine wave. SR = 44100, BufSize = 64
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 923
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 923
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 923
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5036:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5036:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 923
Expression 'paInvalidSampleRate' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2043
Expression 'PaAlsaStreamComponent_InitialConfigure( &self->playback, outParams, self->primeBuffers, hwParamsPlayback, &realSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2717
Expression 'PaAlsaStream_Configure( stream, inputParameters, outputParameters, sampleRate, framesPerBuffer, &inputLatency, &outputLatency, &hostBufferSizeMode )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2838
An error occured while using the portaudio stream
Error number: -9997
Error message: Invalid sample rate
`