raspberrypi/linux #3078: Booting after first update,There is no route in ethernet.

yutayu, 2019-07-15:
**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
Add a clear and concise description of what you think the bug is.

There is no route when I boot after I installed buster to usbhdd.
After I need to add route and network is fine. 

**To reproduce**
List the steps required to reproduce the issue.

dd image of 7/10 to usbhdd depending on manual. 
boot with that usbhdd.
I can use internet automatically,and apt update.
reboot
I cannot use ethernet and ip route results blank.
 
**Expected behaviour**
Add a clear and concise description of what you expected to happen.

I want to use internet soon.

**Actual behaviour**
Add a clear and concise description of what actually happened.

no network there.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

pi@raspberrypi:~ $ sudo apt install raspinfo
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package raspinfo
pi@raspberrypi:~ $ sudo apt search raspinfo
Sorting... Done
Full Text Search... Done

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
pi3B
* Which OS and version (`cat /etc/rpi-issue`)?
pi@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage4

* Which firmware version (`vcgencmd version`)?
pi@raspberrypi:~ $ vcgencmd version
Jul  9 2019 14:40:53 
Copyright (c) 2012 Broadcom
version 6c3fe3f096a93de3b34252ad98cdccadeb534be2 (clean) (release) (start)

* Which kernel version (`uname -a`)?
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.57-v7+ #1244 SMP Thu Jul 4 18:45:25 BST 2019 armv7l GNU/Linux

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

so long lol

**Additional context**
Add any other relevant context for the problem.

kpfleming, 2019-07-15:
Same thing here... 20190709 (4.19.57) kernel has broken LAN (on Pi 3B and 3B+ boards), rolling back to 4.19.42 cures the problem. With .57 kernel I saw lots of messages from the lan78xx module about failures to read data.

lategoodbye, 2019-07-16:
@kpfleming Are you able to provide some dmesg output?

kpfleming, 2019-07-16:
Not directly, as the system fails to boot all the way to userspace. I can capture serial console output, but unfortunately not in the next week as I'll be away from the location of the RPis.

SudhamaniB, 2019-09-17:
Hi,

This issue is not there with kernel version 4.14.98, if you are still not facing this issue, can this issue be closed.

Regards, 
Sudhamani.

yutayu, 2019-09-17:
I agree.

closed 2019-09-17 by yutayu