raspberrypi/linux #3079: Pi 4: uhubctl doesn't power off downstream USB ports

P33M, 2019-07-15:
Vbus still remains powered even if uhubctl is called to turn off *all* USB ports.

mvp, 2019-07-19:
Just curious, can you please post output of `sudo lsusb -v` and output of `sudo uhubctl` on pi 4?

I suspect that USB3 and dual USB2 hubs might have different PPPS properties.

P33M, 2019-07-20:
```
pi@raspberrypi:~$ sudo uhubctl
Current status for hub 2 [1d6b:0003 Linux 4.19.46-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1 [1d6b:0002 Linux 4.19.46-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports]
  Port 1: 0507 power highspeed suspend enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
```
```

pi@raspberrypi:~$ sudo lsusb -v

Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0002 2.0 root hub
  bcdDevice            4.19
  iManufacturer           3 Linux 4.19.46-v7l+ dwc_otg_hcd
  iProduct                2 DWC OTG Controller
  iSerial                 1 fe980000.usb
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0019
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0008
    Ganged power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood        1 * 2 milli seconds
[... 236 lines cut ...]
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x02
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0507 highspeed power suspend enable connect
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

mvp, 2019-07-20:
Per USB 3.0 spec, all USB3 hubs must advertise both USB3 hub and USB2 compatibility hub.
Also, both must have BOS descriptor with the same ContainerID - it allows to identify that these hubs are part of the same physical device.
However, in RPi4 they are advertised in weird way:
```
4-port:
USB 3:
  bcdUSB               3.00
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  iManufacturer           3 Linux 4.19.46-v7l+ xhci-hcd
  iProduct                2 xHCI Host Controller
  nNbrPorts             4
    Per-port power switching
    Per-port overcurrent protection
    ContainerID             N/A

USB 2:
  bcdUSB               2.10
  idVendor           0x2109 VIA Labs, Inc.
  idProduct          0x3431 Hub
  nNbrPorts             4
    Ganged power switching
    Ganged overcurrent protection
    ContainerID             {30eef35c-07d5-2549-b001-802d79434c30}
```
Basically, we have 4-port USB3 and USB2 hubs, which are obviously part of the same physical 4-port hub. However, they advertise different power switching characteristics: USB3 claims per-port power switching, USB2 claims ganged power switching (and is filtered out by uhubctl). In addition to that, USB3 device does not define ContainerID at all, even though by USB 3.0 spec it is required.
Note that to turn power off, both USB3 and USB2 ports must be turned off. Since uhubctl filters out USB2 hub, power is only turned on USB3, and USB2 remains powered.
Question to Raspberry Pi 4 engineers: is ganged power switching on this hub is really true?
```
  bcdUSB               2.10
  idVendor           0x2109 VIA Labs, Inc.
  idProduct          0x3431 Hub
  nNbrPorts             4
    Ganged power switching
    Ganged overcurrent protection
```
If not, we might be able to hack uhubctl to add an exception. However, lack of ContainerID advertisement by USB3 hub still stands, and has to be hacked as well.

There are 2 more 1-port USB hubs, but I don't think they are important for this discussion, as one is OTG controller:
```
1-port:
  bcdUSB               2.00
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0002 2.0 root hub
  nNbrPorts             1
    Per-port power switching
    Per-port overcurrent protection

1-port:
  bcdUSB               2.00
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0002 2.0 root hub
 iManufacturer           3 Linux 4.19.46-v7l+ dwc_otg_hcd
  iProduct                2 DWC OTG Controller
  nNbrPorts             1
  wHubCharacteristic 0x0008
    Ganged power switching
    Per-port overcurrent protection
```

P33M, 2019-07-20:
Thanks for responding.

On the Pi 4, in common with previous board designs, the first externally-accessible power switching/vbus control pin is connected to the USB power switch that controls all downstream ports. On the SMSC devices, there was no way to specify ganged power switching without the use of an EEPROM so the hub reported per-port switching but port 2 switched off power to all of the downstream ports. I think the LAN7515 behaves in the same way.

The broken hub descriptors should be fixed - we have a relationship with VIA labs as they are supplying chips directly to us, so we can feed back any non-compliant behaviours that we come across. The hub descriptor can probably be fixed with a firmware update, but for the sake of testing the hardware - what hacks do I need to do to uhubctl to stop it skipping over the USB2.0 hub?

mvp, 2019-07-21:
You can hack this condition
https://github.com/mvp/uhubctl/blob/master/uhubctl.c#L417-L422

to allow `lpsm == HUB_CHAR_COMMON_LPSM` (ganged power switching) in addition to `HUB_CHAR_INDV_PORT_LPSM` (per-port power switching).
This will force `uhubctl` to consider ganged hubs as if they had per-port power switching.

However, since `ContainerID` is missing for USB3 hub, `uhubctl` will consider 4-port USB3 and USB2 hubs as independent. You will have to power off ports on both hubs to check if VBUS really goes off.

Ideally, we need 2 action items:
1. Fix both USB2+USB3 hub descriptor to claim what is really supported and be consistent.
2. Fix USB3 hub to report BOS `ContainerID` identical to USB2 hub.

Minor note: one of 1-port hubs above reports ganged power switching (OTG controller). Since it is 1-port anyway, perhaps you could change it to per-port? At 1-port capacity, it shouldn't really matter anyway.

P33M, 2019-07-22:
If I nobble the readback of the BOS descriptor, and permit power switching on ganged hubs, then setting port power to off on both the 4-port USB2.0 hub device and the USB3.0 root hub removes downstream power.

It looks like the broken descriptor is the only thing that needs fixing.

mvp, 2019-07-23:
Thanks for the update! Please shout out when these descriptor changes are out, and if you need me to fix anything in uhubctl.

OutsourcedGuru, 2019-07-29:
(Just subscribing to this thread.)

mvp, 2019-10-06:
Any update on fixing broken descriptors in 4B firmware? Can I help somehow?

P33M, 2019-10-07:
I do have a question - how do you do port power control of USB3.0 root hub ports? Linux provides the descriptor in this case but without a BOS/containerID.

mvp, 2019-10-07:
USB3 hubs advertise themselves as 2 virtual hubs: one is USB 3.0, and another is compatibility USB 2.1, but they both share the same physical port. (USB 2.1 to signify that hub is just a bit smarter than old USB 2.0.).
Without special treatment, `uhubctl` sees those hubs as independent entities. However, if you were to turn off power on say USB3 only, hub chip will still keep powered until you turn off power on USB2 counterpart as well.

To address this problem, uhubctl tries to guess correct USB2+USB3 hub pairs, and turn off power on both automatically.
USB3 spec requires that USB 2.1/USB3.x hubs report BOS containerID, which must be the same for all hubs that are part of the same physical device - this is the most reliable way to detect dual USB2/USB3 pair.

Until recently, `uhubctl` was using different method to guess that - it was looking at other clues. E.g. if we have one USB 3 hub, and one USB 2.1, and they seem to have similar topology (location is the same except for bus number), have the same vendor:product, and same number of ports, we would conclude that they must be the dual pair. However, this method was notoriously unreliable, especially if you have multiple USB3 hubs connected, it is virtually impossible to tell which is which without using containerID.
This method wouldn't work for RPI 4B anyway, as USB2 vs USB3 parts have completely different topology attachment, different vendor:product, and event different descriptors (per-port vs ganged).

If Linux does not provide containerID for root USB3 hubs, we have few options:

1. Fix Linux kernel USB driver to expose containerID for root USB3 hubs.
2. Hack uhubctl to have special case for RPI 4B to treat its root USB3 hub as having containerID `{30eef35c-07d5-2549-b001-802d79434c30}` (to match USB 2.1 part).
3. Ask users to turn off power for USB 2.1 and USB 3.0 hubs using 2 separate `uhubctl` commands.

However, all of these methods still require all relevant USB descriptors to report per-port power switching.

henzef, 2019-11-08:
any update on this? It would be really helpful to have working uhubctl on my 4B

P33M, 2019-11-08:
The issue is being investigated by VLI. We don't have a timescale for a fix.

giga, 2019-12-03:
I am also interested in this. As a temporary solution, hacking uhubctl as described above allows per-port power switching? Or does it only allow switching all ports at once?

mvp, 2019-12-03:
I don't know - I don't have Raspberry Pi 4B and don't have access to one. It really depends on what 4B hardware is actually supporting - per port or ganged power switching. Also, it is possible that updated USB firmware can change power switching behavior.

giga, 2019-12-04:
Got a Pi 4B today and I could try it. I patched uhubctl as described above. With a basic device (mouse with led) it works as expected.

I also have an external case for hard drive (with its own power supply) that has an option to go to sleep when unplugged from usb. My plan was to use usb per-port power switch to control it from command-line. The thing is, it only goes to sleep if I turn off all ports, both USB2 and 3. As soon as I turn on power on a port, USB2 or 3, it wakes up. I'm not sure if it specific to raspberry pi as I do not have another hardware with per-port power switching.

mhgue, 2019-12-16:
Using Raspbian GNU/Linux 10 (buster) on Pi4 with 4.19.75-v7l+ and replacing
`if ((lpsm == HUB_CHAR_INDV_PORT_LPSM) &&`
by
`if ((lpsm == HUB_CHAR_INDV_PORT_LPSM || lpsm == HUB_CHAR_COMMON_LPSM) &&`
as described above, lets `hub 3` (OTG) and `hub 1-1` (4 port USB2) appear in addition to `hub 1` (xHCI Host) and `hub 2` (4 port USB3). Using `uhubctl -a off -l 1-1 -p 1` is turning off the power. No need to turn off the corresponding port of the USB3 side too. Trying to do so by `uhubctl -a off -l 2 -p 1` fails with `No compatible smart hubs detected at location 2!`. All others are detected as compatible.

We are using a Pi4 to write FW to several CM3. Using this method is a pretty simple way to turn off the IO board during replacement of the CM3. First I planed to use a GPIO with an external switch, but that way is perfectly cheep and simple. Many thanks to mvp.

P33M, 2019-12-16:
We've received updated firmware from VLI: https://www.raspberrypi.org/forums/viewtopic.php?t=258489

A patch to uhubctl is still required but in a slightly different form to the one posted by @mhgue.

mvp, 2019-12-16:
@P33M , please post output from `sudo lsusb -v` and perhaps also `sudo uhubctl` (with your patch) after you updated USB firmware.

P33M, 2019-12-16:
```
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0002 2.0 root hub
  bcdDevice            4.19
  iManufacturer           3 Linux 4.19.75-v7l+ dwc_otg_hcd
  iProduct                2 DWC OTG Controller
  iSerial                 1 fe980000.usb
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0019
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0008
    Ganged power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood        1 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0000
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Device Descriptor:
  bLength                18
[... 240 lines cut ...]
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0507 highspeed power suspend enable connect
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```
Patched output:
```
Current status for hub 3 [1d6b:0002 Linux 4.19.75-v7l+ dwc_otg_hcd DWC OTG Controller fe980000.usb]
  Port 1: 0000 off
Current status for hub 2 [1d6b:0003 Linux 4.19.75-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1 [1d6b:0002 Linux 4.19.75-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0]
  Port 1: 0507 power highspeed suspend enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
```

mvp, 2019-12-29:
Sorry, it took so long to respond - I was traveling.

I don't see any difference in new firmware behavior.
It has not fixed any of the problems I stated before, namely:

1. USB3 hub still does not report ContainerID - makes impossible to match dual device.
2. USB2 and USB3 hubs still report conflicting type of power switching: it should be consistently ganged or per-port, but they are reported differently, USB2 ganged, USB3 per-port.

All that was done here is to hack uhubctl to treat ganged hubs as per-port.
I believe that it would work exactly the same for old vs new firmware.
If you paid Via for this work, I would ask for refund.

mhgue, 2019-12-30:
The patches are the same, except for line breaks and comments. The difference of the new VL805 firmware release 0137ac is no more `No compatible smart hubs detected at location 2!`.

mvp, 2019-12-31:
@P33M, @mhgue:
I have posted experimental rpi4 branch for uhubctl to support rpi4: https://github.com/mvp/uhubctl/issues/204#issuecomment-569847497
It is ugly to my liking, but should work. Please try it and report if it works as expected.

mvp, 2020-01-03:
Bump: I need someone to try rpi4 fix. I don't have Raspberry Pi 4 (unless you want to send me one 😄 ).
Steps:

1. Flash this USB firmware to your RPi4: https://www.raspberrypi.org/forums/download/file.php?id=33477&sid=d08e808232df0e2ac10eb1af53932f36
2. Build uhubctl with the fix:
```
git clone https://github.com/mvp/uhubctl -b rpi4
cd uhubctl
make
```
3. Check if power switching is working:
```
# off:
sudo ./uhubctl -a 0 -l 2
# on:
sudo ./uhubctl -a 1 -l 2
```

ntova, 2020-01-04:
Unfortunately this does not seem to work. I get a

```
No compatible smart hubs detected at location 2!
Run with -h to get usage info.
```

as result. Firmware version is `000137ac`, branch is `rpi4`

mvp, 2020-01-04:
@ntova, thanks for trying. Can you show output for `sudo uhubctl` and `sudo lsusb -v`?

ntova, 2020-01-04:
```
Current status for hub 2 [1d6b:0003 Linux 4.19.88-1-ARCH xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 1: 0203 power 5gbps U0 enable connect [152d:2509 JMicron Manu Usb production 00A1234567E7]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1 [1d6b:0002 Linux 4.19.88-1-ARCH xhci-hcd xHCI Host Controller 0000:01:00.0]
  Port 1: 0507 power highspeed suspend enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
```

`lsusb -v`: https://gist.github.com/ntova/920fbe512de4f66cb6452c9cf1fb89b7

mvp, 2020-01-04:
@ntova, thanks!

This is looking good to me.
At least it proves we can detect both USB 2 and USB 3 hubs.

As a sanity check, `sudo uhubctl -l 2` should output exactly this:
```
Current status for hub 2 [1d6b:0003 Linux 4.19.88-1-ARCH xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 1: 0203 power 5gbps U0 enable connect [152d:2509 JMicron Manu Usb production 00A1234567E7]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
```

At this point, `sudo uhubctl -a 0 -l 2` should work to turn power on both USB 2 and USB 3 hubs.

Can you also try turning power off by using 2 commands in a row:
```
sudo uhubctl -e -a 0 -l 2
sudo uhubctl -e -a 0 -l 1-1
```
(-e disables automatic USB 2/3 duality handling).

Thanks!

ntova, 2020-01-04:
> As a sanity check, `sudo uhubctl -l 2` should output exactly this:

Sounds logical to me, but unfortunately it outputs this:
```
No compatible smart hubs detected at location 2!
Run with -h to get usage info.
```

By the way, I'm willing to chip in a few dollars to get you a Raspberry Pi 4. I can't buy you one on my own, as I'm a student and a bit short on money but maybe there are a few others and we can get the money together :)

mvp, 2020-01-04:
Thanks, @ntova. I am completely puzzled as to why `sudo uhubctl` would detect hubs `2` and `1-1`, yet `sudo uhubctl -l 2` would complain that hub is not detected - it just doesn't seem to make any sense.

Thanks for offering to chip in, but I don't really need to own rpi4. That said, having access to live rpi4 would be super helpful to understand what's going on. Maybe someone can lend me one for few days?

justinasjaronis, 2020-01-04:
Dear MVP, I am also more than  interested in solution for this issue. If You'd send your pubkey I can grant you access to Raspberry Pi 4 or I can create temporary password right now.

justinasjaronis, 2020-01-04:
Could You write an email to justinas.jaronis@gmail.com,  I would pass access to rpi4 machine. BR

mvp, 2020-01-05:
Thanks to @justinasjaronis providing ssh access to RPi4, I think I have a fix.
Can someone please try building uhubctl from https://github.com/mvp/uhubctl/tree/rpi4, and try it per instructions above https://github.com/raspberrypi/linux/issues/3079#issuecomment-570702895
Thanks!

justinasjaronis, 2020-01-05:
Dear MVP, thanks a lot, on the device You were working, it actually turns off USB port. Should I try on other devices too?

mvp, 2020-01-05:
@justinasjaronis, yes please. Note that on DWC OTG 1 port hubs it may not do anything. Using hub 2 or 1-1 should have identical effect.

ntova, 2020-01-05:
> ```
> # off:
> sudo ./uhubctl -a 0 -l 2
> # on:
> sudo ./uhubctl -a 1 -l 2
> ```

Thank you very much, works for me.

Now we just need to hope for per-port switching...

mvp, 2020-01-05:
Thanks, @ntova. Per-port might work, but I don't think it's likely. Basically this is limited by what hardware actually supports.
I will merge this to master branch.

ntova, 2020-01-05:
In the current state, powering off for example port 4 leaves the power on although uhubctl reports it as powered off. Powering off everything (the command used above) actually works.

mvp, 2020-01-05:
Thanks. This behavior is consistent with ganged power switching - only turning all ports would cut off VBUS.
I will document this in readme.

justinasjaronis, 2020-01-05:
@mvp yes it is, I can confirm that  1-1 and 2 works identically. 

Also I have USB lamp plugged in 4 port, so ./uhubctl -l 2 -p 4 -a on / off affects only this port.

justinasjaronis, 2020-01-05:
Sorry for misinformation. It was a coincidence. Yeah port4 turns off everything, and other ports do nothing

mvp, 2020-01-05:
Oh wait, port 4 controlling all other ports? @justinasjaronis

mvp, 2020-01-05:
I have merged this fix to uhubctl master branch https://github.com/mvp/uhubctl/commit/4aae44ced03980802c5fdadcea16d1bc2f8c0bdb. However, it would be great to know actual requirements for VBUS off - is it really all ports, or just port 4 is enough.

justinasjaronis, 2020-01-05:
Yes exactly. I have testested both USB3 and USB2 ports, all ports get turned on/off by either 1-1 or 2 hub port 4.

justinasjaronis, 2020-01-05:
Other ports power state does not matter anything on any port VBUS

mvp, 2020-01-05:
Thanks, @justinasjaronis. I will update the readme.

justinasjaronis, 2020-01-05:
Anyway, for my business case this is more than enough. Thank you once again ! Will you need the device anymore, or can I turn it off?

mvp, 2020-01-05:
Much thanks for providing access to RPi4, @justinasjaronis!
I don't need your device anymore.

mvp, 2020-01-05:
For the record, I expect this to work with old or new USB firmware exactly the same @P33M. You should still ask Via to fix USB descriptors

m4xr1sk, 2020-02-06:
I have updated the fw to the last version but it doesn't work. Do you have any suggestion?

mvp, 2020-02-06:
1. Last version, which is what? You need vl805 firmware at 00137ac or later.

2. Try turning off all 4 ports using

    sudo uhubctl -l 2 -a 0

(please post output from running this command!)

3. If that doesn't help, try turning power off on USB 2 and USB 3 hubs separately:

    sudo uhubctl -e -l 2 -a 0
    sudo uhubctl -e -l 1-1 -a 0

rob2212, 2020-02-07:
It seems not to work for me.

I have install the USB firmware form here: [https://www.raspberrypi.org/forums/viewtopic.php?t=258489](url)

I have the latest version of uhubctl installed and I use one of the compatible D-Link USB Hubs.

My output for sudo ./uhubctl/uhubctl looks like:
```
Current status for hub 2 [1d6b:0003 Linux 4.19.97-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0103 power enable connect [1cf1:0030 dresden elektronik ingenieurtechnik GmbH ConBee II DE2139920]
  Port 2: 0503 power highspeed enable connect [2101:8500 Action Star USB2.0 Hub]
  Port 3: 0103 power enable connect [046d:c52b Logitech USB Receiver]
  Port 4: 0100 power
Current status for hub 1 [1d6b:0002 Linux 4.19.97-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0]
  Port 1: 0503 power highspeed enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
```
hub1 seems to be the compatible USB hub. I try to power it down with: 
sudo ./uhubctl/uhubctl -l 1 -a 0

```
Current status for hub 2 [1d6b:0003 Linux 4.19.97-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1 [1d6b:0002 Linux 4.19.97-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0]
  Port 1: 0000 off
```
But my device attached to hub still gets power and (all my devices seem to have stopped?)

What is my mistake? (Sorry if not completely clear, I'm new to the RP stuff)

mvp, 2020-02-07:
You need to specify hub 1-1 or 2. Hub 1 is just intermediate 1 port hub, that won't work.

rob2212, 2020-02-08:
Hi mvp, thank you for your response. If I use sudo ./uhubctl/uhubctl -l 1-1 -a 0, it powers down all USB devices.

I thought, there was a way to power down an external USH hub only, while keeping all USB devices directly plugged into the raspi active.

mvp, 2020-02-08:
@rob2212 , you can connect any hub from uhubctl compatibility list to Raspberry Pi and control its ports separately, just be sure to target that hub with -l option, this is covered in uhubctl FAQ.

For the future uhubctl questions, perhaps consider asking them in uhubctl repo. issues.

m4xr1sk, 2020-02-09:
> 1. Last version, which is what? You need vl805 firmware at 00137ac or later.
> 2. Try turning off all 4 ports using
>    sudo uhubctl -l 2 -a 0
> 
> (please post output from running this command!)
> 
> 1. If that doesn't help, try turning power off on USB 2 and USB 3 hubs separately:
>    sudo uhubctl -e -l 2 -a 0
>    sudo uhubctl -e -l 1-1 -a 0

It's working, my mistake.
The problem I suppose is in my camera: INTEL T265
You can find some picture and some details at this link:
https://github.com/IntelRealSense/librealsense/issues/5527#issuecomment-583598029
I don't know if the camera or the rpi doesn't re-enumerate after the power on, maybe is there a way to reset the usb devices detected?

nmeaux, 2020-03-05:
Hello,

On my side i have a pi4b, with vl805 firmware to 000137ad.
When i use :
```
uhubctl -l 2 -a 0 -p 1
uhubctl -l 2 -a 0
uhubctl -a 0
```
This is not working, usb port is just powered off for ~ 1sec.
But uhubctl always show the power at off.

Command :
```
# uhubctl -l 2 -a 0 -p 1
Current status for hub 2 [1d6b:0003 Linux 4.19.97-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 1: 0080 off
Sent power off request
New status for hub 2 [1d6b:0003 Linux 4.19.97-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 1: 0080 off
Current status for hub 1 [1d6b:0002 Linux 4.19.97-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports]
  Port 1: 0507 power highspeed suspend enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
Sent power off request
New status for hub 1 [1d6b:0002 Linux 4.19.97-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports]
  Port 1: 0000 off
```
Syslog :
```
Mar  5 17:18:20 raspberrypi kernel: [ 1973.017867] usb 1-1: USB disconnect, device number 58
Mar  5 17:18:20 raspberrypi kernel: [ 1973.366872] usb 1-1: new high-speed USB device number 59 using xhci_hcd
Mar  5 17:18:20 raspberrypi kernel: [ 1973.549453] usb 1-1: New USB device found, idVendor=2109, idProduct=3431, bcdDevice= 4.21
Mar  5 17:18:20 raspberrypi kernel: [ 1973.549469] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
Mar  5 17:18:20 raspberrypi kernel: [ 1973.549482] usb 1-1: Product: USB2.0 Hub
Mar  5 17:18:20 raspberrypi kernel: [ 1973.551106] hub 1-1:1.0: USB hub found
Mar  5 17:18:20 raspberrypi kernel: [ 1973.551406] hub 1-1:1.0: 4 ports detected

```

Any clue about this ?

rgds

Nicolas

mvp, 2020-03-05:
@nmeaux , please read FAQ at https://github.com/mvp/uhubctl/blob/master/README.md.

Basically, you should use

    sudo uhubctl -l 2 -p 4 -a 0 -r N

where N is a number 100 or higher.
Consider reporting such issues in uhubctl repository.

P33M, 2020-03-16:
This is tested and working on Pi 4 - closing.

closed 2020-03-16 by P33M

kindoshi, 2020-04-24:
> > ```
> > # off:
> > sudo ./uhubctl -a 0 -l 2
> > # on:
> > sudo ./uhubctl -a 1 -l 2
> > ```
> 
> Thank you very much, works for me.
> 
> Now we just need to hope for per-port switching...

I think there is a bit of confusion from 

> 1. Last version, which is what? You need vl805 firmware at 00137ac or later.
>
>     2. Try turning off all 4 ports using
>        sudo uhubctl -l 2 -a 0
>
>
> (please post output from running this [...]

I think there is a bit of confusion regarding the PI4B with update 00137ac or later.
This works, but on the PI 4B either all usb ports are on and off as stated by @ntova :
`# off:
sudo ./uhubctl -a 0 -l 2
# on:
sudo ./uhubctl -a 1 -l 2`

The latest README.md should be updated at the "Raspberry Pi 4B" section with the working commands.

Thanks for this nice tool @mvp !

[3 more replies]