raspberrypi/linux #3081: rpi3b+ hdmi no signal since last update to kernel 4.19.57

baltaner, 2019-07-16:
Hi,
since update last friday to kernel 4.19.57 with apt update; apt upgrade no hdmi device is detected
There are 4 identical pi's connected via hdmi-cabel to 4 Samsung TV (UE50J). Every one of them shows the same problem - no signal on hdmi
The solution for me was to downgrade the kernel to 4.19.50 ( 4.19.50-v7+ #1234)
with rpi-update 458871abedfd68ea417d24577694dbe9d7d4db4c
After that every thing is gonig fine

popcornmix, 2019-07-16:
Is that the newest rpi-update version that works? Identifying the exact version that caused the problem would help narrow the problem down.

baltaner, 2019-07-16:
No - the rpi-update that works is [https://github.com/Hexxeh/rpi-firmware/commit/458871abedfd68ea417d24577694dbe9d7d4db4c]
With every later firmware the problem occours

baltaner, 2019-07-16:
sorry for closing...

popcornmix, 2019-07-16:
Do you have any custom settings in config.txt or cmdline.txt?
Can you ssh in when there is no display?  What does ``tvservice -s`` report? And what does it report when working?

baltaner, 2019-07-16:
now testet with rp-update to the latest firmware:
root@pi420:/home/pi# uname -a
Linux pi420.ssdlbska.local 4.19.58-v7+ #1245 SMP Fri Jul 12 17:25:51 BST 2019 armv7l GNU/Linux
root@pi420:/home/pi# tvservice -s
state 0x40000 [NTSC 4:3], 720x480 @ 60.00Hz, interlaced
With no signal on display...

With older firmware:
root@pi420:/home/pi# uname -a
Linux pi420.ssdlbska.local 4.19.50-v7+ #1234 SMP Thu Jun 13 11:06:37 BST 2019 armv7l GNU/Linux
root@pi420:/home/pi# tvservice -s
state 0x12000a [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive
everything fine...

no special entries in config.txt - I played a little with new port-syntax...
hdmi_force_hotplug=1
hdmi_mode:1=16
hdmi_group:1=1
hdmi_mode:0=16
hdmi_group:0=1
hdmi_group=1
hdmi_mode=16
hdmi_drive=2
config_hdmi_boost=11

popcornmix, 2019-07-16:
Looks like you are getting composite output. What happens if you remove your hdmi settings from config.txt? I'm not sure forcing resolutions on second hdmi display makes sense on a Pi3.

baltaner, 2019-07-16:
now I tested it with no entries in config.txt - same output...

popcornmix, 2019-07-16:
Does it work with no config.txt entries on older version?

baltaner, 2019-07-16:
yes, it works - a little bit ugly because the resolution...
root@pi420:/home/pi# tvservice -s
state 0x12000a [HDMI DMT (4) RGB full 4:3], 640x480 @ 60.00Hz, progressive

popcornmix, 2019-07-16:
I'm going to guess you have a broken [edid](https://en.wikipedia.org/wiki/Extended_Display_Identification_Data). (This can be caused by a bad hdmi cable, or fault with display)
Can you run ``tvservice -d edid.dat`` and if it produces a non zero file attach it here.

(using working firmware with no config.txt settings).

baltaner, 2019-07-16:
I have 4 identical pi's and displays - everything worked fine until last friday...
I changed one hdmi-cable - no result.

both with the new firmware as well as with the old one:
root@pi420:/home/pi# tvservice -d edid-samsung.txt
Nothing written!

I forgot to mention that there is a picture and signal on a 22" LG monitor with DVI-Input with the new firmware...

popcornmix, 2019-07-16:
> I have 4 identical pi's and displays - everything worked fine until last friday...

not quite. You were getting an unsuitable mode by default that you had to workaround with config.txt settings. That was caused by a problem with reading the edid from your displays.

It seems the behaviour has now changed with the added Pi4 support (mainly in supporting more than one hdmi display). I'm sure it's something that can be fixed, but it's not an issue we see here on any displays (which have working edids) so it may take some time to work out what's wrong.

Can you add start_debug=1 to config.txt, reboot and capture output of:
``sudo vcdbg log msg``
with both the working and non working case (with nothing else in config.txt)

baltaner, 2019-07-16:
ok, I understand! Thanks for your help!
Here is the output:
First one is the actual working one (4.19.50):
root@pi420:/home/pi# vcdbg log msg
013810.285: brfs: File read: /mfs/sd/config.txt
013810.591: brfs: File read: 15 bytes
014242.115: HDMI:EDID error reading EDID block 0 attempt 0
014243.421: HDMI:EDID error reading EDID block 0 attempt 1
014244.722: HDMI:EDID error reading EDID block 0 attempt 2
014246.020: HDMI:EDID error reading EDID block 0 attempt 3
014247.319: HDMI:EDID error reading EDID block 0 attempt 4
014248.618: HDMI:EDID error reading EDID block 0 attempt 5
014249.918: HDMI:EDID error reading EDID block 0 attempt 6
014251.216: HDMI:EDID error reading EDID block 0 attempt 7
014252.516: HDMI:EDID error reading EDID block 0 attempt 8
014253.814: HDMI:EDID error reading EDID block 0 attempt 9
014254.877: HDMI:EDID giving up on reading EDID block 0
015122.338: brfs: File read: /mfs/sd/config.txt
015122.512: HDMI:Setting property pixel encoding to Default
015122.572: HDMI:Setting property pixel clock type to PAL
015122.633: HDMI:Setting property content type flag to No data
015122.694: HDMI:Setting property fuzzy format match to enabled
015122.843: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
015320.698: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
015321.995: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
015322.355: hdmi: HDMI:>>>>>>>>>>>>>Rx sensed, reading EDID<<<<<<<<<<<<<
015322.711: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
015324.026: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
015325.333: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
015326.641: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
015327.949: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
015329.257: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
015330.565: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
015331.873: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
015333.181: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
015334.489: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
015335.560: hdmi: HDMI:EDID giving up on reading EDID block 0
015335.638: hdmi: HDMI: No lookup table for resolution group 0
015335.695: hdmi: HDMI: hotplug attached with DVI support
015335.770: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
015336.117: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
015337.422: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
015338.726: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
015340.030: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
015341.335: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
015342.639: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
015343.943: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
015345.248: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
015346.552: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
015347.856: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
015348.925: hdmi: HDMI:EDID giving up on reading EDID block 0
015351.401: hdmi: HDMI: power_on to DMT mode VGA@60
015354.585: hdmi: HDMI: Action callback added to queue to happen at frame 2
015354.650: hdmi: HDMI: Action stop_3d_mode added to queue to happen at frame 2
015354.719: hdmi: HDMI: Action unmute added to queue to happen at frame 3
015354.784: hdmi: HDMI: Action cec_init added to queue to happen at frame 3
015404.234: *** Restart logging
015404.297: brfs: File read: 15 bytes
016261.122: brfs: File read: /mfs/sd/cmdline.txt
016261.214: Read command line from file 'cmdline.txt':
016261.272: 'dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/sda2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait'
016261.607: brfs: File read: 131 bytes
024984.834: brfs: File read: /mfs/sd/kernel7.img
024984.897: Loading 'kernel7.img' to 0x8000 size 0x509c68
024989.308: No kernel trailer - assuming DT-capable
024989.400: brfs: File read: 5282920 bytes
025527.883: brfs: File read: /mfs/sd/bcm2710-rpi-3-b.dtb
025527.949: Loading 'bcm2710-rpi-3-b.dtb' to 0x511c68 size 0x64cb
025670.929: brfs: File read: 25803 bytes
026105.205: brfs: File read: /mfs/sd/config.txt
[... 72 lines cut ...]
016179.235: brfs: File read: /mfs/sd/cmdline.txt
016179.333: Read command line from file 'cmdline.txt':
016179.396: 'dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/sda2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait'
016179.748: brfs: File read: 131 bytes
024764.523: brfs: File read: /mfs/sd/kernel7.img
024764.589: Loading 'kernel7.img' to 0x8000 size 0x50dc28
024769.009: No kernel trailer - assuming DT-capable
024769.095: brfs: File read: 5299240 bytes
025307.358: brfs: File read: /mfs/sd/bcm2710-rpi-3-b.dtb
025307.425: Loading 'bcm2710-rpi-3-b.dtb' to 0x515c28 size 0x675f
025452.497: brfs: File read: 26463 bytes
025886.549: brfs: File read: /mfs/sd/config.txt
028120.796: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
028221.082: Device tree loaded to 0x2eff9400 (size 0x6b6e)
030810.504: vchiq_core: vchiq_init_state: slot_zero = 0xfad80000, is_master = 1
036681.418: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 134, power enable gpio = 133
036982.198: camsubs: Camera not found
036982.278: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 134, power enable gpio = 133
037283.046: camsubs: Camera not found
037283.121: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 134, power enable gpio = 133
037283.883: camsubs: Camera not found
037283.959: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 134, power enable gpio = 133
037284.711: camsubs: Camera not found
037284.783: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 134, power enable gpio = 133
037285.546: camsubs: Camera not found

ephestione, 2019-07-19:
Popping in to report that I have the same issue.
My specific HDMI configuration is:
```
disable_overscan=1
hdmi_force_hotplug=1
hdmi_group=2 #but I tested with 1 as well
hdmi_mode=82 #and I tested this with 16 when group was 1, nothing changed
hdmi_drive=2
config_hdmi_boost=4

```
This is a surveillance monitor, the raspi is always powered on with a dedicated 12V UPS'ed line, while the monitor, a s32f351, gets powered on only when there's people, so I need the hotplug active in order for the raspi to capture the video even when monitor is off for recording purposes.

As soon as I rebooted after latest kernel update I wasn't able to "screenshot" the video output (I use a tool that outputs a jpg file of the display)

I first tried the prerelease kernel with rpi-update, nothing changed.
Then like @baltaner I went back with rpi-update, but only up to 4.19.55.
Just now I found this thread and tried 4.19.55, and the image came back to life.

Since I got peace of mind finally, now I am going to test as per @popcornmix request.

ephestione, 2019-07-19:
Working, with kernel *.50

015752.673: brfs: File read: /mfs/sd/config.txt
015752.979: brfs: File read: 14 bytes
015874.712: HDMI:EDID version 1.3, 1 extensions, screen size 70x39 cm
015874.818: HDMI:EDID features - videodef 0x80 !standby !suspend active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspac
e; preferred format is native; does not support GTF
015874.939: HDMI:EDID found preferred CEA detail timing format: 1920x1080p @ 60 Hz (16)
015875.021: HDMI:EDID found DMT format: code 4, 640x480p @ 60 Hz in established timing I/II
015875.107: HDMI:EDID found DMT format: code 8, 800x600p @ 56 Hz in established timing I/II
015875.187: HDMI:EDID found DMT format: code 9, 800x600p @ 60 Hz in established timing I/II
015875.275: HDMI:EDID found DMT format: code 16, 1024x768p @ 60 Hz in established timing I/II
015875.404: HDMI:EDID standard timings block x 8: 0xB300 81C0 8100 8180 9500 A9C0 0101 0101                                                 015875.512: HDMI:EDID found DMT format: code 58, 1680x1050p @ 60 Hz (16:10) in standard timing 0
015875.618: HDMI:EDID found DMT format: code 85, 1280x720p @ 60 Hz (16:9) in standard timing 1
015875.717: HDMI:EDID found DMT format: code 28, 1280x800p @ 60 Hz (16:10) in standard timing 2
015875.816: HDMI:EDID found DMT format: code 35, 1280x1024p @ 60 Hz (5:4) in standard timing 3
015875.911: HDMI:EDID found DMT format: code 47, 1440x900p @ 60 Hz (16:10) in standard timing 4
015876.005: HDMI:EDID found DMT format: code 83, 1600x900p @ 60 Hz (16:9) in standard timing 5
015889.069: HDMI:EDID parsing v3 CEA extension 0
015889.150: HDMI:EDID monitor support - underscan IT formats:yes, basic audio:no, yuv444:yes, yuv422:yes, #native DTD:1
015889.247: HDMI:EDID found CEA detail timing format: 1280x720p @ 50 Hz (19)
015889.329: HDMI:EDID found CEA detail timing format: 720x576p @ 50 Hz (17)
015889.409: HDMI:EDID found CEA detail timing format: 720x480p @ 60 Hz (2)
015889.494: HDMI:EDID found CEA format: code 16, 1920x1080p @ 60Hz (native)
015889.573: HDMI:EDID found CEA format: code 4, 1280x720p @ 60Hz
015889.645: HDMI:EDID found CEA format: code 31, 1920x1080p @ 50Hz
015889.723: HDMI:EDID found CEA format: code 19, 1280x720p @ 50Hz
015889.794: HDMI:EDID found CEA format: code 18, 720x576p @ 50Hz
015889.867: HDMI:EDID found CEA format: code 3, 720x480p @ 60Hz
015889.929: HDMI:EDID found HDMI VSDB length 5
015889.996: HDMI:EDID HDMI VSDB has physical address 2.0.0.0
015890.056: HDMI:EDID HDMI VSDB has no extension fields
015890.142: HDMI:EDID filtering formats with pixel clock > 162 MHz or h. blanking > 1023
015890.326: HDMI:EDID preferred mode remained as CEA (16) 1920x1080p @ 60 Hz with pixel clock 148 MHz
016191.858: brfs: File read: /mfs/sd/config.txt
016192.031: HDMI:Setting property pixel encoding to Default
016192.089: HDMI:Setting property pixel clock type to PAL
016192.151: HDMI:Setting property content type flag to No data
016192.209: HDMI:Setting property fuzzy format match to enabled
016192.361: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
016390.252: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
016391.551: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
016391.909: hdmi: HDMI:>>>>>>>>>>>>>Rx sensed, reading EDID<<<<<<<<<<<<<
016404.020: hdmi: HDMI:EDID version 1.3, 1 extensions, screen size 70x39 cm
016404.137: hdmi: HDMI:EDID features - videodef 0x80 !standby !suspend active off; colour encoding:RGB444|YCbCr422; sRGB is not default colo
urspace; preferred format is native; does not support GTF
016404.245: hdmi: HDMI:EDID found preferred CEA detail timing format: 1920x1080p @ 60 Hz (16)
016404.338: hdmi: HDMI:EDID found DMT format: code 4, 640x480p @ 60 Hz in established timing I/II
016404.423: hdmi: HDMI:EDID found DMT format: code 8, 800x600p @ 56 Hz in established timing I/II                                           016404.512: hdmi: HDMI:EDID found DMT format: code 9, 800x600p @ 60 Hz in established timing I/II
016404.595: hdmi: HDMI:EDID found DMT format: code 16, 1024x768p @ 60 Hz in established timing I/II
016404.736: hdmi: HDMI:EDID standard timings block x 8: 0xB300 81C0 8100 8180 9500 A9C0 0101 0101
016404.839: hdmi: HDMI:EDID found DMT format: code 58, 1680x1050p @ 60 Hz (16:10) in standard timing 0
016404.944: hdmi: HDMI:EDID found DMT format: code 85, 1280x720p @ 60 Hz (16:9) in standard timing 1
016405.037: hdmi: HDMI:EDID found DMT format: code 28, 1280x800p @ 60 Hz (16:10) in standard timing 2
016405.131: hdmi: HDMI:EDID found DMT format: code 35, 1280x1024p @ 60 Hz (5:4) in standard timing 3
016405.235: hdmi: HDMI:EDID found DMT format: code 47, 1440x900p @ 60 Hz (16:10) in standard timing 4
016405.341: hdmi: HDMI:EDID found DMT format: code 83, 1600x900p @ 60 Hz (16:9) in standard timing 5
016418.400: hdmi: HDMI:EDID parsing v3 CEA extension 0
016418.482: hdmi: HDMI:EDID monitor support - underscan IT formats:yes, basic audio:no, yuv444:yes, yuv422:yes, #native DTD:1
016418.571: hdmi: HDMI:EDID found CEA detail timing format: 1280x720p @ 50 Hz (19)
016418.666: hdmi: HDMI:EDID found CEA detail timing format: 720x576p @ 50 Hz (17)
016418.756: hdmi: HDMI:EDID found CEA detail timing format: 720x480p @ 60 Hz (2)
016418.838: hdmi: HDMI:EDID found CEA format: code 16, 1920x1080p @ 60Hz (native)
016418.913: hdmi: HDMI:EDID found CEA format: code 4, 1280x720p @ 60Hz
016418.995: hdmi: HDMI:EDID found CEA format: code 31, 1920x1080p @ 50Hz
[... cut ...]
061545.408: alloc_compact_internal(space=876736, in_low_region=0, mode=7) failed
071337.518: alloc_compact_internal(space=876736, in_low_region=0, mode=7) failed
081461.905: alloc_compact_internal(space=876736, in_low_region=0, mode=7) failed
119326.872: alloc_compact_internal(space=876736, in_low_region=0, mode=7) failed
130168.017: alloc_compact_internal(space=876736, in_low_region=0, mode=7) failed
178639.984: alloc_compact_internal(space=876736, in_low_region=0, mode=7) failed
189568.031: alloc_compact_internal(space=876736, in_low_region=0, mode=7) failed

ephestione, 2019-07-19:
This is with latest stable 4.19.57, where there is no sign of the existence of a video subsystem:

015183.719: brfs: File read: /mfs/sd/config.txt
015184.028: brfs: File read: 14 bytes
015653.159: brfs: File read: /mfs/sd/config.txt
015653.469: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
015851.485: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
015852.764: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
015852.854: *** Restart logging
015852.918: brfs: File read: 14 bytes
015980.585: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
015980.660: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
016121.352: brfs: File read: /mfs/sd/cmdline.txt
016121.451: Read command line from file 'cmdline.txt':
016121.515: 'dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=3504b13c-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait'
016121.872: brfs: File read: 142 bytes
019659.581: brfs: File read: /mfs/sd/kernel7.img
019659.643: Loading 'kernel7.img' to 0x8000 size 0x50de50
019664.055: No kernel trailer - assuming DT-capable
019664.145: brfs: File read: 5299792 bytes
019766.603: brfs: File read: /mfs/sd/bcm2710-rpi-3-b.dtb
019766.668: Loading 'bcm2710-rpi-3-b.dtb' to 0x515e50 size 0x675f
019910.481: brfs: File read: 26463 bytes
019972.619: brfs: File read: /mfs/sd/config.txt
021816.490: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
021915.992: Device tree loaded to 0x2eff9400 (size 0x6b7a)
023299.916: vchiq_core: vchiq_init_state: slot_zero = 0xfad80000, is_master = 1
028848.965: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 134, power enable gpio = 133
029149.739: camsubs: Camera not found
029149.817: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 134, power enable gpio = 133
029450.592: camsubs: Camera not found
029450.670: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 134, power enable gpio = 133
029451.427: camsubs: Camera not found
029451.500: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 134, power enable gpio = 133
029452.262: camsubs: Camera not found
029452.338: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 134, power enable gpio = 133
029453.094: camsubs: Camera not found

ephestione, 2019-07-19:
BTW, with latest firmware:

```
tvservice -s
state 0x40002 [NTSC 4:3], 720x480 @ 60.00Hz, interlaced
```

```
tvservice -n
device_name=SAM-S32F351
```

```
tvservice -m CEA
Group CEA has 0 modes:
```

```
tvservice -m DMT
Group DMT has 0 modes:
```

ephestione, 2019-07-19:
In attachment is the output of
tvservice -d edid.txt
(with my HDMI directives in config.txt but oh well)
[edid.txt](https://github.com/raspberrypi/linux/files/3411952/edid.txt)

robin0793, 2019-07-25:
Hi, I have exactly the same issue (RPI3B+) with almost the same TV (Samsung UE60J). Any news in this? Can I provide any infos, logs or whatever to help?

yutayu, 2019-07-29:
same here.
3B

JamesH65, 2019-07-30:
There have been a number of changes in this area in the latest firmware, but that has not yet made it to rpi-update. Once it has will flag it here, and if people can then try it out to see if it has helped that would be useful. Should be next couple of days.

baileyt3, 2019-08-01:
Hi, any updates on this, like others, I had a perfectly working 3B+ but had not updated it for some time, I did sudo apt-get update, then sudo apt-get upgrade, re-booted and now no HDMI.
I can SSH ok and have used it to try several options proposed by various sites including this one, nothing works. I have also connected via VNC, when doing that the window is quite small compared to how it used to be, VCN size is related to the resolution, so it looks like its defaulting to:-
state 0x40000 [NTSC 4:3], 720x480 @ 60.00Hz, interlaced

There doesn't seem to be a way to override that setting no matter what you do in the config.txt

Looks like the upgrade has broked HDMI detecion on the 3B+ (so much for testing prior to release)

JamesH65, 2019-08-01:
@baileyt3 Have you tried an rpi-date today?

As for it being broken, in my testing it works fine on the HDMI devices I have, so it's likely some weirdness only seen with some monitors/TV's.

baileyt3, 2019-08-01:
Hi, yes tried twice today, says I'm up-to-date, tried it with an HDMI adaptor to PC VGA convertor which did work before and on an ASUS 1920x1080 HDMI monitor , again it used to work on this also

baileyt3, 2019-08-01:
Meant to add, I have a Pi 3B and have not updated it and that works fine with both (not going to touch that until this gets fixed)

baileyt3, 2019-08-01:
Here's a thing, I read about HDMI on https://www.raspberrypi.org/documentation/configuration/config-txt/video.md

and changed a few settings in my config.txt, I now have HDMI @1920x1080, not sure what happens if I try to change this using raspi-config for my other monitor.
So it really does look like it's not able to detect the monitor settings. (bug fix please :) )

# added by me
hdmi_ignore_edid=0xa5000080
framebuffer_width=1920
framebuffer_height=1080

# uncomment if hdmi display is not detected and composite is being output
hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=2
#hdmi_mode=4

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
hdmi_drive=2

baileyt3, 2019-08-01:
sorry, the copy/paste seems to have screwed up a bit on the comments

pelwell, 2019-08-01:
Hint: Use ` ``` `s for block quotes.

JamesH65, 2019-08-01:
On the Pi that works, when you connect it to one of the devices that don't work on the updated Pi, can you dump the edid that it has got using. Either post the resulting file here, or use base64 to convert to text and C&P result it here.

'tvservice -d fred.edid
base64 fred.edid'

robin0793, 2019-08-01:
From my not updated Pi on my Samsung UE60J:

```AP///////wAR7kUAAQEBAQAZAQOAc0F4KnwRnllHmycQUFQAAACBgJBAAQEBAQEBAQEBAQEBAjqA
GHE4LUBYLEUAEAkAAAAejArQiiDgLRAQPpYABAMAAAAYAAAA/ABERU5PTi1BVlIKICAgAAAA/QA7
PQ8uCAAKICAgICAgAR4CAx5xSpAEBRMUAQIDERIjCQcHgwEAAGYDDAATAAABHQByUdAeIG4oVQAQ
CQAAAB4BHYAYcRwWIFgsJQAQCQAAAJ4BHYDQchwWIBAsJYDEjiEAAJ4BHQC8UtAeILgoVUDEjiEA
AB4AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAApQ==
```

kencormack, 2019-08-01:
> There doesn't seem to be a way to override that setting no matter what you do in the config.txt

I experienced the same thing.  Until the "no HDMI" issue gets resolved, VNC's resolution can be forced to the size you need, using the framebuffer parameters in /boot/config.txt.  Set them to the resolution you need, then reboot the pi.  Your VNC sessions should work as desired.  In the examples below, I set mine to 1920x1080.

framebuffer_width=1920
framebuffer_height=1080

baileyt3, 2019-08-01:
I think you missed my later comments where I added those parameters, alone they don't work to resolve this, I had to add 
hdmi_ignore_edid=0xa5000080

also

this still needs fixing though, if you change to a different monitor, you will need to change these values.

kencormack, 2019-08-02:
Ah, yes, I did miss that.  Sorry about that.  I tried the "hdmi_ignore_edid=0xa5000080" setting and that seems to be working for me as well, until this gets sorted out.

baltaner, 2019-08-02:
Hi,
thanks a lot!
the rpi-update to 4.19.63-v7+ #1249 and 
the only setting: hdmi_ignore_edid=0xa5000080 in /boot/config.txt
did the thing! It works!
One little thing I notice: there is a little black border at the edges of the window...
I think, I can adjust them with the overscan settings in config.txt - but I haven't tried it yet...

Her for information: (blank config.txt with only setting: hdmi_ignore_edid=0xa5000080)
root@pi420:/home/pi# tvservice -s
state 0xa [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive
root@pi420:/home/pi# base64 samsung.dat
AP///////wBMLbAFAAAAAA0TAQOAEAl4Cu6Ro1RMmSYPUFS/74BxT4EAgUCBgJUAlQ+pQLMAAjqA
GHE4LUBYLEUAoFoAAAAeAR0AvFLQHiC4KFVAoFoAAAAeAAAA/QAYSxpRFwAKICAgICAgAAAA/ABT
eW5jTWFzdGVyCiAgAdECAyPxS5MEBRQDEhAfICEiIwkHB4MBAADiAA9nAwwAEAC4LQEdgNByHBYg
ECwlgKBaAAAAngEdgBhxHBYgWCwlAKBaAAAAngEdAHJR0B4gbihVAKBaAAAAHowK0JAgQDEgDEBV
AKBaAAAAGIwK0Iog4C0QED6WAKBaAAAAGAAAVw==

ephestione, 2019-08-02:
@baileyt3 in order to make it clear:
- your HDMI output works on latest firmware if you set hdmi_ignore_edid=0xa5000080 in config.txt whereas if you don't you get blank screen like everyone else
- how did you find out the value to give to that parameter for your monitor?

baileyt3, 2019-08-02:
Hi ephestione, if you check my earlier post, I went through the document here:- https://www.raspberrypi.org/documentation/configuration/config-txt/video.md
thats where I found the line to add, and yes, if I remove that line I get the blank screen.

One other thing, off topic, I noticed that my Wifi is no longer being detected too, so working on resolving that. Looking like a few major bugs in this update.

JamesH65, 2019-08-02:
@robin0793 That edid you provided is odd. Can you dump it again, and run edid-decode on it? You could also try edid-parser, which in this case failed to decode the edid at all.

`Extracted contents:
header:          18 71 38 2d 40 58 2c 45
serial number:   00 10 09 00 00 00 1e 8c 0a d0
version:         8a 20
basic params:    e0 2d 10 10 3e
chroma info:     96 00 04 03 00 00 00 18 00 00
established:     00 fc 00
standard:        44 45 4e 4f 4e 2d 41 56 52 0a 20 20 20 00 00 00
descriptor 1:    fd 00 3b 3d 0f 2e 08 00 0a 20 20 20 20 20 20 01 1e 02
descriptor 2:    03 1e 71 4a 90 04 05 13 14 01 02 03 11 12 23 09 07 07
descriptor 3:    83 01 00 00 66 03 0c 00 13 00 00 01 1d 00 72 51 d0 1e
descriptor 4:    20 6e 28 55 00 10 09 00 00 00 1e 01 1d 80 18 71 1c 16
extensions:      20
checksum:        58

No header found
Manufacturer: @@P Model 9 Serial Number 2350776320
EDID version: 138.32
Digital display
Maximum image size: 45 cm x 16 cm
Gamma: 1.16
DPMS levels: Off
Supported color formats: RGB 4:4:4, YCrCb 4:4:4, YCrCb 4:2:2
Default (sRGB) color space is primary color space
First detailed timing is preferred timing
Established timings supported:
  800x600@72Hz
  800x600@75Hz
  832x624@75Hz
  1280x768@87Hz
  1024x768@60Hz
  1024x768@70Hz
Standard timings supported:
  792x594@65Hz
  872x654@75Hz
  872x872@105Hz
  768x576@82Hz
  904x904@70Hz
  504x504@92Hz
  504x504@60Hz
non-conformant standard timing (0 horiz)
Detailed mode: Clock 2.530 MHz, 544 mm x 32 mm
                 59   69  613 3960 hborder 1
                 46   48   48   54 vborder 30
               +hsync -vsync analog composite 
Detailed mode: Clock 76.830 MHz, 529 mm x 786 mm
               2417 2437 2438 2491 hborder 9
                260  260  310 1033 vborder 7
               +hsync +vsync analog composite 
Detailed mode: Clock 3.870 MHz, 1821 mm x 512 mm
               1536 1555 1555 3072 hborder 81
                  3    3   19   15 vborder 208
               +hsync +vsync 
Detailed mode: Clock 281.920 MHz, 285 mm x 2176 mm
                 40   40   40  125 hborder 113
                 16   17   47   25 vborder 28
               +hsync +vsync digital composite 
Has 32 extension blocks
Checksum: 0x58 (should be 0x78)
EDID block does not conform at all!
	Block has broken checksum
	Bad year of manufacture
	Manufacturer name field contains garbage
`

pelwell, 2019-08-02:
base64 expands the input in a fixed ratio (excluding line ending and padding issues). Since a good EDID in base64 has 4 lines of 76 characters followed by a single line of 40 characters, while @robin0793's has 3 lines of 76 characters followed by a line of 40 character, it seems likely that @robin0793 just missed a line.

JamesH65, 2019-08-02:
Thanks Phil. 

I cannot use the EDID supplied by @baltaner above, since that is a fake EDID used when ignore EDID is set. What I really need is the EDID of a monitor/TV that fails to start up with the latest firmware, so would need to be generated from a Pi that does not yet have that installed.

ephestione, 2019-08-02:
@baileyt3
Got it, thanks, now I remember ending up on that page myself while scavenging for information, and even if I didn't try that parameter specifically, I did try hdmi_safe=1 which includes the edid ignore, but that didn't solve it for me

ephestione, 2019-08-02:
@JamesH65
Did you have the chance of checking the edid that I attached a while ago?

JamesH65, 2019-08-02:
@ephestione That EDID worked fine on my system, booted in to a native CEA 1080p60 (mode 16), as expected given the content of the EDID. What was the problem you were seeing?

ephestione, 2019-08-02:
@JamesH65 simply no video output with any firmware newer than 4.19.50, didn't try the very latest firmware of these days though as I'm physically away for a while and don't want to test when I'm not on the premises

JamesH65, 2019-08-02:
I'm on 4.19.57. There are probably some relevant changes in the later firmwares, so worth trying.

ephestione, 2019-08-02:
I tested up to .58 with no benefits, it's indeed strange that it worked for you with .57.
Will be testing again in a few days

JamesH65, 2019-08-02:
@ephestione Just realised you are using a 3B, will need to redo test on that particular model.

robin0793, 2019-08-02:
You're right, I missed a line, sorry. Here you go. 
```
AP///////wAR7kUAAQEBAQAZAQOAhUt4CiOlpFRNmSYPUFS97wBxT4HAgQCBgJUAqcCzAAEBAjqA
GHE4LUBYLEUAuahCAAAeAR2AGHEcFiBYLCUAxI4hAACeAAAA/ABERU5PTi1BVlIKICAgAAAA/QA7
PQ8uCAAKICAgICAgAVICAztxS5AEEwUUAxIPJB4mNQ9/Bz0ewBUHUF9+A1cGA2d+A19+AYNfAABm
AwwAEwCA4gAP5hFG0ABwAAEdAHJR0B4gbihVABAJAAAAHgEdgNByHBYgECwlgMSOIQAAnowK0Iog
4C0QED6WABAJAAAAGAAAAAAAAAAAAAAAAAAAkg==
```

And the output from edidparser:
```
Enabling fuzzy format match...
Parsing fred.edid...
HDMI:EDID version 1.3, 1 extensions, screen size 133x75 cm
HDMI:EDID features - videodef 0x80 !standby !suspend !active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is native; does not support GTF
HDMI:EDID found monitor name descriptor tag 0xfc
HDMI:EDID monitor name is DENON-AVR
HDMI:EDID found monitor range descriptor tag 0xfd
HDMI:EDID monitor range offsets: V min=0, V max=0, H min=0, H max=0HDMI:EDID monitor range: vertical is 59-61 Hz, horizontal is 15-46 kHz, max pixel clock is 80 MHz
HDMI:EDID monitor range does not support GTF
HDMI:EDID found preferred CEA detail timing format: 1920x1080p @ 60 Hz (16)
HDMI:EDID found CEA detail timing format: 1920x1080i @ 60 Hz (5)
HDMI:EDID established timing I/II bytes are BD EF 00
HDMI:EDID found DMT format: code 4, 640x480p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 5, 640x480p @ 72 Hz in established timing I/II
HDMI:EDID found DMT format: code 6, 640x480p @ 75 Hz in established timing I/II
HDMI:EDID found DMT format: code 9, 800x600p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 10, 800x600p @ 72 Hz in established timing I/II
HDMI:EDID found DMT format: code 11, 800x600p @ 75 Hz in established timing I/II
HDMI:EDID found DMT format: code 16, 1024x768p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 17, 1024x768p @ 70 Hz in established timing I/II
HDMI:EDID found DMT format: code 18, 1024x768p @ 75 Hz in established timing I/II
HDMI:EDID found DMT format: code 36, 1280x1024p @ 75 Hz in established timing I/II
HDMI:EDID standard timings block x 8: 0x714F 81C0 8100 8180 9500 A9C0 B300 0101
HDMI:EDID found DMT format: code 21, 1152x864p @ 75 Hz (4:3) in standard timing 0
HDMI:EDID found DMT format: code 85, 1280x720p @ 60 Hz (16:9) in standard timing 1
HDMI:EDID found DMT format: code 28, 1280x800p @ 60 Hz (16:10) in standard timing 2
HDMI:EDID found DMT format: code 35, 1280x1024p @ 60 Hz (5:4) in standard timing 3
HDMI:EDID found DMT format: code 47, 1440x900p @ 60 Hz (16:10) in standard timing 4
HDMI:EDID found DMT format: code 83, 1600x900p @ 60 Hz (16:9) in standard timing 5
HDMI:EDID found DMT format: code 58, 1680x1050p @ 60 Hz (16:10) in standard timing 6
HDMI:EDID parsing v3 CEA extension 0
HDMI:EDID monitor support - underscan IT formats:no, basic audio:yes, yuv444:yes, yuv422:yes, #native DTD:1
HDMI:EDID found CEA detail timing format: 1280x720p @ 60 Hz (4)
HDMI:EDID found CEA detail timing format: 1920x1080i @ 50 Hz (20)
HDMI:EDID found CEA detail timing format: 720x480p @ 60 Hz (2)
HDMI:EDID found CEA format: code 16, 1920x1080p @ 60Hz (native)
HDMI:EDID found CEA format: code 4, 1280x720p @ 60Hz
HDMI:EDID found CEA format: code 19, 1280x720p @ 50Hz
HDMI:EDID found CEA format: code 5, 1920x1080i @ 60Hz
HDMI:EDID found CEA format: code 20, 1920x1080i @ 50Hz
HDMI:EDID found CEA format: code 3, 720x480p @ 60Hz
HDMI:EDID found CEA format: code 18, 720x576p @ 50Hz
HDMI:EDID found CEA format: code 15, 1440x480p @ 60Hz
HDMI:EDID found CEA format: code 36, 2880x480p @ 60Hz
HDMI:EDID found CEA format: code 30, 1440x576p @ 50Hz
HDMI:EDID found CEA format: code 38, 2880x576p @ 50Hz
HDMI:EDID found audio format 8 channels PCM, sample rate: 32|44|48|88|96|176|192 kHz, sample size: 16|20|24 bits
HDMI:EDID found audio format 6 channels DTS, sample rate: 44|48|88|96 kHz, bitrate: 1536 kbps
HDMI:EDID found audio format 6 channels AC3, sample rate: 32|44|48 kHz, bitrate: 640 kbps
HDMI:EDID found audio format 8 channels DTS-HD, sample rate: 44|48|88|96|176|192 kHz, codec define: 3
HDMI:EDID found audio format 8 channels Dobly Digital+, sample rate: 44|48 kHz, codec define: 3
HDMI:EDID found audio format 8 channels MLP, sample rate: 44|48|88|96|176|192 kHz, codec define: 3
HDMI:EDID found audio format 8 channels DTS-HD, sample rate: 44|48|88|96|176|192 kHz, codec define: 1
HDMI:EDID found HDMI VSDB length 6
HDMI:EDID HDMI VSDB has physical address 1.3.0.0
HDMI:EDID HDMI VSDB supports AI:yes, dual link DVI:no
HDMI:EDID HDMI VSDB deep colour support - 48-bit:no 36-bit:no 30-bit:no DC_yuv444:no
HDMI:EDID HDMI VSDB has no latency information
HDMI:EDID found Video Capability DB length 2
[... 16 lines cut ...]
HDMI:EDID DMT mode (11) 800x600p @ 75 Hz with pixel clock 49 MHz has a score of 36000
HDMI:EDID CEA mode (15) 1440x480p @ 60 Hz with pixel clock 54 MHz has a score of 66472
HDMI:EDID best score mode is now CEA (16) 1920x1080p @ 60 Hz with pixel clock 148 MHz (score 5398248)
HDMI:EDID DMT mode (16) 1024x768p @ 60 Hz with pixel clock 65 MHz has a score of 47185
HDMI:EDID CEA mode (17) 720x576p @ 50 Hz with pixel clock 27 MHz has a score of 66472
HDMI:EDID DMT mode (17) 1024x768p @ 70 Hz with pixel clock 75 MHz has a score of 55050
HDMI:EDID CEA mode (18) 720x576p @ 50 Hz with pixel clock 27 MHz has a score of 66472
HDMI:EDID DMT mode (18) 1024x768p @ 75 Hz with pixel clock 78 MHz has a score of 58982
[... cut ...]
HDMI:EDID DMT mode (47) 1440x900p @ 60 Hz with pixel clock 106 MHz has a score of 102760
HDMI:EDID DMT mode (58) 1680x1050p @ 60 Hz with pixel clock 146 MHz has a score of 130840
HDMI:EDID DMT mode (83) 1600x900p @ 60 Hz with pixel clock 108 MHz has a score of 111400
HDMI:EDID DMT mode (85) 1280x720p @ 60 Hz with pixel clock 74 MHz has a score of 80296
HDMI:EDID preferred mode remained as CEA (16) 1920x1080p @ 60 Hz with pixel clock 148 MHz
HDMI:EDID has HDMI support and audio support
edidparser exited with code 0
```

kencormack, 2019-08-02:
> I cannot use the EDID supplied by @baltaner above, since that is a fake EDID used when ignore EDID is set. What I really need is the EDID of a monitor/TV that fails to start up with the latest firmware, so would need to be generated from a Pi that does not yet have that installed.

Here is the parsed edid data for my HP 22CWA flatscreen HDMI display, that comes up blank with the new kernel.  I grabbed this before updating to the new kernel, and before adding "hdmi_ignore_edid=0xa5000080" to my config.txt...

Enabling fuzzy format match...
Parsing edid.dat...
HDMI:EDID version 1.3, 1 extensions, screen size 48x27 cm
HDMI:EDID features - videodef 0x80 !standby !suspend active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is native; does not support GTF
HDMI:EDID found monitor range descriptor tag 0xfd
HDMI:EDID monitor range offsets: V min=0, V max=0, H min=0, H max=0
HDMI:EDID monitor range: vertical is 50-60 Hz, horizontal is 30-80 kHz, max pixel clock is 170 MHz
HDMI:EDID monitor range does not support GTF
HDMI:EDID found monitor name descriptor tag 0xfc
HDMI:EDID monitor name is HP_22cwa
HDMI:EDID found monitor S/N descriptor tag 0xff
HDMI:EDID found preferred CEA detail timing format: 1920x1080p @ 60 Hz (16)
HDMI:EDID established timing I/II bytes are A1 08 00
HDMI:EDID found DMT format: code 4, 640x480p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 9, 800x600p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 16, 1024x768p @ 60 Hz in established timing I/II
HDMI:EDID standard timings block x 8: 0xD1C0 81C0 A9C0 B300 9500 8180 8100 0101
HDMI:EDID found DMT format: code 82, 1920x1080p @ 60 Hz (16:9) in standard timing 0
HDMI:EDID found DMT format: code 85, 1280x720p @ 60 Hz (16:9) in standard timing 1
HDMI:EDID found DMT format: code 83, 1600x900p @ 60 Hz (16:9) in standard timing 2
HDMI:EDID found DMT format: code 58, 1680x1050p @ 60 Hz (16:10) in standard timing 3
HDMI:EDID found DMT format: code 47, 1440x900p @ 60 Hz (16:10) in standard timing 4
HDMI:EDID found DMT format: code 35, 1280x1024p @ 60 Hz (5:4) in standard timing 5
HDMI:EDID found DMT format: code 28, 1280x800p @ 60 Hz (16:10) in standard timing 6
HDMI:EDID parsing v3 CEA extension 0
HDMI:EDID monitor support - underscan IT formats:yes, basic audio:yes, yuv444:yes, yuv422:yes, #native DTD:1
HDMI:EDID found CEA detail timing format: 1920x1080p @ 60 Hz (16)
HDMI:EDID found CEA detail timing format: 1920x1080p @ 50 Hz (31)
HDMI:EDID found CEA format: code 16, 1920x1080p @ 60Hz (native)
HDMI:EDID found CEA format: code 31, 1920x1080p @ 50Hz
HDMI:EDID found CEA format: code 4, 1280x720p @ 60Hz
HDMI:EDID found CEA format: code 19, 1280x720p @ 50Hz
HDMI:EDID found CEA format: code 3, 720x480p @ 60Hz
HDMI:EDID found CEA format: code 2, 720x480p @ 60Hz
HDMI:EDID found CEA format: code 18, 720x576p @ 50Hz
HDMI:EDID found CEA format: code 17, 720x576p @ 50Hz
HDMI:EDID found CEA format: code 1, 640x480p @ 60Hz
HDMI:EDID found audio format 2 channels PCM, sample rate: 32|44|48 kHz, sample size: 16|20|24 bits
HDMI:EDID found HDMI VSDB length 7
HDMI:EDID HDMI VSDB has physical address 1.0.0.0
HDMI:EDID HDMI VSDB supports AI:no, dual link DVI:no
HDMI:EDID HDMI VSDB deep colour support - 48-bit:no 36-bit:no 30-bit:no DC_yuv444:no
HDMI:EDID HDMI VSDB max TMDS clock 170 MHz
HDMI:EDID HDMI VSDB has no latency information
HDMI:EDID filtering formats with pixel clock > 162 MHz or h. blanking > 1023
HDMI:EDID best score mode initialised to CEA (1) 640x480p @ 60 Hz with pixel clock 25 MHz (score 0)
HDMI:EDID best score mode is now CEA (1) 640x480p @ 60 Hz with pixel clock 25 MHz (score 61864)
HDMI:EDID best score mode is now CEA (2) 720x480p @ 60 Hz with pixel clock 27 MHz (score 66472)
HDMI:EDID CEA mode (3) 720x480p @ 60 Hz with pixel clock 27 MHz has a score of 66472
HDMI:EDID best score mode is now CEA (4) 1280x720p @ 60 Hz with pixel clock 74 MHz (score 135592)
HDMI:EDID DMT mode (4) 640x480p @ 60 Hz with pixel clock 25 MHz has a score of 18432
HDMI:EDID DMT mode (9) 800x600p @ 60 Hz with pixel clock 40 MHz has a score of 28800
HDMI:EDID best score mode is now CEA (16) 1920x1080p @ 60 Hz with pixel clock 148 MHz (score 5398248)
HDMI:EDID DMT mode (16) 1024x768p @ 60 Hz with pixel clock 65 MHz has a score of 47185
HDMI:EDID CEA mode (17) 720x576p @ 50 Hz with pixel clock 27 MHz has a score of 66472
HDMI:EDID CEA mode (18) 720x576p @ 50 Hz with pixel clock 27 MHz has a score of 66472
HDMI:EDID CEA mode (19) 1280x720p @ 50 Hz with pixel clock 74 MHz has a score of 117160
HDMI:EDID DMT mode (28) 1280x800p @ 60 Hz with pixel clock 83 MHz has a score of 86440
HDMI:EDID CEA mode (31) 1920x1080p @ 50 Hz with pixel clock 148 MHz has a score of 4232360
HDMI:EDID DMT mode (35) 1280x1024p @ 60 Hz with pixel clock 108 MHz has a score of 103643
HDMI:EDID DMT mode (47) 1440x900p @ 60 Hz with pixel clock 106 MHz has a score of 102760
HDMI:EDID DMT mode (58) 1680x1050p @ 60 Hz with pixel clock 146 MHz has a score of 130840
HDMI:EDID DMT mode (82) 1920x1080p @ 60 Hz with pixel clock 148 MHz has a score of 149416
HDMI:EDID DMT mode (83) 1600x900p @ 60 Hz with pixel clock 108 MHz has a score of 111400
HDMI:EDID DMT mode (85) 1280x720p @ 60 Hz with pixel clock 74 MHz has a score of 80296
HDMI:EDID preferred mode remained as CEA (16) 1920x1080p @ 60 Hz with pixel clock 148 MHz
HDMI:EDID has HDMI support and audio support
edidparser exited with code 0

ephestione, 2019-08-03:
Updated to latest firmware on my 3B with rpi-update:
```
uname -r
4.19.63-v7+
```
video out works as intended now, yay!!
I mean, at least it should.
I rebooted remotely after update, and the monitor is actually off, as in totally off since AC mains to it are shut down, RPi is on its 12V permanent line.
I don't know how HDMI standard works, maybe even a non-powered monitor gives out info about itself when connected to HDMI.

```
tvservice -s
state 0xa [HDMI DMT (82) RGB full 16:9], 1920x1080 @ 60.00Hz, progressive
tvservice -n
device_name=SAM-S32F351
tvservice -m CEA
Group CEA has 9 modes:
           mode 1: 640x480 @ 60Hz 4:3, clock:25MHz progressive
           mode 2: 720x480 @ 60Hz 4:3, clock:27MHz progressive
           mode 3: 720x480 @ 60Hz 16:9, clock:27MHz progressive
           mode 4: 1280x720 @ 60Hz 16:9, clock:74MHz progressive
  (native) mode 16: 1920x1080 @ 60Hz 16:9, clock:148MHz progressive
           mode 17: 720x576 @ 50Hz 4:3, clock:27MHz progressive
           mode 18: 720x576 @ 50Hz 16:9, clock:27MHz progressive
           mode 19: 1280x720 @ 50Hz 16:9, clock:74MHz progressive
           mode 31: 1920x1080 @ 50Hz 16:9, clock:148MHz progressive
tvservice -m DMT
Group DMT has 11 modes:
           mode 4: 640x480 @ 60Hz 4:3, clock:25MHz progressive
           mode 8: 800x600 @ 56Hz 4:3, clock:36MHz progressive
           mode 9: 800x600 @ 60Hz 4:3, clock:40MHz progressive
           mode 16: 1024x768 @ 60Hz 4:3, clock:65MHz progressive
           mode 28: 1280x800 @ 60Hz 16:10, clock:83MHz progressive
           mode 35: 1280x1024 @ 60Hz 5:4, clock:108MHz progressive
           mode 47: 1440x900 @ 60Hz 16:10, clock:106MHz progressive
           mode 58: 1680x1050 @ 60Hz 16:10, clock:146MHz progressive
  (prefer) mode 82: 1920x1080 @ 60Hz 16:9, clock:148MHz progressive
           mode 83: 1600x900 @ 60Hz 16:9, clock:108MHz progressive
           mode 85: 1280x720 @ 60Hz 16:9, clock:74MHz progressive
```

base64 edid:

```
AP///////wBMLSQNQ05aWhkbAQOARid4KiOtpFRNmSYPR0ojCACzAIHAgQCBgJUAqcABAQEBAjqA
GHE4LUBYLEUAuokhAAAeAAAA/QAyPB5REQAKICAgICAgAAAA/ABTMzJGMzUxCiAgICAgAAAA/wBI
NFpKNjAwNjMwCiAgAXkCAxGxRpAEHxMSA2UDDAAgAAEdALxS0B4guChVQLqJIQAAHowK0JAgQDEg
DEBVALqJIQAAGIwK0Iog4C0QED6WALqJIQAAGAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAg==

```

baileyt3, 2019-08-03:
Thanks for the update, it does look like its now detecting the monitor, I will try udate again tomorrow, one additional thing not related to HDMI, but after the update I also find WiFi and Camera no longer work, hope this update fixes everything, cos its a mess at the moment

baileyt3, 2019-08-06:
Managed to do and update today, still have to use the manual fix for HDMI, Wifi still not being detected and camera still doesn't work.
However, I downloaded the latest Pi image and installed on a USB stick and everything works as expected. Next step it to update/upgrade and see if it still works.
Installing a new image is a real pain though as I run my Pi off an SSD hard drive and to have to reinstall everything would take forever.

JamesH65, 2019-08-06:
@baileyt3 So, a fresh install of Buster on a USB and you still need the ignore_edid config, but the camera and wifi work OK?

baileyt3, 2019-08-06:
no, fresh install and it worked straight away, no problems at all, doing the update/upgrade right now, will report back as soon as it's finished

baileyt3, 2019-08-06:
Completed the update/upgrade and it all still works, so it looks like the problem is if you upgrade to the latest version from a much older installation it breaks many things and there doesn't seem to be a way to fix them all without a fresh install. ;((

kencormack, 2019-08-07:
My "much older version" was stretch, with everything current.the day before the kernel upgrade.  (This was on a Pi 3B.). It's not a USB boot issue, and not an "upgrade from much older version" issue.  It seems to be a kernel unable to deal with the edid of certain HDMI displays.

JamesH65, 2019-08-07:
@kencormack Can you provide the EDID please - I see you posted a parsed version, but I would like to try the binary.

```
tvservice -d fred.edid
base64 fred.edid
```
Then C&P the results here.

orohde, 2019-08-08:
Hi,
unfortunately I discovered exactly the same problem with Stretch yesterday. Updated the kernel and suddenly my monitor stopped working.
By editing the config.txt like described here I was able to revitalize the monitor but only in unusable 640x480 resolution. Everything else was unreadable.

I am using a 3B+

thank god I made a backup before updating the kernel, so I could revert back and get this information:

```
uname -r 
4.14.98-v7+

tvservice -s
state 0x12000a [HDMI DMT (16) RGB full 4:3], 1024x768 @ 60.00Hz, progressive

tvservice -n
device_name=KEB-display

tvservice -m CEA
Group CEA has 11 modes:
           mode 1: 640x480 @ 60Hz 4:3, clock:25MHz progressive 
           mode 2: 720x480 @ 60Hz 4:3, clock:27MHz progressive 
           mode 3: 720x480 @ 60Hz 16:9, clock:27MHz progressive 
  (native) mode 4: 1280x720 @ 60Hz 16:9, clock:74MHz progressive 
           mode 15: 720x480 @ 60Hz 16:9, clock:54MHz x2 progressive 
  (native) mode 16: 1920x1080 @ 60Hz 16:9, clock:148MHz progressive 
           mode 17: 720x576 @ 50Hz 4:3, clock:27MHz progressive 
           mode 18: 720x576 @ 50Hz 16:9, clock:27MHz progressive 
           mode 19: 1280x720 @ 50Hz 16:9, clock:74MHz progressive 
           mode 30: 720x576 @ 50Hz 16:9, clock:54MHz x2 progressive 
           mode 31: 1920x1080 @ 50Hz 16:9, clock:148MHz progressive 

tvservice -m DMT
Group DMT has 10 modes:
           mode 4: 640x480 @ 60Hz 4:3, clock:25MHz progressive 
           mode 5: 640x480 @ 72Hz 4:3, clock:31MHz progressive 
           mode 9: 800x600 @ 60Hz 4:3, clock:40MHz progressive 
  (prefer) mode 16: 1024x768 @ 60Hz 4:3, clock:65MHz progressive 
           mode 32: 1280x960 @ 60Hz 4:3, clock:108MHz progressive 
           mode 35: 1280x1024 @ 60Hz 5:4, clock:108MHz progressive 
           mode 42: 1400x1050 @ 60Hz 4:3, clock:121MHz progressive 
           mode 47: 1440x900 @ 60Hz 16:10, clock:106MHz progressive 
           mode 82: 1920x1080 @ 60Hz 16:9, clock:148MHz progressive 
           mode 85: 1280x720 @ 60Hz 16:9, clock:74MHz progressive 

base64 keb.edid
AP///////wAsolYBAQEBAQ4cAQOAHhp4PmQ1pVRPnicSUFQpCADRwNHA0cCVAJBAgYCBQIHAAjqA
GHE4LUBYLCUALAQRAAAeAAAA/ABkaXNwbGF5CiAgICAgAAAA/wAKICAgICAgICAgICAgAAAA/QAw
PR5QDwEKICAgICAgAc8CA0PxSQIQEBASDx8ehCMJBweDAQAAZwMMABAAOERn2F3EAUSAAWgaAAAB
CSg85uUPAAAGAOMFwADiAP/mBgUBWVlJAjqAGHE4LUBYLCUALAQRAAAeAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAATA==
```

Would be great if you could help.

Berchemer, 2019-08-11:
Same problem here. Upgraded a running system fresh installed at the begin of Juli 2019.
After update and upgrade I had no HDMI-output anymore.

Not only on one raspberry. I have two different raspberry 3B+ connected to two different  TVs. One is a Medion TV and the other one is an older SONY-TV. Both with CEC-Support to use the TV-Remote automatically over HDMI.

Now apt-get update and apt-get upgrade this morning.
Adding "hdmi_ignore_edid=0xa5000080" to /boot/config.txt gives me  an hdmi-output again. But usage of CEC inside KODI does not work anymore.

My Data:
MEDION-TV:

uname -r
4.19.58-v7+

uname -a
Linux raspberrypi-1OG-Salon 4.19.58-v7+ #1245 SMP Fri Jul 12 17:25:51 BST 2019 armv7l GNU/Linux

tvservice -n
device_name=SAM-SyncMaster

tvservice -s
state 0xa [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive

 tvservice -m CEA
Group CEA has 14 modes:
           mode 1: 640x480 @ 60Hz 4:3, clock:25MHz progressive
           mode 2: 720x480 @ 60Hz 4:3, clock:27MHz progressive
           mode 3: 720x480 @ 60Hz 16:9, clock:27MHz progressive
           mode 4: 1280x720 @ 60Hz 16:9, clock:74MHz progressive
           mode 5: 1920x1080 @ 60Hz 16:9, clock:74MHz interlaced
  (prefer) mode 16: 1920x1080 @ 60Hz 16:9, clock:148MHz progressive
           mode 17: 720x576 @ 50Hz 4:3, clock:27MHz progressive
           mode 18: 720x576 @ 50Hz 16:9, clock:27MHz progressive
  (native) mode 19: 1280x720 @ 50Hz 16:9, clock:74MHz progressive
           mode 20: 1920x1080 @ 50Hz 16:9, clock:74MHz interlaced
           mode 31: 1920x1080 @ 50Hz 16:9, clock:148MHz progressive
           mode 32: 1920x1080 @ 24Hz 16:9, clock:74MHz progressive
           mode 33: 1920x1080 @ 25Hz 16:9, clock:74MHz progressive
           mode 34: 1920x1080 @ 30Hz 16:9, clock:74MHz progressive

tvservice -d fred.edid
base64 fred.edid
`AP///////wBMLbAFAAAAAA0TAQOAEAl4Cu6Ro1RMmSYPUFS/74BxT4EAgUCBgJUAlQ+pQLMAAjqA
GHE4LUBYLEUAoFoAAAAeAR0AvFLQHiC4KFVAoFoAAAAeAAAA/QAYSxpRFwAKICAgICAgAAAA/ABT
eW5jTWFzdGVyCiAgAdECAyPxS5MEBRQDEhAfICEiIwkHB4MBAADiAA9nAwwAEAC4LQEdgNByHBYg
ECwlgKBaAAAAngEdgBhxHBYgWCwlAKBaAAAAngEdAHJR0B4gbihVAKBaAAAAHowK0JAgQDEgDEBV
AKBaAAAAGIwK0Iog4C0QED6WAKBaAAAAGAAAVw==`

I also would be happy, if you could help me :-)

dimov-cz, 2019-09-09:
+1

Same story here, thanks for the tip with "rpi-update 458871abedfd68ea417d24577694dbe9d7d4db4c".

dimov-cz, 2019-09-10:
After struggling with no hdmi problem, tvservice not working properly and no CEC after upgrade to buster. I did clean install today with current Buster to SD card (Raspberry Pi 3 Model B Rev 1.2). Everything worked again like a charm. But because I'm using external SSD i cloned this new installation and all the problems are back again. So maybe it's related to firmware, bootcode.bin..

[3 more replies]