raspberrypi/linux #3084: 4.19.58 doesn't boot on 3B+

trilean, 2019-07-17:
...at least there's no HDMI output.

I'm compiling my own kernels because I'm running Gentoo on my Pi. I have a working config + image for 4.19.25. Using this config on the current tree (rpi-4.19.y/4.19.58) only results in the rainbow pattern on HDMI.

Now I've tested `bcmrpi3_defconfig`. The only changes are:
- no module support - everything is built in
- built in firmware (`CONFIG_EXTRA_FIRMWARE`)
- `CONFIG_RTL8192CU` is not set (wouldn't link)

Still I'm just getting the rainbow pattern. It's the same on 5.2.1.

I'm using the dtbs built with my kernel and the latest `bootcode.bin` / `fixup.dat` / `start.elf`.

Any ideas how to debug this?

lategoodbye, 2019-07-24:
In case the RPi 3B+ is able to mount the rootfs, there should be the kernel log available under /var/log/kern.log

Another way to debug this issue is to connect a serial TTL adapter to the debug UART (GPIO14,15).

trilean, 2019-07-28:
I dug up my USB-serial converter. The serial console works fine. After a couple of reboots the monitor suddenly came to life, but the image was badly distorted. It might be an issue with the hardware. Though if it is I don't understand why it works with an older kernel.

closed 2019-07-28 by trilean