raspberrypi/linux #3085: Kernel 5.1.16-v7l+ unable to reboot Pi4B

TheEnbyperor, 2019-07-17:
**Describe the bug**
Kernel is unable to reboot the processor of the pi4b

**To reproduce**
1) use kernel 5.1.16-v7l+
2) `sudo reboot`
3) system goes through reboot sequence
4) kernel hangs trying to reboot

**Expected behaviour**
kernel reboots the processor

**Actual behaviour**
kernel does not reboot the processo

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi?
Pi4B
* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-06-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 150e25c4f8123a4c9c63e8dca1b4737fa6c1135c, stage2
* Which firmware version (`vcgencmd version`)?
Jun 20 2019 16:04:31 
Copyright (c) 2012 Broadcom
version 407b1da8fa3d1a7108cb1d250f5064a3420d2b7d (clean) (release) (start)
* Which kernel version (`uname -a`)?
Linux 0a9d2112 5.1.16-v7l+ #3 SMP Sun Jul 7 09:44:08 BST 2019 armv7l GNU/Linux

pelwell, 2019-07-17:
5.1.18-v7l+ reboots just fine for me:

1. Try with no external hardware attached.
2. Try with an updated, firmware and "official" 4.19 kernel (`sudo rpi-update`).
3. Now replace the kernel with your own.
4. Try building the latest 5.1.18 kernel yourself (you didn't get a 5.1 kernel pre-built from us), but I doubt it will make a difference.

lategoodbye, 2019-07-17:
@TheEnbyperor Did you compiled the watchdog driver as a module?
In this case you are affected by a regression which was introduced by Linux 5.1.
Here is the [fix](https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/?h=next-20190717&id=215e06f0d18d5d653d6ea269e4dfc684854d48bf).

JamesH65, 2019-07-30:
@TheEnbyperor Any luck with the fix? If so, please close the issue.