raspberrypi/linux #3087: linux-raspberrypi-4.19.58-1 causes kernel panic on raspberry pi 2 (arch linux arm)

nightBulb, 2019-07-18:
**Describe the bug**
Kernel Panic on boot.

**To reproduce**
- Update packages using `pacman -Syu` (updating linux-raspberrypi package to 4.19.58-1)
- Reboot
- Kernel Panic

**Expected behaviour**
System to boot / not result in Kernel Panic

**Actual behaviour**
Kernel Panic

**System**
[raspinfo.txt](https://github.com/raspberrypi/linux/files/3408615/raspinfo.txt)

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**

Current workaround is to 
- restore `/boot` from backup to `/boot` on memorycard manually
- boot system with several systemd services failing (no modules for old kernel)
- install old kernel `pacman -U /var/cache/pacman/pkg/linux-raspberrypi-4.19.57-1-armv7h.pkg.tar.xz`
- reboot
- normal functional system

nightBulb, 2019-07-18:
The issue might be related to #3084

lategoodbye, 2019-07-19:
This needs more information:
- What kernel panic does occur (please a dump of the console or at least a photo)
- Where is your rootfs located sd card or USB device?

nightBulb, 2019-07-20:
The rootfs is located on USB Flash Drive,

(Memory Cards (any) were very unreliable for root in case of power failure)

I'll try to get photo of kernel panic.

nightBulb, 2019-07-20:
The following is the image of boot log, just before kernel panic

[image]

nightBulb, 2019-07-20:
@lategoodbye The above data is as requested (hopefully),

I currently have held back update:

`linux-raspberrypi: ignoring package upgrade (4.19.57-1 => 4.19.58-1)`

Because its also the primary server at home, and,
recovering from kernel panics is a lengthy, disruptive process,

however, I can try to intentionally cause kernel panic by updating software,
if needed.

johnny-cash, 2019-07-21:
I have a raspberry 3b and the same thing happens with the kernel 4.19.58-1 in arch linux arm.

I have the system / on a usb disk and it gives me kernel panic. Also, if I install the whole system on an SD card and try to mount USB devices with this version of the kernel, it gives me print_req_error and I / O errors.

dkadioglu, 2019-07-21:
I have a Raspberry Pi 3B+ with Arch Linux on a SD Card. With kernel 4.19.58 my Sundtek USB Tuner is not working anymore. With 4.19.57 everything works as expected. Maybe there is something bad in general with 4.19.58 and USB devices? If you need further info, please ask.

lategoodbye, 2019-07-21:
@nightBulb Thanks for the photo. It seems that the USB device isn't accessible (-110 = ETIMEDOUT). 

But a trace of the real kernel panic would be still necessary.

nightBulb, 2019-07-21:
@lategoodbye This is the kernel panic stack trace, (hopefully as needed)

[image]

Also I discovered that linux-raspberrypi-4.19.57-2 also causes, what appears to be the same problem.
(Kernel panic and all ... )

linux-raspberrypi-4.19.57-1 seems to be the last sane kernel. 

And so, changing the title seems appropriate.

lategoodbye, 2019-07-21:
@nightBulb Sorry for being inprecise, we need the beginning of the kernel panic.

Appart from that your issue looks like #3067

nightBulb, 2019-07-21:
@lategoodbye Is there any place where the Bootlog is stored,
cause the beginning part of kernel panic scrolls up quickly on screen  ...

lategoodbye, 2019-07-21:
Without rootfs there is chance to store the file.

The best solution is to connect serial TTL adapter to the debug UART.

In case USB still working you could try to enlarge the kernel log buffer by adding `log_buf_len=1M` to cmdline.txt

nightBulb, 2019-07-21:
@lategoodbye I meant, how do you scroll screen after kernel panic?

I do not have  a TTL adapter either.

One thing to note though, If I remove the USB rootfs, the bootloader (or whatever it is that boots rpi) drops me to ash terminal.

after I re-inser rootfs USB and call init, it panics, and I can no longer run any commands even from bootloader ash 😁

lategoodbye, 2019-07-21:
Sometimes it's possible to scroll back via page up/down. But this depends on the crash.

nightBulb, 2019-07-22:
The (shift +) Page up/down keys do not seem to be working especially after kernel panic,

[image]

I even tried the following,

- Remove USB Flash Drive Containing root
- reboot
- boot process drops to `ash`
- Re-insert USB Flash Drive containing root
- run  `./init` script in `rootfs` terminal (ash)
- wait for `reset high speed USB device using dwc_otg` error/kernel-message to occur
- exit
- kernel panic

No method to scroll the screen.

pelwell, 2019-07-22:
Adding `framebuffer_height=2160` to config.txt will get twice as many lines on screen.

dave-p, 2019-07-22:
Here is a console log showing the issue. It is from a Pi2 v1.1 running Arch Linux, with the /boot directory on SD and the OS on an SSD connected via USB - the Toshiba HDD mentioned in the log is just the usb-sata interface.

I've had a similar issue with another Pi2 v1.2 which is used as a TVHeadend server; the OS is on SD but the videos are on a USB HDD. With this device there is no crash but playback of video stops after a while and the disk is inaccessible until the Pi is rebooted.

The latest software I know is working is Linux 4.19.55 and bootloader/firmware 20190624.

[minicom.txt](https://github.com/raspberrypi/linux/files/3417538/minicom.txt)

nightBulb, 2019-07-22:
I too managed to get the picture of kernel panic using @pelwell  's idea.

I also have more pictures of the same kernel panic, if this one is not clear, I can upload them if needed ..

[image]

tsaarni, 2019-07-22:
I'm seeing the same bug on Raspberry Pi 3 Model B, booting from external USB SSD.
[kernel-4.19.58-1-ARCH-boot-from-usb-device-fails.txt](https://github.com/raspberrypi/linux/files/3418215/kernel-4.19.58-1-ARCH-boot-from-usb-device-fails.txt)

pelwell, 2019-07-22:
The common factors here appear to be kernel version and Arch Linux - it's not a problem we've seen on Raspbian, and it would be enlightening if one of you would install Raspbian on a spare card to confirm (or not) that it boots OK.

dave-p, 2019-07-22:
OK I installed the current Raspbian Buster Lite onto an RPi3 (kernel 4.19.57-V7+), then configured it to boot from SD but use an SSD as the root FS. I then ran a dist-upgrade which updated the kernel to 4.19.58. The system booted as normal. I moved the SD card and SSD to the same RPi2 as in my previous test and it continued to boot correctly. It's not quite the same test (the SSD and usb-sata interface were different) but it does suggest an Arch problem.

Arch forums have one post with the same issue:
https://archlinuxarm.org/forum/viewtopic.php?f=60&t=13792

[minicom2.txt](https://github.com/raspberrypi/linux/files/3418916/minicom2.txt)

tsaarni, 2019-07-22:
I have been iterating over the kernel configuration parameters that Arch kernel has changed recently. See diff [here](https://github.com/archlinuxarm/PKGBUILDs/commit/ad0cf064936a9b9057c21cc6b7f39a57d5f7ce9f#diff-f2360ac15be17cca1379bbe5df910a75).   I'm not 100% sure I'm on a right track, but it seems to me that CONFIG_VMSPLIT_3G=y (instead of VMSPLIT_2G) is the change that triggers the bug for me on RPI3.  

Can experts here deny or confirm that this could cause USB boot to not work?

dave-p, 2019-07-23:
I've built a new stock Foundation kernel (which is now at version 4.19.59) and it boots my Arch RPi3 with rootfs on SSD. This does seem to be an Arch kernel config issue.

pelwell, 2019-07-23:
The Pi2 bringup was a long time ago now, but from what I recall we switched to VMSPLIT_2G to avoid the need for HIGHMEM on a 1GB part. Does the Arch config have HIGHMEM enabled?

dave-p, 2019-07-23:
From https://archlinuxarm.org/packages/armv7h/linux-raspberrypi/files/config.v7 (also config.v6)
CONFIG_HIGHMEM=y

tsaarni, 2019-07-23:
Changing to CONFIG_HIGHMEM=n seems to fix the problem too.

lategoodbye, 2019-07-23:
> Changing to CONFIG_HIGHMEM=n seems to fix the problem too.

This is not a fix, it's only a workaround.

tsaarni, 2019-07-23:
Yes I agree, sorry for bad wording.

tsaarni, 2019-07-23:
This is not my area at all, but just in case this is relevant:  Arch config enables also CONFIG_ARM_LPAE and CONFIG_ARCH_DMA_ADDR_T_64BIT which results in `typedef u64 dma_addr_t`.  Interestingly, during compilation I see some warnings from `dwc_otg` driver e.g.

```
drivers/usb/host/dwc_otg/dwc_otg_fiq_fsm.c: In function ‘fiq_increment_dma_buf’:
drivers/usb/host/dwc_otg/dwc_otg_fiq_fsm.c:243:30: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
  struct fiq_dma_blob *blob = (struct fiq_dma_blob *) st->dma_base;
                              ^
drivers/usb/host/dwc_otg/dwc_otg_fiq_fsm.c:252:14: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
  hcdma.d32 = (dma_addr_t) &blob->channel[n].index[i].buf[0];
              ^
drivers/usb/host/dwc_otg/dwc_otg_fiq_fsm.c: In function ‘fiq_iso_out_advance’:
drivers/usb/host/dwc_otg/dwc_otg_fiq_fsm.c:292:30: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
  struct fiq_dma_blob *blob = (struct fiq_dma_blob *) st->dma_base;
                              ^
drivers/usb/host/dwc_otg/dwc_otg_fiq_fsm.c:304:14: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
  hcdma.d32 = (dma_addr_t) blob->channel[n].index[i].buf;
              ^
```

lategoodbye, 2019-07-24:
@tsaarni The motivation behind the config changes was the introduction of the Raspberry Pi 4. So a temporary workaround would a separate config for BCM2711. But it's possible that the Arch guys don't want to maintain another image.

nightBulb, 2019-07-25:
I just checked, 
the problem exists in Linux 4.19.59-1-Arch as well,

the following is the kernel panic for linux-raspberrypi-4.19.59-1

[image]

tsaarni, 2019-07-26:
@kmihelich what is your opinion about the situation? There might not be fix any time soon. Would it be reasonable to introduce another Arch kernel package with old config as a workaround for those that need to boot from USB devices.

zertyz, 2019-07-26:
I'm facing the same issue on my raspberry Pi 2 with archlinuxarm & kernel 4.19.59, after upgrading from before rPi4 was taken into consideration on the arch packages. I'm booting from an raid array of 4 USB sticks + sd card.

In my case, I am able to boot (without those freaking USB errors) when I use 256m of GPU ram (or more). Decreasing that a little (192, 160, 128...) sometimes issues a failure into initializing a random CPU (only 3 raspberries shows on my debug monitor and the boot process hangs for more than I can wait trying to wake the cpu up).

When downgrading my system to the versions before the update, everything works fine.

GazzaC, 2019-07-27:
> I'm facing the same issue on my raspberry Pi 2 with archlinuxarm & kernel 4.19.59, after upgrading from before rPi4 was taken into consideration on the arch packages. I'm booting from a raid array of 4 USB sticks + sd card.
> 
> In my case, **I am able to boot** (without those freaking USB errors) **when I use 256m of GPU ram** (or more). Decreasing that a little (192, 160, 128...) sometimes issues a failure into initializing a random CPU (only 3 raspberries shows on my debug monitor and the boot process hangs for more than I can wait trying to wake the cpu up).
> 
> When downgrading my system to the versions before the update (don't remember which), everything works fine.

Set my GPU to 256m, its a workaround.

nightBulb, 2019-07-27:
I can confirm that,
setting `gpu_mem=256` in `/boot/config.txt` allows the system to boot normally with updated kernel 
:+1: :+1:  ,

(I tested with `4.19.59-1`, which is the latest at the time of testing)

but, that leaves around `742MB` ( **998M (total mem) - 256M (gpu mem)** ) of memory for the remaining OS & programs :-1:  :-1: ,
which leaves around **500M** of usable RAM for user programs.

running RPi headless (for example, as a home server) with 256M GPU is a serious waste of (already constrained) RAM.

GazzaC, 2019-07-27:
> I can confirm that,
> setting `gpu_mem=256` in `/boot/config.txt` allows the system to boot normally with updated kernel
> +1 +1 ,
> 
> (I tested with `4.19.59-1`, which is the latest at the time of testing)
> 
> but, that leaves around `742MB` ( **998M (total mem) - 256M (gpu mem)** ) of memory for the remaining OS & programs -1 -1 ,
> which leaves around **500M** of usable RAM for user programs.
> 
> running RPi headless (for example, as a home server) with 256M for GPU is a serious waste of (already constrained) RAM.

We definitely need a fix for this one.

JamesH65, 2019-07-28:
Interesting. That should give us enough information to track it down, clearly something is trying to allocate a LOT of memory on startup and failing. Framebuffers? Perhaps try max_framebuffers=0 if running headless?

pelwell, 2019-07-28:
It will be something to do with HIGHMEM - with VMSPLIT_3G the kernel doesn't have enough address range to get to all memory. We have a configuration for Pi 2/3 that works, and the Arch devs appear to have tried to make all devices share (at least some of) the Pi4 configuration and broken it.

semeion, 2019-07-30:
Post in archlinux-arm, related with this issue:
https://archlinuxarm.org/forum/viewtopic.php?f=60&t=13821

semeion, 2019-08-03:
Someone know if the issue got fixed? 
The changelog of the new kernel:

https://archlinuxarm.org/packages/armv7h/linux-raspberrypi/log?id=991876d8084a43c69a82aed88d0c03aeae8bea12

pelwell, 2019-08-03:
Not by that change list, no.

erikschreier, 2019-08-04:
I can confirm the gpu_mem=256 line avoids the Kernelpanic but when updating the System using pacman -Syu there will be a new problem. After reboot, the Raspberry Pi stops booting completely and the Screen stays black, green LED Appears for for about 5 Seconds and then blinking a few times.

The same Problem appears when Flashing Raspbian Buster on an USB Drive and also when updating Stretch to the new Firmware.

Device: Raspberry Pi 3 B+ (1.3 UK)
Drive: Sandisk Ultra 64Gb / Intenso Aluline 16 GB

All Imges run with no Problems when flashed to SD Cards. Logs are not possible cause the Pi wont even boot to Rainboscreen so sorry.

The Raspberry Pi 3 B Boots the drive with no Problem so i think theres some problem in the new firmware or Bootloader of the Raspberry Pi Files of the 3 B+.

nightBulb, 2019-08-04:
Update: `linux-raspberrypi-4.19.63-1` also causes kernel panic :-1: :-1: .

with similar 
`reset high speed usb device using dwc_otg`
`Error Read 64/`... error

nightBulb, 2019-08-04:
Just updated main comment of this issue with additional context and links to Arch Linux ARM forums

nightBulb, 2019-08-05:
Can anyone confirm @erikschreier's  [observation](https://github.com/raspberrypi/linux/issues/3087#issuecomment-518026139) that the issue affects Raspbian as well ?

And problem with `gpu_mem=256` workaround.

nightBulb, 2019-08-11:
Just an update linux-raspberrypi-**4.19.65-1**-armv7h still has( / causes...???)  kernel panic.
@kmihelich @pelwell 

Is there any other solution / workaround besides wasting 256 MB RAM on 997 MB RAM device
OR
staying at an older kernel version (probably a security hazard...)

pelwell, 2019-08-11:
The solution is the same it has always been - her the Arch devs to unbreak their Pi 2 configuration. Or run Raspbian.

semeion, 2019-08-11:
I still waiting for a solution... I hope it come on next days.

pelwell, 2019-08-11:
Can you not use one of our kernels with the Arch userland?

zertyz, 2019-08-11:
> The solution is the same it has always been - her the Arch devs to unbreak their Pi 2 configuration. Or run Raspbian.

The problem seems not to be related to archlinuxarm's Pi 2 configuration, since it also happens on Raspbian, as reported some posts ago and verified by me, on my Pi 2 -- Booting raspbian from MMC, but with USB drives attached is enough to reveal the problem -- even if the USB drives are not involved in the boot process.

Maybe the title for this issue should be reconsidered and the Archlinuxarm part removed?

semeion, 2019-08-11:
@zertyz, the issue with USB happens with raspbian too? Relevant info!
I am using archlinux-arm in a rpi3b, i don´t have issue with kernel panic, but my USB ports can´t access drives anymore, if i try i get lot of errors.

tsaarni, 2019-08-11:
We have already troubleshooted this problem here.  It is related to the USB driver and it is triggered by kernel config settings CONFIG_VMSPLIT_3G=y and CONFIG_HIGHMEM=y.  The problem appeared when these settings were taken into use in Arch raspberry kernel by @kmihelich who we have not reached for comments (see diff [here](https://github.com/archlinuxarm/PKGBUILDs/commit/ad0cf064936a9b9057c21cc6b7f39a57d5f7ce9f#diff-f2360ac15be17cca1379bbe5df910a75)).

Nobody seems to be working with HIGHMEM support for the USB driver.  Therefore, like suggested here before, the only feasible way forward seems to be either to revert the change in Arch raspberry kernel package, or by moving to different Linux distro.

esiqveland, 2019-08-12:
I came here after seeing this with archlinuxarm and kernel 4.19.65.
Booting with `gpu_mem=256` makes kernel able to boot.

I found these older issues that might be related with regards to `CONFIG_HIGHMEM=y` and `CONFIG_VMSPLIT_3G=y`:

https://github.com/raspberrypi/linux/issues/1641

And some even older:

https://github.com/raspberrypi/linux/issues/1394

graysky2, 2019-08-13:
> The solution is the same it has always been - her the Arch devs to unbreak their Pi 2 configuration. Or run Raspbian.

@pelwell - Check your email for our exchange around 12-July.  My memory is that in order to get the RPi4 to boot, the changes to the config file that @tsaarni linked were needed.

From my notes, they were these 3:
1) "System type>Multiple platform selection" to disable "ARMv6 based platforms (ARM11)"
2) Enable LPAE
3) Select a 1G/3G split

I cannot remember if the HIGHMEM option is selected in doing this or not.

pelwell, 2019-08-13:
I'm not sure what you are saying. The issue is about why current Arch images don't run on Pi 2, and you've just described what we had to do on Pi 4. Note that the Pi 4 uses a completely different USB controller, so the problems reported here with the dwc_otg driver and HIGHMEM don't apply.

graysky2, 2019-08-13:
@pelwell - I thought the hypothesis is that the problem for RPi2 was introduced when Arch ARM merged those support options including HIGHMEM to allow for RPi4 support.

> We have already troubleshooted this problem here. It is related to the USB driver and it is triggered by kernel config settings CONFIG_VMSPLIT_3G=y and CONFIG_HIGHMEM=y. The problem appeared when [...]

pelwell, 2019-08-13:
> I thought the hypothesis is that the problem for RPi2 was introduced when Arch ARM merged those support options including HIGHMEM to allow for RPi4 support.

Yes, that is the hypothesis. I still don't get your point - you just seem to be restating the problem.

graysky2, 2019-08-13:
> Yes, that is the hypothesis. I still don't get your point - you just seem to be restating the problem.

If we disable HIGHMEM will there by any ill-effects for Pi4 boards?  If not, would it be a good test to have someone with Arch ARM and an affected RPi2 compile the current kernel using the following patch which disables HIGHMEM and see if it fixes the USB issue?

https://gist.github.com/graysky2/5dee027e153fadc98659a85f32aeddbc

pelwell, 2019-08-13:
> If we disable HIGHMEM will there by any ill-effects for Pi4 boards?

Yes - the kernel won't be able to address all memory, and it may even crash - I don't remember the precise failure mechanism.

graysky2, 2019-08-13:
@pelwell - OK.  I still feel like having an affected RPi2 user try compiling/booting with that config patched disablign HIGHMEM will be a test that it is causing the problem.  If it does, Arch ARM might need to consider another kernel package for RPi2 with HIGHMEM disabled, no?

[17 more replies]

closed 2019-08-19 by popcornmix