raspberrypi/linux #3092: mmc0: timeout waiting for hardware interrupt (without SD card)

burtyb, 2019-07-20:
I'm not sure if this is the same problem as #2802 . I'm not using an SD card in the Pi Zero (booting with rpiboot, USB Gadget Ethernet and NFS root) but I'm seeing the timeout. The Pi Zero had just been set idle after booting Buster (Lite) over USB.

System Information
------------------

Raspberry Pi Zero Rev 1.3
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage2

Linux p1 4.19.57+ #1244 Thu Jul 4 18:42:50 BST 2019 armv6l GNU/Linux
Hardware        : BCM2835
Revision        : 900093
Serial          : 000000000bd39641
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Jul  9 2019 14:40:53
Copyright (c) 2012 Broadcom
version 6c3fe3f096a93de3b34252ad98cdccadeb534be2 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem                                 1K-blocks    Used Available Use% Mounted on
udev                                          214596       0    214596   0% /dev
tmpfs                                          44312    1980     42332   5% /run
172.19.180.254:/var/lib/clusterctrl/nfs/p1  30442880 7979136  21194624  28% /
tmpfs                                         221552       0    221552   0% /dev/shm
tmpfs                                           5120       4      5116   1% /run/lock
tmpfs                                         221552       0    221552   0% /sys/fs/cgroup
tmpfs                                          44308       0     44308   0% /run/user/1000

Filename                                Type            Size    Used    Priority

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: (none)
raspberrypi-sys-mods:
  Installed: 20190429
openbox:
  Installed: (none)
lxpanel:
  Installed: (none)
pcmanfm:
  Installed: (none)
rpd-plym-splash:
  Installed: (none)

Networking Information
----------------------

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet x.x.x.x  netmask x.x.x.x
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
[... 690 lines cut ...]
[33807.299884] 9f60: d94cb900 c003d498 d8ed7f58 d952de88 d8e69fac d8e69f80 c0042fd0 c003d4a4
[33807.313506] 9f80: ffffffff d934c040 c0042eb4 00000000 00000000 00000000 00000000 00000000
[33807.327111] 9fa0: 00000000 d8e69fb0 c00090ac c0042ec0 00000000 00000000 00000000 00000000
[33807.340769] 9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[33807.354426] 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[33807.368127] [<c054aa4c>] (bcm2835_sdhost_finish_command) from [<c054c0bc>] (bcm2835_sdhost_request+0x4b4/0x684)
[33807.383747] [<c054c0bc>] (bcm2835_sdhost_request) from [<c0529b44>] (__mmc_start_request+0x78/0x164)
[33807.398387] [<c0529b44>] (__mmc_start_request) from [<c052a8cc>] (mmc_start_request+0x94/0xc4)
[33807.412471] [<c052a8cc>] (mmc_start_request) from [<c052a964>] (mmc_wait_for_req+0x68/0xe0)
[33807.426306] [<c052a964>] (mmc_wait_for_req) from [<c052aa54>] (mmc_wait_for_cmd+0x78/0xb4)
[33807.440109] [<c052aa54>] (mmc_wait_for_cmd) from [<c0534a78>] (mmc_app_cmd+0xbc/0x110)
[33807.453561] [<c0534a78>] (mmc_app_cmd) from [<c0534b58>] (mmc_wait_for_app_cmd+0x8c/0xfc)
[33807.467277] [<c0534b58>] (mmc_wait_for_app_cmd) from [<c0534d34>] (mmc_send_app_op_cond+0xb8/0x148)
[33807.481828] [<c0534d34>] (mmc_send_app_op_cond) from [<c0534888>] (mmc_attach_sd+0x44/0x178)
[33807.495736] [<c0534888>] (mmc_attach_sd) from [<c052cd60>] (mmc_rescan+0x448/0x498)
[33807.508881] [<c052cd60>] (mmc_rescan) from [<c003d228>] (process_one_work+0x130/0x3a0)
[33807.522339] [<c003d228>] (process_one_work) from [<c003d4cc>] (worker_thread+0x34/0x530)
[33807.535971] [<c003d4cc>] (worker_thread) from [<c0042fd0>] (kthread+0x11c/0x158)
[33807.548891] [<c0042fd0>] (kthread) from [<c00090ac>] (ret_from_fork+0x14/0x28)
[33807.558892] Exception stack(0xd8e69fb0 to 0xd8e69ff8)
[33807.566709] 9fa0:                                     00000000 00000000 00000000 00000000
[33807.580217] 9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[33807.593664] 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[33807.602999] Code: e5d530a4 e3130010 1a0000e9 e3e0306d (e5803020)
[33807.611770] ---[ end trace 27dee134a46220fb ]---

lategoodbye, 2019-07-25:
How reproducible is this (e.g. always within 1 day)?

burtyb, 2019-07-28:
Yes, it's reproducible within a day with different hardware models.

Raspberry Pi Zero W Rev 1.1
>[13775.134657] mmc0: timeout waiting for hardware interrupt.
[13775.134681] [36e1119e] CMD  37 0
[13775.134689] [36e1119e] REQ> d9c4fd38 0
[13775.134695] [36e111ac] TSK< d9c4fd38 0
[13775.134700] [36e111b4] TSK> d9c4fd38 0
[13775.134706] [36e111c4] IOS< 61a80 0
[13775.134711] [36e111cb] REQ< d9c4fe00 10800
[13775.134717] [36e111cd] CMD< 1 0
[13775.134723] [36e111d1] FCM< d9c4fe00 d9c4fe60
[13775.134728] [36e1150a] CMD  1 0
[13775.134734] [36e1150b] REQ> d9c4fe00 0
[13775.134739] [36e11518] TSK< d9c4fe00 0
[13775.134745] [36e11521] TSK> d9c4fe00 0
[13775.134750] [36e11531] IOS< 0 0
[13775.134755] [36e11c62] IOS< 0 0
[13775.134761] [36e14f84] IOS< 493e0 0
[13775.134766] [36e182aa] RST< 0 0
[13775.134771] [36e1d0ce] REQ< d9c4fe08 10800
[13775.134777] [36e1d0d1] CMD< 34 c00
[13775.134784] [36e1d0d7] FCM< d9c4fe08 d9c4fe68
[13775.134789] [36e1daad] CMD  34 0
[13775.134795] [36e1daad] REQ> d9c4fe08 0
[13775.134800] [36e1dabf] TSK< d9c4fe08 0
[13775.134806] [36e1dac8] TSK> d9c4fe08 0
[13775.134811] [36e1dadd] REQ< d9c4fe08 10800
[13775.134817] [36e1dadd] CMD< 34 80000c08
[13775.134822] [36e1dae2] FCM< d9c4fe08 d9c4fe68
[13775.134828] [36e1e4c0] CMD  34 0
[13775.134833] [36e1e4c0] REQ> d9c4fe08 0
[13775.134839] [36e1e4ce] TSK< d9c4fe08 0
[13775.134844] [36e1e4d6] TSK> d9c4fe08 0
[13775.134850] [36e1e4e6] IOS< 493e0 0
[13775.134855] [36e1ec21] REQ< d9c4fe30 10800
[13775.134860] [36e1ec22] CMD< 0 0
[13775.134866] [36e1ec28] FCM< d9c4fe30 d9c4fe90
[13775.134871] [36e1ed02] FCM> d9c4fe30 0
[13775.134877] [36e1ed02] CMD  0 0
[13775.134882] [36e1ed03] REQ> d9c4fe30 0
[13775.134887] [36e1ed12] TSK< d9c4fe30 0
[13775.134893] [36e1ed19] TSK> d9c4fe30 0
[13775.134898] [36e1f245] IOS< 493e0 0
[13775.134904] [36e1f759] REQ< d9c4fe30 10800
[13775.134910] [36e1f75a] CMD< 8 1aa
[13775.134916] [36e1f75e] FCM< d9c4fe30 d9c4fe90
[13775.134921] [36e1fb7f] CMD  8 0
[13775.134927] [36e1fb7f] REQ> d9c4fe30 0
[13775.134932] [36e1fb8e] TSK< d9c4fe30 0
[13775.134937] [36e1fb95] TSK> d9c4fe30 0
[13775.134943] [36e1fbaa] REQ< d9c4fdd8 10800
[13775.134948] [36e1fbab] CMD< 5 0
[13775.134954] [36e1fbaf] FCM< d9c4fdd8 d9c4fe38
[13775.134959] [36e1ffd9] CMD  5 0
[13775.134964] [36e1ffda] REQ> d9c4fdd8 0
[13775.134970] [36e1ffe7] TSK< d9c4fdd8 0
[13775.134976] [36e1ffec] TSK> d9c4fdd8 0
[13775.134981] [36e1fffc] REQ< d9c4fdd8 10800
[13775.134987] [36e1fffd] CMD< 5 0
[13775.134992] [36e20003] FCM< d9c4fdd8 d9c4fe38
[13775.134998] [36e20409] CMD  5 0
[13775.135003] [36e2040a] REQ> d9c4fdd8 0
[13775.135009] [36e20416] TSK< d9c4fdd8 0
[13775.135014] [36e2041b] TSK> d9c4fdd8 0
[13775.135020] [36e20428] REQ< d9c4fdd8 10800
[13775.135025] [36e20429] CMD< 5 0
[13775.135030] [36e2042d] FCM< d9c4fdd8 d9c4fe38
[13775.135036] [36e20839] CMD  5 0
[13775.135041] [36e20839] REQ> d9c4fdd8 0
[... 585 lines cut ...]
[26665.885559] bf20: d98f4b54 c09eb7ec c09f38c0 00000008 d8dbb098 d98f4b40 d9c3bf7c d9c3bf48
[26665.899217] bf40: c003d4e0 c003d118 d9c3bf7c d9c3bf58 c0042858 d8dbb080 d8dc76c0 00000000
[26665.912872] bf60: d98f4b40 c003d4ac d8dbb098 da953e88 d9c3bfac d9c3bf80 c0042fe4 c003d4b8
[26665.926521] bf80: ffffffff d8dc76c0 c0042ec8 00000000 00000000 00000000 00000000 00000000
[26665.940132] bfa0: 00000000 d9c3bfb0 c00090ac c0042ed4 00000000 00000000 00000000 00000000
[26665.953729] bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[26665.967380] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[26665.981069] [<c054aabc>] (bcm2835_sdhost_finish_command) from [<c054c12c>] (bcm2835_sdhost_request+0x4b4/0x684)
[26665.996706] [<c054c12c>] (bcm2835_sdhost_request) from [<c0529bb4>] (__mmc_start_request+0x78/0x164)
[26666.011376] [<c0529bb4>] (__mmc_start_request) from [<c0529d24>] (mmc_wait_for_req_done+0x84/0x114)
[26666.025930] [<c0529d24>] (mmc_wait_for_req_done) from [<c052aa10>] (mmc_wait_for_req+0xa4/0xe0)
[26666.040101] [<c052aa10>] (mmc_wait_for_req) from [<c052aac4>] (mmc_wait_for_cmd+0x78/0xb4)
[26666.053833] [<c052aac4>] (mmc_wait_for_cmd) from [<c0536fb4>] (mmc_send_io_op_cond+0xa8/0x12c)
[26666.067974] [<c0536fb4>] (mmc_send_io_op_cond) from [<c0536a44>] (mmc_attach_sdio+0x44/0x398)
[26666.082023] [<c0536a44>] (mmc_attach_sdio) from [<c052cdb4>] (mmc_rescan+0x42c/0x498)
[26666.095379] [<c052cdb4>] (mmc_rescan) from [<c003d23c>] (process_one_work+0x130/0x3a0)
[26666.108804] [<c003d23c>] (process_one_work) from [<c003d4e0>] (worker_thread+0x34/0x530)
[26666.122354] [<c003d4e0>] (worker_thread) from [<c0042fe4>] (kthread+0x11c/0x158)
[26666.135212] [<c0042fe4>] (kthread) from [<c00090ac>] (ret_from_fork+0x14/0x28)
[26666.145262] Exception stack(0xd9c3bfb0 to 0xd9c3bff8)
[26666.153058] bfa0:                                     00000000 00000000 00000000 00000000
[26666.166609] bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[26666.180128] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000
[26666.189446] Code: e5d530a4 e3130010 1a0000e9 e3e0306d (e5803020)
[26666.198212] ---[ end trace 744dafd43af08788 ]---

lategoodbye, 2019-07-28:
Did it ever worked without this warning (regression)?
Is the system still usable in this state?

burtyb, 2019-07-30:
I don't remember seeing it previously but that was with a very old kernel/firmware, it'll take some time to work out where it started.

The Pi Zero LED is no longer on (which is an easy way to spot which one had the oops) and obviously the SD no longer works if you put one in.

lategoodbye, 2019-07-30:
Only to get this right:
- the OTP of the Raspberry Pi Zero hasn't be changed
- the Raspberry Pi Zero is connected via OTG to a PC
- the PC runs rpiboot, DHCP server and a NFS root (buster image)

Skywalker-11, 2019-08-26:
Not sure if it is related or if I just have a bad sd card but I get similar errors on a Pi 2 Model B with Debian 10. From my syslog and apt logs it seems like it appeared after the update to kernel1.20190819-1 or one of these: `libraspberrypi-bin:armhf (1.20190718-1, 1.20190819-1), libraspberrypi-dev:armhf (1.20190718-1, 1.20190819-1), libraspberrypi-doc:armhf (1.20190718-1, 1.20190819-1), raspberrypi-kernel:armhf (1.20190718-1, 1.20190819-1), raspberrypi-bootloader:armhf (1.20190718-1, 1.20190819-1), libraspberrypi0:armhf (1.20190718-1, 1.20190819-1)`
```
 kernel: mmc0: timeout waiting for hardware interrupt.
 kernel: [86f70f33] PRD2 2 0
 kernel: [86f70f3c] PRD3 b699e984 0
 kernel: [86f70f3d] PDM> b9f261a0 0
 kernel: [86f70f3e] REQ< b9f260c0 10801
 kernel: [86f70f3f] CMD< 12 15faa48
 kernel: [86f70f44] CMDD d0 200
 kernel: [86f70f45] SDMA b9f261a0 ba677640
 kernel: [86f70f48] FCM< b9f260c0 b9f26138
 kernel: [86f70f49] RSP  900 0
 kernel: [86f70f49] FCM> b9f260c0 0
 kernel: [86f70f4a] CMD  12 200
.....
 kernel: mmc0: timeout waiting for hardware interrupt.
 kernel: [c7991d9f] FCM< b91efc00 b91efc68
 kernel: [c7991da1] RSP  900 0
 kernel: [c7991dad] FCM> b91efc00 0
 kernel: [c7991dae] CMD  d 0
 kernel: [c7991dae] REQ> b91efc00 0
 kernel: [c7991dc3] TSK< b91efc00 0
 kernel: [c7991dd1] TSK> b91efc00 0
 kernel: [c7991de1] PRD< b9f2a9a0 0
 kernel: [c7991de2] PRD1 ba677640 0
```

roro-86, 2019-11-28:
I have the same issue on a raspberry pi 4 4gb. It network boots and all file systems are mounted over NFS but it still complains on the HW interrupt for mmc0

```
Model           : Raspberry Pi 4 Model B Rev 1.1
```

```
[54903.847095] mmc0: Timeout waiting for hardware cmd interrupt.
[54903.847106] mmc0: sdhci: ============ SDHCI REGISTER DUMP ===========
[54903.847118] mmc0: sdhci: Sys addr:  0x00000000 | Version:  0x00001002
[54903.847128] mmc0: sdhci: Blk size:  0x00000000 | Blk cnt:  0x00000000
[54903.847137] mmc0: sdhci: Argument:  0x00000000 | Trn mode: 0x00000000
[54903.847147] mmc0: sdhci: Present:   0x1fff0001 | Host ctl: 0x00000001
[54903.847156] mmc0: sdhci: Power:     0x0000000f | Blk gap:  0x00000080
[54903.847165] mmc0: sdhci: Wake-up:   0x00000000 | Clock:    0x0000f447
[54903.847174] mmc0: sdhci: Timeout:   0x00000000 | Int stat: 0x00000000
[54903.847183] mmc0: sdhci: Int enab:  0x00ff1003 | Sig enab: 0x00ff1003
[54903.847192] mmc0: sdhci: ACmd stat: 0x00000000 | Slot int: 0x00000000
[54903.847201] mmc0: sdhci: Caps:      0x45ee6432 | Caps_1:   0x0000a525
[54903.847211] mmc0: sdhci: Cmd:       0x0000371a | Max curr: 0x00080008
[54903.847219] mmc0: sdhci: Resp[0]:   0x00000000 | Resp[1]:  0x00000000
[54903.847228] mmc0: sdhci: Resp[2]:   0x00000000 | Resp[3]:  0x00000000
[54903.847236] mmc0: sdhci: Host ctl2: 0x00000000
[54903.847245] mmc0: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0x00000000
[54903.847253] mmc0: sdhci: ============================================
```

nkfs, 2019-12-10:
Same issue here as well, also network booting and no SD card inserted. 

 kernel: mmc0: Timeout waiting for hardware cmd interrupt.
 kernel: mmc0: sdhci: ============ SDHCI REGISTER DUMP ===========
 kernel: mmc0: sdhci: Sys addr:  0x00000000 | Version:  0x00001002
 kernel: mmc0: sdhci: Blk size:  0x00000000 | Blk cnt:  0x00000000
 kernel: mmc0: sdhci: Argument:  0x00000000 | Trn mode: 0x00000000
 kernel: mmc0: sdhci: Present:   0x1fff0001 | Host ctl: 0x00000001
 kernel: mmc0: sdhci: Power:     0x0000000f | Blk gap:  0x00000080
 kernel: mmc0: sdhci: Wake-up:   0x00000000 | Clock:    0x0000f447
 kernel: mmc0: sdhci: Timeout:   0x00000000 | Int stat: 0x00000000
 kernel: mmc0: sdhci: Int enab:  0x00ff1003 | Sig enab: 0x00ff1003
 kernel: mmc0: sdhci: ACmd stat: 0x00000000 | Slot int: 0x00000000
 kernel: mmc0: sdhci: Caps:      0x45ee6432 | Caps_1:   0x0000a525
 kernel: mmc0: sdhci: Cmd:       0x00000502 | Max curr: 0x00080008
 kernel: mmc0: sdhci: Resp[0]:   0x00000000 | Resp[1]:  0x00000000
 kernel: mmc0: sdhci: Resp[2]:   0x00000000 | Resp[3]:  0x00000000
 kernel: mmc0: sdhci: Host ctl2: 0x00000000
 kernel: mmc0: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0x00000000
 kernel: mmc0: sdhci: ============================================

timg236, 2019-12-10:
This is just harmless log spam caused by the Linux SDHCI driver assuming that if there is no card detect pin then the card will be present so it tries to use it periodically. The interrupt times out and some dev_info/dev_err messages appear in the log. You could just disable those messages, it's a low priority but I'm sure we'd accept a downstream PR if someone wants to find a better fix.

burtyb, 2019-12-11:
> This is just harmless log spam

As an SD card can no longer be used once the error has occurred I wouldn't call it harmless.

timg236, 2019-12-11:
That sounds like a different issue. SD seems to work with network boot but you will get these errors without sdcards present

qxsnap, 2020-02-22:
> I have the same issue on a raspberry pi 4 4gb. It network boots and all file systems are mounted over NFS but it still complains on the HW interrupt for mmc0

@roro-86 @nkfs If you are net booting or don't need the sdcard after boot, add this to /boot/config.txt to eliminate the mmc0 errors:

`dtparam=sd_poll_once=on`

llevet, 2020-04-29:
Hi qxsnap,

Adding dtparam=sd_poll_once=on in /boot/config.txt is working for me.

nabossha, 2020-06-20:
@qxsnap thanks, solved the same issue for me when booting to SSD without SD-card on pi4 and beta-firmware

vianpl, 2020-10-20:
Hi @timg236, 
> This is just harmless log spam caused by the Linux SDHCI driver assuming that if there is no card detect pin then the card will be present so it tries to use it periodically. The interrupt times out [...]

I started looking into this as using _sd_poll_once_ isn't an option for general-purpose distros (users might want to boot over the network and hotplug SD cards). Funnily enough, the message starts showing up only after setting the core clock to 500MHz through the firmware clock driver. Be it by adding "core_freq_min=500" to config.txt or enabling the vc4 driver (which will set the core clock to at least 500MHz).

This can also be seen by applying this patch in rpi-5.9.y (the option isn't there on previous kernel versions):
```
diff --git a/drivers/clk/clk.c b/drivers/clk/clk.c
index 0a9261a099bd..81de2db0667c 100644
--- a/drivers/clk/clk.c
+++ b/drivers/clk/clk.c
@@ -3028,7 +3028,7 @@ static int clk_dump_show(struct seq_file *s, void *data)
 }
 DEFINE_SHOW_ATTRIBUTE(clk_dump);
 
-#undef CLOCK_ALLOW_WRITE_DEBUGFS
+#define CLOCK_ALLOW_WRITE_DEBUGFS  1
 #ifdef CLOCK_ALLOW_WRITE_DEBUGFS
 /*
  * This can be dangerous, therefore don't provide any real compile time
```

And playing around with `fw-clk-core's` frequency in `/sys/kernel/debug/clk/fw-clk-core/clk_rate`. The errors will show up after ramping up the frequency, and stop after lowering it. Any ideas?

timg236, 2020-10-20:
Sorry, I'm not very familiar with the sdhci-iproc driver, AFAIK there's no option to use card detect because it's muxed with WL_ON so the driver must poll, it would be nice if the driver was less chatty about errors if in 'card-detect'broken' mode

MichaIng, 2021-07-30:
Probably discussions can be merged (#3657).

A bigger problem than the error logs is probably the constant CPU load this implies: https://github.com/MichaIng/DietPi/issues/4602

Not sure whether this can be reduced somehow, by re-attempting less frequently when no SD card is present, a longer interrupt timeout or such.

Also how many users do really use SD card hot plugging? As booting from USB becomes more and more common (in my experience), probably it makes sense to poll once only by default and hide the repeated polling behind an SD card hotplug detection option.