raspberrypi/linux #3094: USB in standby then inaccessible after 2 attempts

minegameYTB, 2019-07-21:
Hello I found a bug with the USB key and the raspberry pi zero with the current kernel (4.19.57+)
the USB key becomes inactive in the few seconds that we put the USB key in the raspberry pi.

**To reproduce**
put in the USB port of the raspberry pi a USB key with an activity led (for example the USB key where we put the micro-sd card) and we wait a few seconds and we notice that the led activity the USB flash drive is off.

**Expected behaviour**
The LED of the USB key must remain lit

**Actual behaviour**
A few seconds after putting the USB key in the USB port, the USB key to the activity led off

**System**

* The model of the raspberry pi is a Pi zero W
* `cat /etc/rpi-issue`

`Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage5`

* `vcgencmd version`

`Jul  9 2019 14:40:53 
Copyright (c) 2012 Broadcom
version 6c3fe3f096a93de3b34252ad98cdccadeb534be2 (clean) (release) (start)`

* `uname -a`

`Linux rpi-zero 4.19.57+ #1244 Thu Jul 4 18:42:50 BST 2019 armv6l GNU/Linux`

pelwell, 2019-07-21:
What kennel messages does plugging in the USB device generate?
Use `sudo dmesg -C` to clear the log, then plug in the device and run `dmesg`.

minegameYTB, 2019-07-21:
Here is the answer to the `dmesg` command
`[ 4910.331622] usb 1-1: USB disconnect, device number 4
[ 4914.031586] Indeed it is in host mode hprt0 = 00021501
[ 4914.241383] usb 1-1: new high-speed USB device number 5 using dwc_otg
[ 4914.242814] Indeed it is in host mode hprt0 = 00001101
[ 4914.489633] usb 1-1: New USB device found, idVendor=0bda, idProduct=0153, bcdDevice=57.13
[ 4914.489655] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4914.489666] usb 1-1: Product: USB2.0-CRW
[ 4914.489675] usb 1-1: Manufacturer: Generic
[ 4914.489682] usb 1-1: SerialNumber: 20120926571200000
[ 4914.514149] ums-realtek 1-1:1.0: USB Mass Storage device detected
[ 4914.536849] scsi host1: usb-storage 1-1:1.0
[ 4915.594670] scsi 1:0:0:0: Direct-Access     Generic- SD/MMC/MS PRO    1.00 PQ: 0 ANSI: 4
[ 4915.610691] sd 1:0:0:0: Attached scsi generic sg0 type 0
[ 4916.412407] sd 1:0:0:0: [sdb] 30892032 512-byte logical blocks: (15.8 GB/14.7 GiB)
[ 4916.413191] sd 1:0:0:0: [sdb] Write Protect is off
[ 4916.413217] sd 1:0:0:0: [sdb] Mode Sense: 2f 00 00 00
[ 4916.414035] sd 1:0:0:0: [sdb] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[ 4916.445352]  sdb: sdb1 sdb2 < sdb5 sdb6 sdb7 >
[ 4916.463689] sd 1:0:0:0: [sdb] Attached SCSI removable disk
[ 4920.249638] EXT4-fs (sdb7): mounted filesystem with ordered data mode. Opts: (null)
[ 4923.079114] FAT-fs (sdb6): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.`

pelwell, 2019-07-21:
I don't see any errors in there, and the file system claims to be mounted.
1. Can you access the files on there?
2. Is the only issue that the LED is off?
3. What makes you think the LED should remain on? That's not my experience with card readers, SSDs etc. - they have _activtity_ LEDs, not _mounted/power_ LEDs.

minegameYTB, 2019-07-21:
when I put the usb card reader in the raspberry pi it is perfectly recognized and a few seconds after the activity led turns off.
I can still access the file but after or during the check if the files are still accessible the raspberry pi unmounts and goes up the usb card reader (always reverify if the file are accessible the card reader dismounts and does not go back until next start of the raspberry pi)

pelwell, 2019-07-21:
Please repeat the kernel log method above, but only capture the log once the drive has powered down and accesses have failed.

minegameYTB, 2019-07-21:
The result of `dmesg`
`[  939.270689] usb 1-1: USB disconnect, device number 2
[  943.220618] Indeed it is in host mode hprt0 = 00021501
[  943.430502] usb 1-1: new high-speed USB device number 3 using dwc_otg
[  943.431969] Indeed it is in host mode hprt0 = 00001101
[  943.678596] usb 1-1: New USB device found, idVendor=0bda, idProduct=0153, bcdDevice=57.13
[  943.678616] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  943.678643] usb 1-1: Product: USB2.0-CRW
[  943.678652] usb 1-1: Manufacturer: Generic
[  943.678662] usb 1-1: SerialNumber: 20120926571200000
[  943.697400] ums-realtek 1-1:1.0: USB Mass Storage device detected
[  943.727696] scsi host0: usb-storage 1-1:1.0
[  944.733201] scsi 0:0:0:0: Direct-Access     Generic- SD/MMC/MS PRO    1.00 PQ: 0 ANSI: 4
[  944.745877] sd 0:0:0:0: Attached scsi generic sg0 type 0
[  945.553467] sd 0:0:0:0: [sda] 30892032 512-byte logical blocks: (15.8 GB/14.7 GiB)
[  945.554180] sd 0:0:0:0: [sda] Write Protect is off
[  945.554205] sd 0:0:0:0: [sda] Mode Sense: 2f 00 00 00
[  945.554930] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[  945.578924]  sda: sda1 sda2 < sda5 sda6 sda7 >
[  945.595323] sd 0:0:0:0: [sda] Attached SCSI removable disk
[  948.219450] EXT4-fs (sda7): mounted filesystem with ordered data mode. Opts: (null)
[  950.160757] FAT-fs (sda6): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[ 1003.543414] Indeed it is in host mode hprt0 = 00001101
[ 1003.760902] usb 1-1: reset high-speed USB device number 3 using dwc_otg
[ 1003.761463] Indeed it is in host mode hprt0 = 00001101
[ 1003.980927] usb 1-1: device descriptor read/64, error -71
[ 1004.101361] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1004.101399] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1004.101424] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1004.101470] Indeed it is in host mode hprt0 = 00001101
[ 1004.310949] usb 1-1: device descriptor read/64, error -71
[ 1004.431169] Indeed it is in host mode hprt0 = 00001101
[ 1004.640939] usb 1-1: reset high-speed USB device number 3 using dwc_otg
[ 1004.641521] Indeed it is in host mode hprt0 = 00001101
[ 1004.850905] usb 1-1: device descriptor read/64, error -71
[ 1004.971181] Indeed it is in host mode hprt0 = 00001101
[ 1005.212781] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00
[ 1005.212840] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 01 2e e1 08 00 00 08 00
[ 1005.212860] print_req_error: I/O error, dev sda, sector 19849480
[ 1059.654892] Indeed it is in host mode hprt0 = 00001101
[ 1059.881422] usb 1-1: reset high-speed USB device number 3 using dwc_otg
[ 1059.882087] Indeed it is in host mode hprt0 = 00001101
[ 1060.101393] usb 1-1: device descriptor read/64, error -71
[ 1060.222002] Indeed it is in host mode hprt0 = 00001101
[ 1060.431441] usb 1-1: device descriptor read/64, error -71
[ 1060.551650] Indeed it is in host mode hprt0 = 00001101
[ 1060.761534] usb 1-1: reset high-speed USB device number 3 using dwc_otg
[ 1060.762119] Indeed it is in host mode hprt0 = 00001101
[ 1060.971409] usb 1-1: device descriptor read/64, error -71
[ 1061.092007] Indeed it is in host mode hprt0 = 00001101
[ 1061.301405] usb 1-1: device descriptor read/64, error -71
[ 1061.421637] Indeed it is in host mode hprt0 = 00001101
[ 1061.631534] usb 1-1: reset high-speed USB device number 3 using dwc_otg
[ 1061.631707] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1061.851519] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1061.851564] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1061.851593] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1062.071416] usb 1-1: device not accepting address 3, error -71
[ 1062.071536] Indeed it is in host mode hprt0 = 00001101
[ 1062.291475] usb 1-1: reset high-speed USB device number 3 using dwc_otg
[ 1062.731468] usb 1-1: device not accepting address 3, error -71
[ 1062.731801] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00
[ 1062.731834] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 00 2c a1 40 00 00 01 00
[ 1062.731853] print_req_error: I/O error, dev sda, sector 2924864
[ 1062.732501] usb 1-1: USB disconnect, device number 3
[ 1062.733345] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00
[ 1062.733379] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 00 2c a1 41 00 00 01 00
[ 1062.733397] print_req_error: I/O error, dev sda, sector 2924865
[ 1062.733561] print_req_error: I/O error, dev sda, sector 2924866
[ 1062.737792] print_req_error: I/O error, dev sda, sector 2924867
[... 641 lines cut ...]
[ 1064.272030] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1064.272058] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1064.272104] Indeed it is in host mode hprt0 = 00001101
[ 1065.001582] usb 1-1: device descriptor read/64, error -71
[ 1065.121687] Indeed it is in host mode hprt0 = 00001101
[ 1065.761577] usb 1-1: new high-speed USB device number 5 using dwc_otg
[ 1065.761904] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1065.761938] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1065.761964] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1065.762027] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1065.762056] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1065.762083] NYET/NAK/ACK/other in non-error case, 0x00000002
[... cut ...]
[ 1066.421784] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1066.421814] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1066.421841] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1066.421903] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1066.421931] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1066.421956] NYET/NAK/ACK/other in non-error case, 0x00000002
[ 1066.422002] Indeed it is in host mode hprt0 = 00001101
[ 1066.841558] usb 1-1: device descriptor read/64, error -71
[ 1066.961710] usb usb1-port1: attempt power cycle`

pelwell, 2019-07-21:
That's better.

What sort of power supply are you using with the Pi Zero?

Have you tried connecting the USB device through a powered hub?

minegameYTB, 2019-07-21:
I use as a power supply an external battery
and I use an otg cable for the usb and not a powered hub

pelwell, 2019-07-21:
Do you have another supply you can try? Or do you have a current rating for the battery? @P33M may have other suggestions, but I'm thinking it may be failing because it is underpowered.

minegameYTB, 2019-07-21:
I can connect it to an electric power supply

JamesH65, 2019-07-30:
@minegameYTB What were the results of using a solid power supply? This does look power supply related.

minegameYTB, 2019-07-30:
I connect the raspberry pi zero has a power supply and the problem continues, I will try with another version of the kernel with the tool rpi-update

muesli4, 2019-08-16:
Not sure whether this is related but someone from `#archlinuxarm` on `freenode` pointed me to this issue. I have a USB stick constantly plugged in. This will result in very slow boot (2 minute delay at least) and reading from the USB stick completely failing. The USB stick works on another computer.

I downgraded to kernel 4.19.57 and the issue completely disappeared.

I've added the output of `journalctl -b`: [journalctl-output.txt](https://github.com/raspberrypi/linux/files/3511140/journalctl-output.txt)

P33M, 2020-01-14:
What happens if you disable autosuspend? Add `usbcore.autosuspend=-1` to /boot/cmdline.txt (and reboot).