raspberrypi/linux #3096: Fluctuating SCLK of SPI with increase in frequency

BlinkOnceTwice, 2019-07-23:
**Describe the bug**
I was working on Raspberry Pi 3 B+ , Raspbian Stretch Desktop with package OS kernel 4.14.98-v7 and i was testing protocols and i am stuck at SPI. I am currently connecting Salae logic analyzer from a linux desktop to visualize the signal.

These are my connections

(Pin 21) MISO - MOSI (Pin 19) and Logic Analyzer 0
 
(Pin 19) MOSI - MISO (Pin 21) and Logic Analyzer 1

(Pin 23) SCLK - Logic Analyzer 2

(Pin 11) CS - Logic Analyzer 3

I am running spidev_test.c from https://github.com/torvalds/linux/blob/master/tools/spi/spidev_test.c to test my SPI.  

I get logic voltage level of 3.3 for a frequency level of 300K, but as soon as i go above, it constantly decreases. The behavior is unpredictable. Even MOSI voltage falls a bit low around 2.2-2.5 at high frequency (> 5Mhz ). Only Chip select has a stable 3.3v level. I tried with the recent Buster OS with both desktop only and desktop and package versions and i am seeing the same behavior. 
I am supplying the power using recommended canaKit. I tried powering up from a laptop also as an alternative. I am connecting to a Samsung Desktop, one mouse , one USB and a power supply. I don't know where the contention is occurring.
I tried Forum, But didn't get any reply what so ever. Please help.

**System**

* Which model of Raspberry Pi? **Pi 3 B+**
* Which OS and version (`cat /etc/rpi-issue`)? Raspbian Stretch
* Which firmware version (`vcgencmd version`)? version 2e98b31d18547962e564bdf88e573df7085c29b
* Which kernel version (`uname -a`)? 4.14.98
[image]

pelwell, 2019-07-23:
One of us is confused. You appear to be sampling the analogue voltages at about 3MHz, while the SCLK is running at about 6MHz, but you need a sampling frequency twice the signal rate to have a chance of avoid sampling errors.

What you appear to have built is a bitstream DAC: the rapidly changing digital output is being smoothed (filtered) by the sampling process (as it should), so you end up with an average. The data lines (MOSI & MISO) show this beautifully - mainly 1s -> high voltage, mainly 0s -> low voltage. With a 50% duty cycle (where the 1s and 0s are the same length) the sampled voltage will be the midpoint between the high and the low, i.e. about 1.6V. The small gaps between the bytes increases the proportion of 0s on the SCLK line, causing the average to drop down to the 1.43V you are seeing.

>  I am getting the expected data on MOSI.

It sounds like everything it working as it should.

BlinkOnceTwice, 2019-07-23:
Hey thank you for the input. It was really stupid of me to not look into the fundamental and i wasted weeks trying to find the contention in SPI bus debugging boot logs and messing with daemons and others processes. Sorry for wasting your time. 
Thank You. I am closing this issue.

closed 2019-07-23 by BlinkOnceTwice