raspberrypi/linux #3102: TL-WN725N (TP-Link WiFi adapter) regression on Buster

HankB, 2019-07-23:
I am unable to get this adapter working on Buster. It works fine with the install image for Stretch (2019-04-08-raspbian-stretch.img)

**Describe the bug**
GUI widget and raspi-config both report "No wireless interfaces found." 

**To reproduce**
Copy a Buster install image to a microSD card and boot to installation procedure with WiFi dongle installed. no WiFi adapter is discovered. 

**Expected behaviour**
with wpa_suppplicant.conf in /boot, WiFi connection should come up. (This works with Stretch but not Buster.)

**Actual behaviour**
No WiFi hardware detected.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link,
https://pastebin.com/bffePRJ8

 or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi 2B

* Which OS and version (`cat /etc/rpi-issue`)?
2019-07-10-raspbian-buster.img

* Which firmware version (`vcgencmd version`)?
```
hbarta@raspigold4gb:~ $ vcgencmd version
Jul  9 2019 14:40:53 
Copyright (c) 2012 Broadcom
version 6c3fe3f096a93de3b34252ad98cdccadeb534be2 (clean) (release) (start)
hbarta@raspigold4gb:~ $ 
```

* Which kernel version (`uname -a`)?
```
hbarta@raspigold4gb:~ $ uname -a
Linux raspigold4gb 4.19.58-v7+ #1245 SMP Fri Jul 12 17:25:51 BST 2019 armv7l GNU/Linux
hbarta@raspigold4gb:~ $ 
```

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.
dmesg:
```
[   73.712352] usb 1-1.5: new high-speed USB device number 6 using dwc_otg
[   73.843828] usb 1-1.5: New USB device found, idVendor=0bda, idProduct=8179, bcdDevice= 0.00
[   73.843858] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   73.843871] usb 1-1.5: Product: 802.11n NIC
[   73.843886] usb 1-1.5: Manufacturer: Realtek
[   73.843916] usb 1-1.5: SerialNumber: 00E04C0001
[   73.936824] lib80211: common routines for IEEE802.11 drivers
[   73.936851] lib80211_crypt: registered algorithm 'NULL'
[   73.967611] r8188eu: module is from the staging directory, the quality is unknown, you have been warned.
[   73.992533] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
[   74.029680] usbcore: registered new interface driver r8188eu
[   74.992671] MAC Address = 60:e3:27:1a:08:3e
[   74.996486] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   84.065424] logitech-hidpp-device 0003:046D:1017.0007: HID++ 1.0 device connected.
```
lsusb:
```
pi@raspberrypi:~ $ lsusb
Bus 001 Device 006: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
Bus 001 Device 005: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 004: ID 04d9:0132 Holtek Semiconductor, Inc. 
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@raspberrypi:~ $ 
```

network stuff
```
pi@raspberrypi:~ $ ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:aa:e6:2f brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.156/24 brd 192.168.1.255 scope global noprefixroute eth0
       valid_lft forever preferred_lft forever
    inet6 2601:249:e00:6b24::16a9/128 scope global dynamic noprefixroute 
       valid_lft 7038sec preferred_lft 4338sec
    inet6 2601:249:e00:6b24:722f:c402:c3eb:544c/64 scope global dynamic mngtmpaddr noprefixroute 
       valid_lft 86388sec preferred_lft 14388sec
    inet6 fe80::2d51:b30b:140a:9dd6/64 scope link 
       valid_lft forever preferred_lft forever
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 60:e3:27:1a:08:3e brd ff:ff:ff:ff:ff:ff
pi@raspberrypi:~ $ ifconfig wlan0
wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 60:e3:27:1a:08:3e  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

pi@raspberrypi:~ $ 
```
modules
```
pi@raspberrypi:~ $ lsmod|egrep "8188|80211"
r8188eu               438272  0
lib80211               16384  1 r8188eu
cfg80211              614400  1 r8188eu
rfkill                 28672  2 cfg80211
pi@raspberrypi:~ $ 
```
installed firmware
```
pi@raspberrypi:~ $ dpkg -l firmware-realtek
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name             Version           Architecture Description
+++-================-=================-============-==================================================
ii  firmware-realtek 1:20190114-1+rpt2 all          Binary firmware for Realtek wired/wifi/BT adapters
pi@raspberrypi:~ $ 
```
Note: If I insatall the micro-SD card in a Pi with on-board WiFi or a different dongle (Edimax) the WiFi operates normally.

Also ref https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=246587

Thanks!

macmpi, 2019-07-23:
You may want to blacklist `r8188eu` driver and use staging `8188eu`, or install [MrEngman's](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=62371&sid=2a7dda1bcd301769b6c465b2bf4079d6) which has been quite reliable so far (even in AP mode).

HankB, 2019-07-24:
Thank you for the prompt reply.
Can you help with two questions?
I've blacklisted the module but cannot find a giuide to using a kernel module from staging. Can you point to a guide for that?
I use a couple of these dongles with Pis that are operated headless and distant from Ethernet. Will a kernel update cause WiFi to not work again until I intervene manually to correct the situation (for either solution?) 

Apparently I also need a guide for blacklisting modules. I created the following file:
```
pi@raspberrypi:~ $ cat /etc/modules-load.d/raspi-blacklist.conf 
blacklist r8188eu

pi@raspberrypi:~ $ 
```
and on following boot the module looks like it still loaded.
```
pi@raspberrypi:~ $ lsmod |egrep "8188|80211"
r8188eu               438272  0
lib80211               16384  1 r8188eu
cfg80211              614400  1 r8188eu
rfkill                 28672  2 cfg80211
pi@raspberrypi:~ $
```

Thanks!

macmpi, 2019-07-24:
Just follow MrEngman's install procedure using the script mentioned in post: it will do all for you.

HankB, 2019-07-24:
Does MrEngman's install procedure need to be performed every time the kernel is updated? I need a solution that can be performed on a headless system only accessible via ssh.
If it is needed following each kernel upgrade, can it be run post upgrade and before rebooting? In other words, will it work for installed kernels or just the running kernel?
Thanks!

macmpi, 2019-07-24:
Did that driver provide any benefit / fix your issue?

Yes, it needs to be reinstalled after each kernel update. You may download/install the necessary archive by yourself if you prefer (location is predictable).
I suggest you ask any question on those driver deployment on the related forum thread.

HankB, 2019-07-25:
Yes. That process gets the dongle working. (I haven't performed extensive testing but it comes up and gets an IP address.)
It looks like there is a way to update the module when a new kernel is installed by providing a commit number (for the kernel?) I will ask about that on the thread. I will need to figure that out before I update the kernel. Perhaps a working driver will make it into the next kernel.

macmpi, 2019-07-26:
Glad it worked for you.
Overtime, [`rtl8xxxu`](https://github.com/torvalds/linux/tree/master/drivers/net/wireless/realtek/rtl8xxxu) driver, included in kernel may provide proper support for 8188eu. I have not checked status of this one lately. (note it is blacklisted by default in Raspbian Buster I think).

Meanwhile, other options if you want to build from sources:
https://github.com/lwfinger/rtl8188eu  (likely source of MrEngman)
https://github.com/quickreflex/rtl8188eus

JamesH65, 2019-12-20:
Closing this issue as questions answered/issue resolved.

closed 2019-12-20 by JamesH65