raspberrypi/linux #3105: kernel >  4.19.50-1-ARCH  broken support on mmal hardware decoding

enricoT, 2019-07-24:
**Describe the bug**
After upgrade 

upgraded linux-raspberrypi (4.19.50-1 -> 4.19.58-1)
upgraded linux-raspberrypi-headers (4.19.50-1 -> 4.19.58-1)
upgraded raspberrypi-bootloader (20190611-1 -> 20190715-1)
upgraded raspberrypi-bootloader-x (20190611-1 -> 20190715-1)
upgraded raspberrypi-firmware (20190611-1 -> 20190715-1)

mmal hardware support for ffmpeg mpv is broken, i need to downgrade

downgraded linux-raspberrypi (4.19.58-1 -> 4.19.50-1)
downgraded linux-raspberrypi-headers (4.19.58-1 -> 4.19.50-1)
downgraded raspberrypi-bootloader (20190715-1 -> 20190611-1)
downgraded raspberrypi-bootloader-x (20190715-1 -> 20190611-1)
downgraded raspberrypi-firmware (20190715-1 -> 20190611-1)

to use mpv with hardware acceleration
(+) Video --vid=1 (*) (mpeg2video 720x576 25.000fps)
 (+) Audio --aid=1 (*) (ac3 2ch 48000Hz)
 (+) Subs  --sid=1 (*) (dvd_subtitle)
...
Using hardware decoding (mmal),

thank you
**To reproduce**
List the steps required to reproduce the issue.

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

pelwell, 2019-07-24:
Is this Arch Linux? And running on which hardware?

It's possibly a duplicate (or related to) https://github.com/raspberrypi/linux/issues/3087.

enricoT, 2019-07-25:
is a Raspberry pi 3B with mmal code registration for mmal hardware decoding.
in /boot/config.txt last line is:
decode_MPG2=0x.......
the boot is ok, but if i upgrade, mpv is not able to play dvd or movies in mpeg2 format with hardware acceleration, i have to go back to old version. It is not a problem related to ffmpeg-mmal o mpv-rpi make and build, after upgrade the first test, was to rebuild ffmpeg and mpv, but even with new make of ffmpeg and mpv, i had to downgrade to get hardware decoding. I have two  rpi 3b machines, and after upgrade i have the same problem on sbc number 2 (is the same model 3b) , the only way to see movies with hardware decoding is to downgrade.

pelwell, 2019-07-25:
If it works on Raspbian then you'll have to ask the Arch devs, but it does sound like it could be caused by the new VMSPLIT_3G + LPAE + HIGHMEM settings they've switched to to match the Pi 4.

closed 2019-12-20 by JamesH65