raspberrypi/linux #3107: Timezone selection in Buster from raspi-config not working for Pacific-New

OutsourcedGuru, 2019-07-25:
**Describe the bug**
In Stretch, the `sudo raspi-config` -> `Localisation` -> `Timezone` command would list all the US-based timezones as seen in the `/usr/share/zoneinfo` area, as appropriate. Specifically, `Pacific-New` is the one I use the most since it's `PST/-8` (California).

In Buster, `Pacific-New` is no longer a choice. Selecting `Pacific` equates to `US/Alaska` which is a -9 offset from Greenwich. In fact, most of the zones which incorporate hyphens now appear in the pulldown list for Buster: `East-Indiana` (no), `Indiana-Starke` (yes) and `Pacific-New` (no).

**To reproduce**
* sudo raspi-config
* Localisation Options
* Change Timezone
* US

**Expected behaviour**
I expect that `Pacific-New`, for example, be an option in the list.

**Actual behaviour**
It doesn't appear in the list.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? `Pi3B`
* Which OS and version (`cat /etc/rpi-issue`)?
```
Raspberry Pi reference 2018-04-18
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, d6c238c1b2b1b070a574d3e25048ca442e3e221f, stage2
```

* Which firmware version (`vcgencmd version`)?
```
Jul  9 2019 14:42:57 
Copyright (c) 2012 Broadcom
version 6c3fe3f096a93de3b34252ad98cdccadeb534be2 (clean) (release) (start_x)
```

* Which kernel version (`uname -a`)?
```
Linux octopi 4.19.57-v7+ #1244 SMP Thu Jul 4 18:45:25 BST 2019 armv7l GNU/Linux
```
**Logs**
N/A

**Additional context**
What's also weird is the following unexpected behavior. The underlying `dpkg-reconfigure` which raspi-config uses to update the timezone—once selected within the interface—won't even change to this timezone if done in a non-interactive way.

## As root:
```
rm /etc/localtime
echo "US/Pacific-New" > /etc/timezone
dpkg-reconfigure -f noninteractive tzdata
 # Appears to not work, leaving the original or perhaps badly parsing Pacific-New as Pacific
 Current default time zone: 'US/Alaska'
 Local time is now:      Thu Jul 25 09:52:53 AKDT 2019.
 Universal Time is now:  Thu Jul 25 17:52:53 UTC 2019.
```

OutsourcedGuru, 2019-07-25:
Here is the screenshot from the US region, noting that East-Indiana and Pacific-New are missing:

[image]

OutsourcedGuru, 2019-07-25:
For what it's worth, I next chose `Update` from `raspi-config` but it looks like it's the most recent version.

```
Get:1 http://archive.raspberrypi.org/debian buster InRelease [25.1 kB]             
Get:2 http://archive.raspberrypi.org/debian buster/main armhf Packages [212 kB]                           
Get:3 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]                                                                            
Get:4 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]                                                                  
Fetched 13.2 MB in 49s (268 kB/s)                                                                                                                    
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
raspi-config is already the newest version (20190709).
0 upgraded, 0 newly installed, 0 to remove and 10 not upgraded.
Sleeping 5 seconds before reloading raspi-config
```

XECDesign, 2019-07-25:
This isn't a Linux kernel or even a Raspbian issue.

It was removed because it's not a valid timezone.

https://mm.icann.org/pipermail/tz/2009-February/015448.html

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=815200

OutsourcedGuru, 2019-07-25:
So what's the available -8 offset timezone for the US then?

XECDesign, 2019-07-25:
From what I gather, you're afteeer -7, not -8, since you're on daylight time.
https://www.timeanddate.com/time/zone/usa/los-angeles

> If Pacific-New has been removed then why is it still in the /usr/share/zoneinfo/US folder? I'm sure it's used in more lookups than my own collection of scripts.

I don't know, you should take it up with ICANN and Debian. From what I gather, it's still there because it's still defined in tzdata (although marked deprecated), it's just not a valid timezone so it was removed from the list (see debian bug report linked earlier)

https://en.wikipedia.org/wiki/List_of_tz_database_time_zones

That page suggests that all of the US/ time zones are deprecated and you should use America/Los_Angeles

But again, this really isn't the place for it. This tracker is for issues with the linux kernel.

OutsourcedGuru, 2019-07-25:
Duly-noted on the underlying changes. I can't even imagine the number of shell scripts which are now broken due to the US/Pacific -> America/Los_Angeles changeout, though.

In my humble opinion, whoever deprecated that has never tried to select from a pulldown list which would be now populated with hundreds of city names in North/South/Central America versus a short list of timezones in the states.

closed 2019-07-29 by XECDesign