raspberrypi/linux #3113: [Question] How does Raspbian know about an i2c real-time clock?

etherbrat, 2019-07-30:
The SferaLabs Strato Pi has an MCP7940 i2c RTC chip. SferaLabs provide simple code for a service to load the necessary i2c drivers and initialise the System time from the RTC at startup; however it appears that Raspbian detects the chip and will respond to hwclock commands even if this service is disabled (!). I cannot work out how it does this. /boot/config only loads i2C-dev. I can see that hwmon is loaded although I have no idea how that happens (and it may not be relevant). And in /var/log/boot.log I can see that the system has detected an MCP7940 chip and set it up as rtc-0. How does all of this happen??

pelwell, 2019-07-30:
I would guess that config.txt includes `dtoverlay=i2c-rtc,mcp7940x`. Run (at a shell prompt) `dtoverlay -h i2c-rtc` for more information. If that line is absent then some script must be running `echo mcp7940x 0x6f > /sys/bus/i2c/devices/i2c-1/new_device`. Either way, an out-of-the-box Raspbian image won't detect an RTC.

6by9, 2019-07-30:
Or they've added a HAT EEPROM with the appropriate configuration to setup the rtc.

pelwell, 2019-07-30:
`ls /proc/device-tree/hat` will tell you if there is a HAT EEPROM. To find out what the EEPROM is doing, add `dtdebug=1` to config.txt and run `sudo vcdbg log msg >& log.txt` after booting, then `nano log.txt` (or use your editor of choice).

6by9, 2019-07-30:
> Or they've added a HAT EEPROM with the appropriate configuration to setup the rtc.

Then again, the schematics say not.
https://www.sferalabs.cc/files/strato/doc/stratopi-base-sch.pdf
Nothing is connected to pins 27&28 on CN2.

etherbrat, 2019-07-30:
I inherited the Strato Pi from a developer. Perhaps there is some detritus left over from another project. I’ll start afresh. Thanks for your help. N

Neil Higgins (iPhone)
higgins-demack@bigpond.com

etherbrat, 2019-07-31:
Indeed, echo mcp7940x 0x6f > /sys/bus/i2c/devices/i2c-1/new_device, is included in the startup script for the service, but as I understand it, this would not be persisted across boot-ups.

JamesH65, 2019-12-20:
Closing this issue as questions answered/issue resolved.

closed 2019-12-20 by JamesH65