raspberrypi/linux #3114: [RPi 4 B][4.19.y] WiFi AP makes Kernel panic

kamarya, 2019-07-31:
**System**
Raspberry Pi 4 B
Linux Kernel Branch: 4.19.y
Linux Kernel Version: 4.19.58
SHA1 : d5dc848c982dff2e020f294e384447efe6ea6617
Distribution: Yocto Poky (meta-raspberrypi)

**Context of the issue**

Running _hostapd_ on the on-board WiFi interface as an access point. The AP works fine in the beginning and the traffic passes through. However after a few minutes RPi freezes and I get kernel panics about hostapd (see logs). Before having the panic about hostapd I get these interesting lines:
```
[  979.419611] bcm2835-dma fe007000.dma: swiotlb buffer is full (sz: 1536 bytes)
[  979.426928] bcm2835-dma fe007000.dma: DMA: Out of SW-IOMMU space for 1536 bytes
[  979.434559] mmc-bcm2835 fe300000.mmcnr: dma_map_sg returned zero length
[  989.661972] mmc1: Timeout waiting for hardware interrupt.
[  995.168392] ***** psta->aid (54) out of bounds
```

I also get kernel messages at the boot time complaining about cfg80211 (line 752 in sme.c)
This might be related to the fix of @agherzan (see SHA1 above) and it might still need more fixes. I have not tested this with other versions.

**Kernel logs**

```
root@raspberrypi4-64:~# [  965.706974] WARNING: CPU: 0 PID: 158 at net/wireless/sme.c:752 __cfg80211_connect_result+0x2fc/0x328 [cfg80211]
[  965.717220] Modules linked in: nfc bnep bluetooth ecdh_generic ipv6 iptable_filter xt_state xt_conntrack iptable_nat ipt_MASQUERADE nf_nat_ipv4 nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip_tables x_tables 88XXau(O) brcmfmac brcmutil vc4 sha256_generic bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) v4l2_common cfg80211 videobuf2_vmalloc snd_soc_core videobuf2_memops videobuf2_v4l2 videobuf2_common snd_compress videodev snd_pcm_dmaengine rfkill snd_pcm media snd_timer snd vc_sm_cma(C) vchiq(C) uio_pdrv_genirq uio fixed
[  965.763812] CPU: 0 PID: 158 Comm: kworker/u8:2 Tainted: G        WC O      4.19.58 #1
[  965.771756] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[  965.777737] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[  965.783662] pstate: 40000005 (nZcv daif -PAN -UAO)
[  965.788589] pc : __cfg80211_connect_result+0x2fc/0x328 [cfg80211]
[  965.794836] lr : __cfg80211_connect_result+0x1d4/0x328 [cfg80211]
[  965.801016] sp : ffffff800a633cc0
[  965.804376] x29: ffffff800a633cc0 x28: ffffff8008c46000 
[  965.809774] x27: ffffff800915bcb8 x26: ffffffc07a0f6fb8 
[  965.815172] x25: dead000000000100 x24: dead000000000200 
[  965.820568] x23: ffffff800a633d08 x22: ffffff8008c48688 
[  965.825965] x21: ffffffc07cdae000 x20: ffffffc07a346c00 
[  965.831361] x19: ffffffc079bd8c18 x18: ffffffc077c49694 
[  965.836757] x17: 0000000000000000 x16: 0000000000000000 
[  965.842154] x15: 0000000000000003 x14: 0100081000087f00 
[  965.847550] x13: 0000000000000000 x12: 0000000000000028 
[  965.852945] x11: 0101010101010101 x10: 7f7f7f7f7f7f7f7f 
[  965.858340] x9 : fefefefefefefeff x8 : 0000000000000001 
[  965.863736] x7 : 0000000000000000 x6 : ffffff8008c48000 
[  965.869131] x5 : 0000000000000000 x4 : ffffff800888acb0 
[  965.874526] x3 : ffffff8008c48dd0 x2 : 0000000000000000 
[  965.879922] x1 : 0000000000000000 x0 : 0000000000000000 
[  965.885317] Call trace:
[  965.887862]  __cfg80211_connect_result+0x2fc/0x328 [cfg80211]
[  965.893751]  cfg80211_process_wdev_events+0x120/0x188 [cfg80211]
[  965.899903]  cfg80211_process_rdev_events+0x40/0x98 [cfg80211]
[  965.905876]  cfg80211_event_work+0x24/0x38 [cfg80211]
[  965.911005]  process_one_work+0x1c4/0x420
[  965.915071]  worker_thread+0x50/0x408
[  965.918788]  kthread+0x108/0x138
[  965.922066]  ret_from_fork+0x10/0x1c
[  965.925690] ---[ end trace c74340692d12f570 ]---
[  979.419611] bcm2835-dma fe007000.dma: swiotlb buffer is full (sz: 1536 bytes)
[  979.426928] bcm2835-dma fe007000.dma: DMA: Out of SW-IOMMU space for 1536 bytes
[  979.434559] mmc-bcm2835 fe300000.mmcnr: dma_map_sg returned zero length
[  989.661972] mmc1: Timeout waiting for hardware interrupt.
[  995.168392] ***** psta->aid (54) out of bounds
[ 1088.478020] INFO: task hostapd:336 blocked for more than 120 seconds.
[ 1088.484597]       Tainted: G        WC O      4.19.58 #1
[ 1088.490037] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 1088.498043] hostapd         D    0   336      1 0x00000000
[ 1088.503661] Call trace:
[ 1088.506206]  __switch_to+0xa8/0xf0
[ 1088.509668]  __schedule+0x254/0x858
[... 166 lines cut ...]
[ 1209.691582]  brcmf_sdio_dataworker+0x17a8/0x2218 [brcmfmac]
[ 1209.697259]  process_one_work+0x1c4/0x420
[ 1209.701349]  worker_thread+0x50/0x408
[ 1209.705088]  kthread+0x108/0x138
[ 1209.708388]  ret_from_fork+0x10/0x1c
[ 1209.712038] INFO: task kworker/0:2:560 blocked for more than 120 seconds.
[ 1209.718945]       Tainted: G        WC O      4.19.58 #1
[ 1209.724356] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 1209.732335] kworker/0:2     D    0   560      2 0x00000028
[ 1209.737940] Workqueue: events linkwatch_event
[ 1209.742400] Call trace:
[ 1209.744886]  __switch_to+0xa8/0xf0
[ 1209.748358]  __schedule+0x254/0x858
[ 1209.751920]  schedule+0x3c/0xa0
[ 1209.755128]  schedule_preempt_disabled+0x1c/0x30
[ 1209.759838]  __mutex_lock.isra.0+0x1a4/0x4f8
[ 1209.764193]  __mutex_lock_slowpath+0x24/0x30
[ 1209.768549]  mutex_lock+0x48/0x58
[ 1209.771934]  rtnl_lock+0x1c/0x28
[ 1209.775230]  linkwatch_event+0x14/0x40
[ 1209.779054]  process_one_work+0x1c4/0x420
[ 1209.783143]  worker_thread+0x50/0x408
[ 1209.786880]  kthread+0x108/0x138
[ 1209.790178]  ret_from_fork+0x10/0x1c
```

kamarya, 2019-07-31:
Warning at the boot
```
[   17.137453] WARNING: CPU: 3 PID: 59 at net/wireless/sme.c:752 __cfg80211_connect_result+0x2fc/0x328 [cfg80211]
[   17.147611] Modules linked in: ip_tables x_tables 88XXau(O) brcmfmac brcmutil bcm2835_v4l2(C) sha256_generic vc4 bcm2835_mmal_vchiq(C) v4l2_common videobuf2_vmalloc cfg80211 snd_soc_core videobuf2_memops videobuf2_v4l2 videobuf2_common videodev snd_compress snd_pcm_dmaengine rfkill snd_pcm media snd_timer vc_sm_cma(C) snd vchiq(C) uio_pdrv_genirq uio fixed
[   17.179770] CPU: 3 PID: 59 Comm: kworker/u8:1 Tainted: G         C O      4.19.58 #1
[   17.187626] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[   17.193620] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[   17.199546] pstate: 20000005 (nzCv daif -PAN -UAO)
[   17.204474] pc : __cfg80211_connect_result+0x2fc/0x328 [cfg80211]
[   17.210723] lr : __cfg80211_connect_result+0x1d4/0x328 [cfg80211]
[   17.216903] sp : ffffff800915bcc0
[   17.220263] x29: ffffff800915bcc0 x28: ffffff8008c46000 
[   17.225662] x27: ffffffc07d4d0070 x26: ffffffc07d4d0020 
[   17.231059] x25: dead000000000100 x24: dead000000000200 
[   17.236456] x23: ffffff800915bd08 x22: ffffff8008c48688 
[   17.241854] x21: ffffffc07aaf2000 x20: ffffffc07c664000 
[   17.247250] x19: ffffffc07c662218 x18: 0000000000000000 
[   17.252645] x17: 0000000000000000 x16: 0000000000000002 
[   17.258041] x15: 0000000000000000 x14: 6f9dcd0f5a2c0001 
[   17.263436] x13: 0000000000000040 x12: 0000000000000228 
[   17.268832] x11: 0000000000000000 x10: 0000000000001920 
[   17.274226] x9 : ffffff800915bb10 x8 : ffffffc07cf15380 
[   17.279623] x7 : 0000000000000000 x6 : ffffff8008c48000 
[   17.285019] x5 : 0000000000000000 x4 : ffffff80080d2820 
[   17.290413] x3 : ffffff8008c48dd0 x2 : 0000000000000000 
[   17.295808] x1 : 0000000000000000 x0 : 0000000000000000 
[   17.301204] Call trace:
[   17.303748]  __cfg80211_connect_result+0x2fc/0x328 [cfg80211]
[   17.309637]  cfg80211_process_wdev_events+0x120/0x188 [cfg80211]
[   17.315789]  cfg80211_process_rdev_events+0x40/0x98 [cfg80211]
[   17.321762]  cfg80211_event_work+0x24/0x38 [cfg80211]
[   17.326892]  process_one_work+0x1c4/0x420
[   17.330958]  worker_thread+0x50/0x408
[   17.334676]  kthread+0x108/0x138
[   17.337954]  ret_from_fork+0x10/0x1c
[   17.341579] ---[ end trace 4d1e1b8a1f49bad1 ]---
```

kamarya, 2019-07-31:
The same issue with version 4.19.60 (SHA1 20565c94ca877a099a3c9c5fa39a0380f3d16491) and the warning message at the boot.

```
[   17.224367] WARNING: CPU: 1 PID: 7 at net/wireless/sme.c:752 __cfg80211_connect_result+0x2fc/0x328 [cfg80211]
[   17.234434] Modules linked in: iptable_nat ipt_MASQUERADE nf_nat_ipv4 nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip_tables x_tables 88XXau(O) brcmfmac brcmutil bcm2835_v4l2(C) sha256_generic bcm2835_mmal_vchiq(C) vc4 v4l2_common videobuf2_vmalloc cfg80211 videobuf2_memops videobuf2_v4l2 snd_soc_core videobuf2_common videodev snd_compress rfkill snd_pcm_dmaengine snd_pcm media snd_timer vc_sm_cma(C) snd vchiq(C) uio_pdrv_genirq uio fixed
[   17.274460] CPU: 1 PID: 7 Comm: kworker/u8:0 Tainted: G         C O      4.19.60 #1
[   17.282229] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[   17.288211] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[   17.294137] pstate: 40000005 (nZcv daif -PAN -UAO)
[   17.299064] pc : __cfg80211_connect_result+0x2fc/0x328 [cfg80211]
[   17.305311] lr : __cfg80211_connect_result+0x1d4/0x328 [cfg80211]
[   17.311491] sp : ffffff800805bcc0
[   17.314850] x29: ffffff800805bcc0 x28: ffffff8008c46000 
[   17.320249] x27: ffffffc07d4d0070 x26: ffffffc07d4d0020 
[   17.325646] x25: dead000000000100 x24: dead000000000200 
[   17.331042] x23: ffffff800805bd08 x22: ffffff8008c48688 
[   17.336437] x21: ffffffc07a769000 x20: ffffffc07c6b4800 
[   17.341832] x19: ffffffc07c095618 x18: 0000000000000000 
[   17.347227] x17: 0000000000000000 x16: 0000000000000000 
[   17.352622] x15: 0000000000000000 x14: 0100081000087f00 
[   17.358017] x13: 0000000000000000 x12: 0000000000000020 
[   17.363413] x11: 0101010101010101 x10: ffffffff7f7fff7f 
[   17.368809] x9 : fefefebf7965a607 x8 : 0000000000000001 
[   17.374204] x7 : ffffffc07d5a2c00 x6 : ffffff8008c48000 
[   17.379600] x5 : 0000000000000000 x4 : ffffff800888def0 
[   17.384995] x3 : ffffff8008c48dd0 x2 : 0000000000000000 
[   17.390389] x1 : 0000000000000000 x0 : 0000000000000000 
[   17.395784] Call trace:
[   17.398331]  __cfg80211_connect_result+0x2fc/0x328 [cfg80211]
[   17.404222]  cfg80211_process_wdev_events+0x120/0x188 [cfg80211]
[   17.410374]  cfg80211_process_rdev_events+0x40/0x98 [cfg80211]
[   17.416346]  cfg80211_event_work+0x24/0x38 [cfg80211]
[   17.421476]  process_one_work+0x1c4/0x420
[   17.425541]  worker_thread+0x50/0x408
[   17.429260]  kthread+0x108/0x138
[   17.432536]  ret_from_fork+0x10/0x1c
[   17.436160] ---[ end trace 955367a198e4a3fd ]---
```

ED6E0F17, 2019-08-07:
This is interesting: you seem to have hit issue #1988 in a new way, and I REALLY want to get that one closed.

It might be safe to close this as a duplicate of that one, since I am fairly sure that the wifi issue is not related to your broken 64bit dma

kamarya, 2019-08-07:
Any idea about why the kernel freezes ? and about `hostapd` timeout panic message repeating after the freeze?

I am not sure where these come from.

ED6E0F17, 2019-08-08:
kamarya: The DMA fails 14 seconds after the wifi "cfg80211_connect_result" issue in the above logs, but you suggest that these Warnings happen more often than the "Hung Task." - The "swiotlb buffer is full" message is the one you should focus on, and check in your dmesg that the DMA is being set up properly.

kamarya, 2019-08-23:
I have built again the kernel with the latest SHA1 9798c94b34db4a32707f7f501616dbcbd088a696 on Aug 21 2019. Broadcom WiFi in AP mode stops working after a while due to the same issue.

yaroslavros, 2019-08-23:
https://github.com/raspberrypi/linux/pull/3164 is not merged yet. Can you manually apply https://github.com/raspberrypi/linux/pull/3164/commits/9dfbe9f500e85d2f751a240f6144785f237d795a and https://github.com/raspberrypi/linux/pull/3164/commits/1395119b339e946fb87f4ff9da2c85c15947734d to give it another try?

kamarya, 2019-08-23:
> #3164 is not merged yet. Can you manually apply [9dfbe9f](https://github.com/raspberrypi/linux/commit/9dfbe9f500e85d2f751a240f6144785f237d795a) and [1395119](https://github.com/raspberrypi/linux/commit/1395119b339e946fb87f4ff9da2c85c15947734d) to give it another try?

I will give it a try as soon as I am back home :)

kamarya, 2019-08-24:
@yaroslavros after applying the proposed patch in #3164  on top of 9798c94 the issue has not been reproduced yet. The fix looks promising and I will keep testing it.

agherzan, 2019-09-09:
We should be able to close this issue now that the DMA leaks PR was merged.

closed 2019-09-09 by pelwell