raspberrypi/linux #3115: vdsomunge: Failed to map vdso.so.dbg: Invalid argument

gpanders, 2019-07-31:
**Describe the bug**
Add a clear and concise description of what you think the bug is.
`make zImage` produces
```
  CALL    scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  MUNGE   arch/arm/vdso/vdso.so.dbg
./arch/arm/vdso/vdsomunge: Failed to map arch/arm/vdso/vdso.so.dbg: Invalid argument
arch/arm/vdso/Makefile:44: recipe for target 'arch/arm/vdso/vdso.so.dbg' failed
make[1]: *** [arch/arm/vdso/vdso.so.dbg] Error 1
Makefile:1044: recipe for target 'arch/arm/vdso' failed
make: *** [arch/arm/vdso] Error 2
```

**To reproduce**
This is using a clean `git clone https://github.com/raspberrypi/linux`. I also downloaded the `tools` repo and added `tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin` to my PATH and set `CROSS_COMPILE` to `arm-linux-gnueabihf-`. This is all running on Ubuntu 18.04 in a VM.

**Expected behaviour**
zImage builds normally.

**Actual behaviour**
See error message above.

Note that building the modules and dtbs with `make modules dtbs` works as expected.

**System**
N/A (this error occurs while cross-compiling the kernel on my host machine).

**Additional context**
This is on the default branch `rpi-4.19.y`.

pelwell, 2019-07-31:
I cross compile kernels every day - it's not broken.

Follow our kernel building guide (https://www.raspberrypi.org/documentation/linux/kernel/building.md) without changes, then report back with your findings.

gpanders, 2019-07-31:
I did follow that guide without changes, so please consider my initial comment to be my report back.

JamesH65, 2019-07-31:
"make zImage" is not in those instructions anywhere, to cross compile use

make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- zImage

gpanders, 2019-07-31:
Fair enough, but
```bash
export ARCH=arm
export CROSS_COMPILE=arm-linux-gnueabihf-
export KERNEL=kernel7
make zImage
```
has the exact same effect, which is what I used. As a sanity check, I just tried it again specifying those environment variables on the command line, which had the same result.

pelwell, 2019-07-31:
```
$ base64 > edid.bin <<EOF
AP///////wBjGAAAAAAAABIYAQOAAAAACtelollKliQUUFSjCACBwAEBAQEBAQEBAQEBAQEBfyFW
qlEAHjBGjzMAP0MhAAAeqRoAoFAAFjAwIDcAP0MhAAAaAAAA/QAyTB5QEAAKICAgICAgAAAA/ABB
QUEKICAgICAgICAgAbQCAyR0UJAFBAMHAgYBHxQTEhYRFSAjCQcDgwEAAGYDDAAgAAABHQC8UtAe
ILgoVUDEjiEAAB4BHYDQchwWIBAsJYDEjiEAAJ6MCtCKIOAtEBA+lgATjiEAABiMCtCQIEAxIAxA
VQATjiEAABgAAAAAAAAAAAAAAAAAAAAAAAAA8w==
EOF
```

pelwell, 2019-07-31:
Please post the output of the following commands:
```
$ ls -l arch/arm/vdso
$ file arch/arm/vdso/vdsomunge
$ ldd arch/arm/vdso/vdsomunge
```

JamesH65, 2019-07-31:
Anyway, just trying this out, so far all working fine. The only thing I can think this might be is the following statement is spread over two lines in the docs, but is actually one line.

"make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2711_defconfig"

gpanders, 2019-07-31:
```console
$ ls -l arch/arm/vdso
total 68
-rw-r--r-- 1 vagrant vagrant  2481 Jul 30 21:00 Makefile
-rw-r--r-- 1 vagrant vagrant   287 Jul 30 21:00 datapage.S
-rw-r--r-- 1 vagrant vagrant   968 Jul 30 21:10 datapage.o
-rw-r--r-- 1 vagrant vagrant     0 Jul 30 21:10 modules.builtin
-rw-r--r-- 1 vagrant vagrant     0 Jul 30 21:14 modules.order
-rw-r--r-- 1 vagrant vagrant   972 Jul 30 21:00 vdso.S
-rw-r--r-- 1 vagrant vagrant  4330 Jul 31 07:28 vdso.lds
-rw-r--r-- 1 vagrant vagrant  2126 Jul 30 21:00 vdso.lds.S
-rwxr-xr-x 1 vagrant vagrant  3757 Jul 31 07:28 vdso.so.raw
-rwxr-xr-x 1 vagrant vagrant 13264 Jul 31 07:28 vdsomunge
-rw-r--r-- 1 vagrant vagrant  5716 Jul 30 21:00 vdsomunge.c
-rw-r--r-- 1 vagrant vagrant  5672 Jul 30 21:00 vgettimeofday.c
-rw-r--r-- 1 vagrant vagrant  2796 Jul 30 21:10 vgettimeofday.o
```

```console
$ file arch/arm/vdso/vdsomunge
arch/arm/vdso/vdsomunge: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/l, for GNU/Linux 3.2.0, BuildID[sha1]=757ccf1ded632c31d27c13187204073241fb4447, not stripped
```

```console
$ ldd arch/arm/vdso/vdsomunge
        linux-vdso.so.1 (0x00007ffe207f9000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f4710f8c000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f4711580000)
```

gpanders, 2019-08-01:
So my theory turned out to be correct. Once I saw that the error was coming from a failed `mmap` I realized what the problem was. I am compiling the Linux sources inside a VM with the source directory being a shared folder with the host. This reasonably caused the `mmap` to fail (probably due to some kind of permissions issues with files in a shared folder). This also explains why I wasn't able to find this error message on Google, since it is such a unique situation.

Cloning the sources to a "native" path inside the VM has solved the issue.

closed 2019-08-01 by gpanders