raspberrypi/linux #3118: status of v4l2 m2m hw decode support in 5.2/5.3 64bit mainline kernel

warpme, 2019-08-02:
I'm creating this ticket purely to track progress of v4l2 m2m hw decode support for rpi3/4 on mainline kernels with aarch64 arch.

Reason: in my project (mythtv appliance minimyth2) I want to have single kernel for rpi3/4, amlogic s905/s912, rockchip 33xx and allwinner A64/H6. 

So far I have this for all listed except rpi3/4.
 
What is timeline/status of v4l2 m2m hw decode on mainline?

6by9, 2019-08-02:
> What is timeline/status of v4l2 m2m hw decode on mainline?

When we have resource to attempt to upstream it.
Sorry, upstreaming stuff is not a priority.

warpme, 2019-11-13:
Guys, how v4l2 m2m hw decode support in 64bit mainline kernel is going? 
Any update? 
Do you think we can expect it in month, quarter, year, never, etc?
We (mythtv team) progressing nicely with v4l2 m2m + DRM Prime so having single common SD card image for RPI/AmLogic/AllWinner SoC would be nice....

JamesH65, 2019-12-20:
@6by9 Wasn't this pretty much done?

6by9, 2019-12-20:
It's implemented in our downstream tree, not in mainline.

As per my original answer
> Sorry, upstreaming stuff is not a priority.

Our main driver to get stuff into mainline is going to be Libcamera.

A common SD card may be tricky seeing as you need the FAT boot partition for the Pi, but that's not my problem.

JamesH65, 2019-12-20:
Sounds like we will not do doing specifically what is requested here, so closing.

closed 2019-12-20 by JamesH65