raspberrypi/linux #3119: Error pairing with Secure Simple Pairing (SSP) disabled

Douglas6, 2019-08-02:
There seems to be a problem with Bluetooth pairing when Secure Simple Pairing is disabled. Several posts on this forum alerted me to the problem (which I can verify): https://www.raspberrypi.org/forums/view ... 9&t=195090 and https://www.raspberrypi.org/forums/view ... 8&t=244241.

To reproduce the issue:

Flash the latest Buster full image and upgrade as of today.
Disable SSP with 'sudo hciconfig hci0 sspmode 0'
Enter bluetoothctl and make the Pi discoverable.
Attempt to pair from another device (I used an Android phone running Pie, ironically.)
No joy.
An hcidump transcription goes like this:

HCI sniffer - Bluetooth packet analyzer ver 5.50
device: hci0 snap_len: 1500 filter: 0xffffffff

HCI Event: Command Complete (0x0e) plen 4
Write Simple Pairing Mode (0x03|0x0056) ncmd 1
status 0x00
HCI Event: Command Complete (0x0e) plen 4
Write Current IAC LAP (0x03|0x003a) ncmd 1
status 0x00
HCI Event: Command Complete (0x0e) plen 4
Write Scan Enable (0x03|0x001a) ncmd 1
status 0x00
HCI Event: Connect Request (0x04) plen 10
bdaddr 0C:CB:85:0D:6B:AB class 0x5a020c type ACL
HCI Event: Command Status (0x0f) plen 4
Accept Connection Request (0x01|0x0009) status 0x00 ncmd 1
HCI Event: Command Complete (0x0e) plen 10
PIN Code Request Negative Reply (0x01|0x000e) ncmd 1
status 0x00 bdaddr 0C:CB:85:0D:6B:AB
HCI Event: Connect Complete (0x03) plen 11
status 0x18 handle 12 bdaddr 0C:CB:85:0D:6B:AB type ACL encrypt 0x00
Error: Pairing Not Allowed

Attempts with a dongle that does not support SSP worked fine, which is why I suspect Raspbian rather than BlueZ (as well as the lack of outcry from other distros).

pi@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage4
pi@raspberrypi:~ $ vcgencmd version
Jul 9 2019 14:40:53
Copyright (c) 2012 Broadcom
version 6c3fe3f096a93de3b34252ad98cdccadeb534be2 (clean) (release) (start)
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.58-v7+ #1245 SMP Fri Jul 12 17:25:51 BST 2019 armv7l GNU/Linux

sakthivelj, 2021-02-02:
same here also, anyone found any solution for this?. I am using **raspberry PI 4 b** and **android 10**

baderas, 2021-02-24:
Can confirm, same happens on the raspberry pi 3 b

bhaumik90, 2021-05-02:
Facing same issue with RPi 3 and Android 10. However, I was able to connect iPhone to the RPi.

ThibautPlg, 2022-09-22:
Any news on this subject ? I have the same problem here with a Pi4 and Android12. 
Bluetooth through my linux desktop worked on first try. Seems like a Android related issue.

Christiancoding, 2024-05-21:
I found this:
https://www.youtube.com/watch?v=lxmLorWYE-c

maximko, 2024-09-29:
Problem is still there with rpi4b and latest Raspberry Pi OS.