raspberrypi/linux #3120: RPI4 ARM64 error: sdhci-iproc fe340000.emmc2: swiotlb buffer is full, 4.19.63-v8-g75f1d14ce

satmandu, 2019-08-02:
RPI4 4Gb running 4.19.63-v8-g75f1d14ce arm64 ubuntu-current.
Should I be worried by this error I'm seeing from dmesg:

```
[23813.111533] sdhci-iproc fe340000.emmc2: swiotlb buffer is full (sz: 65536 bytes)
[23813.111542] sdhci-iproc fe340000.emmc2: DMA: Out of SW-IOMMU space for 65536 bytes
[23813.124991] sdhci-iproc fe340000.emmc2: swiotlb buffer is full (sz: 65536 bytes)
[23813.124999] sdhci-iproc fe340000.emmc2: DMA: Out of SW-IOMMU space for 65536 bytes
[23813.132689] ------------[ cut here ]------------
[23813.132730] WARNING: CPU: 3 PID: 118 at drivers/mmc/host/sdhci.c:968 sdhci_send_command+0xc4c/0xd38
[23813.132733] Modules linked in: binfmt_misc dm_mod bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) v4l2_common videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media brcmfmac vc_sm_cma(C) brcmutil spidev sha256_generic cfg80211 rfkill vchiq(C) spi_bcm2835 uio_pdrv_genirq uio sch_fq_codel ip_tables x_tables ipv6 btrfs zstd_decompress zstd_compress xxhash lzo_compress zlib_deflate raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq raid1 raid0 linear md_mod i2c_bcm2835 fixed
[23813.132842] CPU: 3 PID: 118 Comm: kworker/3:1H Tainted: G         C        4.19.63-v8-g75f1d14ce #1
[23813.132845] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[23813.132853] Workqueue: kblockd blk_mq_run_work_fn
[23813.132859] pstate: 00000085 (nzcv daIf -PAN -UAO)
[23813.132864] pc : sdhci_send_command+0xc4c/0xd38
[23813.132868] lr : sdhci_send_command+0xc4c/0xd38
[23813.132870] sp : ffffff80098539f0
[23813.132873] x29: ffffff80098539f0 x28: ffffff8008c16000 
[23813.132879] x27: 0000000000000002 x26: 0000000000000000 
[23813.132885] x25: ffffffc0bbb94000 x24: ffffff8008c18688 
[23813.132891] x23: 0000000000418958 x22: 0000000000000005 
[23813.132897] x21: ffffffc0bb2391f0 x20: ffffffc0bb239270 
[23813.132903] x19: ffffffc0bbb94500 x18: ffffff8008c18688 
[23813.132909] x17: 0000000000000001 x16: 0000000000000007 
[23813.132915] x15: 0000000000000010 x14: 0720072007200720 
[23813.132921] x13: 00000000000047e0 x12: ffffff8008d11000 
[23813.132926] x11: ffffff8008c2c000 x10: ffffff8008d11f20 
[23813.132932] x9 : 0000000000000000 x8 : 0000000000000004 
[23813.132938] x7 : ffffff8008d11000 x6 : 00000040b73ad000 
[23813.132944] x5 : 0000000000000002 x4 : 0000000000000000 
[23813.132950] x3 : ffffffc0bffac9d8 x2 : 3949e550559fe500 
[23813.132955] x1 : 0000000000000000 x0 : 0000000000000024 
[23813.132962] Call trace:
[23813.132967]  sdhci_send_command+0xc4c/0xd38
[23813.132971]  sdhci_request+0xb0/0xf8
[23813.132975]  __mmc_start_request+0x88/0x1d8
[23813.132977]  mmc_start_request+0x88/0xb8
[23813.132981]  mmc_blk_mq_issue_rq+0x288/0x784
[23813.132985]  mmc_mq_queue_rq+0x10c/0x278
[23813.132989]  blk_mq_dispatch_rq_list+0xb0/0x508
[23813.132993]  blk_mq_do_dispatch_sched+0x6c/0x108
[23813.132997]  blk_mq_sched_dispatch_requests+0x138/0x188
[23813.133000]  __blk_mq_run_hw_queue+0xb0/0x118
[23813.133004]  blk_mq_run_work_fn+0x28/0x38
[23813.133008]  process_one_work+0x1c4/0x420
[23813.133011]  worker_thread+0x50/0x408
[23813.133014]  kthread+0x108/0x138
[23813.133018]  ret_from_fork+0x10/0x1c
[23813.133021] ---[ end trace c9246121e463e30c ]---
```
Fullish dmesg here: https://paste.ubuntu.com/p/BsTRn6NVHJ/ (which also has wifi errors.)

```
cat /boot/firmware/config.txt 
enable_uart=1
#kernel=kernel8.bin
kernel=kernel8.img
device_tree_address=0x03000000
dtparam=i2c_arm=on
dtparam=spi=on
arm_64bit=1
armstub=armstub8-gic.bin
enable_gic=1
initramfs initrd.img followkernel

# Disable Ethernet LEDs
dtparam=eth_led0=14
dtparam=eth_led1=14

# Disable the PWR LED
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off

# Disable the Activity LED
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off

#dtoverlay=upstream
device_tree=bcm2711-rpi-4-b.dtb
total_mem=3072
```

```
cat /boot/firmware/cmdline.txt 
net.ifnames=0 dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fsck.mode=force fsck.repair=yes
```

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi?  Pi 4B
* Which OS and version (`cat /etc/rpi-issue`)?
```cat /etc/issue
Ubuntu Eoan Ermine (development branch) \n \l```
* Which firmware version (`vcgencmd version`)?
```sudo /opt/vc/bin/vcgencmd version
Jul 25 2019 17:52:38 
Copyright (c) 2012 Broadcom
version f09a65b56d867fa06b7c572d4bacea3e25fe5756 (clean) (release) (start)```
* Which kernel version (`uname -a`)?  ```uname -a
Linux rpi4 4.19.63-v8-g75f1d14ce #1 SMP PREEMPT Fri Aug 2 01:06:27 UTC 2019 aarch64 aarch64 aarch64 GNU/Linux```

pelwell, 2019-08-02:
That is to be expected at the moment - set total_mem to 1024 as a workaround.

See https://github.com/raspberrypi/linux/issues/3032#issuecomment-506912449 for more details.

satmandu, 2019-08-02:
Thanks. As per https://github.com/raspberrypi/linux/issues/3032#issuecomment-511214995 I'm going to try using ```sdhci.debug_quirks=96``` with ```total_mem=3072``` for a bit to see if the device is functional. It seems to be booting just fine:

```
[    1.232533] sdhci: Secure Digital Host Controller Interface driver
[    1.232556] sdhci: Copyright(c) Pierre Ossman
[    1.232960] mmc-bcm2835 fe300000.mmcnr: could not get clk, deferring probe
[    1.233333] sdhci-pltfm: SDHCI platform and OF driver helper
[    1.236327] ledtrig-cpu: registered to indicate activity on CPUs
[    1.236405] hidraw: raw HID events driver (C) Jiri Kosina
[    1.236570] usbcore: registered new interface driver usbhid
[    1.236592] usbhid: USB HID core driver
[    1.236858] Initializing XFRM netlink socket
[    1.236900] NET: Registered protocol family 17
[    1.237004] Key type dns_resolver registered
[    1.237870] registered taskstats version 1
[    1.237898] Loading compiled-in X.509 certificates
[    1.245282] uart-pl011 fe201000.serial: cts_event_workaround enabled
[    1.245398] fe201000.serial: ttyAMA0 at MMIO 0xfe201000 (irq = 19, base_baud = 0) is a PL011 rev2
[    1.248658] console [ttyS0] disabled
[    1.248704] fe215040.serial: ttyS0 at MMIO 0x0 (irq = 22, base_baud = 62500000) is a 16550
[    1.279198] usb 1-1: New USB device found, idVendor=2109, idProduct=3431, bcdDevice= 4.20
[    1.279485] console [ttyS0] enabled
[    1.282785] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[    1.330400] brcmstb_thermal fd5d2200.thermal: registered AVS TMON of-sensor driver
[    1.363780] usb 1-1: Product: USB2.0 Hub
[    1.365099] hub 1-1:1.0: USB hub found
[    1.371729] mmc-bcm2835 fe300000.mmcnr: mmc_debug:0 mmc_debug2:0
[    1.378530] hub 1-1:1.0: 4 ports detected
[    1.384943] mmc-bcm2835 fe300000.mmcnr: DMA channel allocated
[    1.410989] sdhci-iproc fe340000.emmc2: Linked as a consumer to regulator.1
[    2.650643] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
[    2.657862] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    2.665069] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    2.670688] mmc0: SDHCI controller on fe340000.emmc2 [fe340000.emmc2] using PIO
[    2.679627] of_cfs_init
[    2.681146] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
[    2.682254] of_cfs_init: OK
[    2.689335] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    2.699282] Freeing unused kernel memory: 960K
[    2.703985] Run /init as init process
[    2.739645] random: fast init done
[    2.773254] mmc1: new high speed SDIO card at address 0001
[    2.804663] mmc0: new ultra high speed DDR50 SDXC card at address aaaa
[    2.813108] mmcblk0: mmc0:aaaa SC128 119 GiB
[    2.818118] usb 1-1.1: new high-speed USB device number 3 using xhci_hcd
[    2.827452]  mmcblk0: p1 p2
```

closed 2019-08-05 by satmandu