raspberrypi/linux #3124: Strange duplicated code after merge

psgreco, 2019-08-03:
After commit a21b98653ecf7b2f71906228d9965d8174a1c275 there is a strange duplicated atomic_inc/atomic_dec in drivers/staging/vc04_services/vchiq-mmal/mmal-vchiq.c which looks more like a bad merge than actual code.

Even more, git blame seems to point to 2 different versions of the same patch.

https://github.com/raspberrypi/linux/commit/a21b98653ecf7b2f71906228d9965d8174a1c275#diff-7aba1a24b36b535c3c8817c7dda105c6

6by9, 2019-08-05:
Thanks for the report. Yes, it's a bad merge from upstream. They picked up a load of my patches and backported them to 4.19, and it looks like the merge has gone wrong.
I'll sort out a PR.

JamesH65, 2019-09-17:
@6by9 Did this get done? Closeable?

6by9, 2019-09-17:
Yes, see the linked commits.

closed 2019-09-17 by JamesH65