raspberrypi/linux #3125: rpi kernel missing usb struct adapter item pm_usage_cnt

jshenry1963, 2019-08-04:
Please let me know if this is the wrong place to post this, but I have tracked it to a change that was made in the kernel that causes this anomaly.

In the linux kernel version 4.19.41 the item pm_usage_cnt was removed from the usb struct adapter. It was considered a duplicate counter.  Unfortunately, there are several packages that rely on this item. And some of these cannot be changed. In the main linux kernel it was added back in version 4.20.
One of the 3rd party source packages I am using requires it to know if it is safe to "put" the interface or not.

For the Raspberry Pi 3 B+, I can go back a version in the kernel and use that as a work around for now until the raspberry pi kernel is brought up to the core of 4.20 (which has the pm_usage_cnt added back in)
However, for the Pi 4, this is not an option. A Pi 4 compatible kernel does not include this change yet.

On the Pi 4
uname -a
Linux raspberrypi 4.19.58-v7l+ #1245 armv71 GNU/Linux
OS and Version
Raspberry Pi reference 2019-07-10
If this is the wrong forum, please let me know where to post this, and, if you need anymore info, please let me know.

Thanks
John

pelwell, 2019-08-04:
I will happily back port a fix patch - do you have the commit hash for the fix? If not, I'm sure I can find it.

jshenry1963, 2019-08-04:
It was originally in the code, then reported as a duplicate and removed because of a bug....
Here is the link to the change to remove it, with the description of why.
If I need to get the 3rd party to use a different counter instead of porting this back in, let me know. 
Appreciate the help
https://github.com/raspberrypi/linux/commit/83c6688d679c1d95ed79650f4ec6e702548c39ef#diff-d81b85761c12e5ebdcaca4309ff73cfe

pelwell, 2019-08-04:
It looks as though the reason pm_usage_cnt is in the 4.20 kernel is that the commit removing it wasn't back-ported. The patch was introduced in 5.1, then because it is a bug fix it got back-ported to 4.19 which is a Long Term Stable kernel. Reverting that patch is not a viable option long term (and would reintroduce the bug), so I think the 3rd party code needs another strategy.

jshenry1963, 2019-08-04:
Thank you very much for your time. I agree 100%, I do have the source code for this and modified it to use ->dev.power.usage_count, instead of ->pm_usage_cnt, and seems to work just fine, compiles and runs well on the Pi 3B+.
I will go ahead and mark this one as closed / not a problem. My bad for bringing it up before I researched a bit more.
I'll inform the owner of the code and let them know what we found / y'all said.

Take care, and thanks
John

closed 2019-08-04 by jshenry1963