raspberrypi/linux #3126: Pi 3 B+ fails to boot from usb when using linux-firmware-20190717.bf13a71-1

erikschreier, 2019-08-05:
**Describe the bug**
When i flash Raspbian Stretch (2019-04-08) on an USB Drive its booting with no Problems, i can update nearly all Packages with no Problems except the linux-firmware package.

When updating the linux-firmware Package on Stretch or Arch Linux ARM my Pi 3 B+ (rev 1.3UK) stops booting from the USB Drive, The green LED shows up for a few seconds and after a while it starts blinking 3 times.

Same Problem appears when flashing Raspbian Buster 2019-07-12 and 2019-06-24 to an USB Drive, it wont boot up from the beginning.

On Arch Linux ARM the exactly same Problem happens when updating the linux-firmware to 20190717.bf13a71-1.

When flashing the Images to an SD Card or cloning the USB drives to a SD Card the Pi boots with no Problems.

The USB Drive boots with no Problems on my Pi 3 B so i think its a Problem exclusive on Pi 3 B+ or may for the rev 1.3 of the Pi 3+ cause same Problems are reportet by different Users in the german and UK Forum but there are some telling that the 3 B+ will boot.

Ive tested the Scenario using 2 Sandisk Ultra 64Gb USB 3.0 and an Intenso Aluline 16Gb USB 2.0, these drives working with no Problems before the Update and also running well when installing Ubuntu Mate or Stretch <= 2019-04-08 withouth the linux-firmware Update.

**To reproduce**
Writing the Raspbian Buster Image on my Fedora 29 Desktop to the USB Drive using dd or ddrescue (tested both) using bs=1M and bs=512

Also extracting the Arch Linux ARM arm7l archive to an prepared USB drive following the official installation Guide for armv7 Hardware.

After updating the linux-firmware package and rebooting (or shutting down and repowering) the RPI the Bug stops the Boot Process of the 3 B+.

Bug can be reproduced in 100% of the installations.

**Expected behaviour**
The Pi will boot after the Update into the Displaymanager.

**Actual behaviour**
The green LED appears for about 3 to 5 Seconds (depending on the Speed of the Drive) and starts then blinking 3 times in intervall.

**System**
* Model: RPI 3 B+ (rev 1.3 UK)
* OS: Raspbian Stretch 2019-04-08, Raspbian Buster 2019-07-12, Arch Linux armv7l
* Firmware: 2019727-1 (ARCH)
* Kernel: 4.19.63-1-ARCH

**Logs**
  Not possible

**Additional context**
  Bought the 3 B+ on December 2018 in Germany, the Inteno Aluline 16 GB USB 2.0 is 2 Years old, the two Sandisk Ultra 64Gb USB 3.0 3 Months.

  When cloning the Drive to an SD Card there will be no errors and the Pi 3 B+ will start with no problem from the cloned SD Card (after rewriting .cmdline.txt and fstab in ARCH)

JamesH65, 2019-12-20:
@erikschreier Is this still present after an apt update/full-upgrade ?

gorky, 2020-01-08:
Happening with the clean image from 2019-09-26-raspbian-buster-lite.zip on a PI3 using an USB HDD.  Was running automatic updates, a few months back the Screen stopped working.  I thought it was because I had broken the HDMI plug, so didn't think to grab the syslog or check for recent upgrades (sorry).
2 days ago, system stopped booting & w/o screen, couldn't log in.  So started with fresh image on an SD, it worked.  Same version on the HDD, no screen.  Tried a clean install on HDD, no boot from external HDD with 2019-07-10-raspbian-buster-lite.zip.  As in, no splash, no raspi's.  But I could boot to login prompt from an SD card.

Flashed 2019-09-26-raspbian-buster-lite.zip  onto SD & HDD.  Can boot from SD.  HDD gives 4 raspi's, and then flashing cursor, and that's it.  No additional boot progress, just hangs.

I did set program_usb_boot_mode=1 in the config.txt file on both the SD card & the HDD before booting.

Current workaround for booting the PI3B is too use SD card to provide /boot, and the load OS from HDD.

gorky, 2020-01-08:
Subsequent boots fail to even get the Raspi's to display, or any green LEDs.