raspberrypi/linux #3127: RPI4 arm64 wireless issues w/ power management

satmandu, 2019-08-05:
Tons of these errors in dmesg:

```
[ 3629.107094] brcmfmac: brcmf_fill_bss_param: Failed to get bss info (-110)
[ 3631.658501] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 3636.778595] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 3636.787384] brcmfmac: brcmf_fill_bss_param: Failed to get bss info (-110)
[ 3639.338646] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 3641.898708] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 3641.907354] brcmfmac: brcmf_cfg80211_get_station: GET STA INFO failed, -110
[ 3644.458799] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 3647.018797] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 3647.027715] brcmfmac: brcmf_cfg80211_get_station: GET STA INFO failed, -110
[ 3649.578831] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 3652.138927] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 3652.147566] brcmfmac: brcmf_cfg80211_get_station: GET STA INFO failed, -110
[ 3654.698978] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 3657.259021] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 3657.267675] brcmfmac: brcmf_cfg80211_get_station: GET STA INFO failed, -110
[ 3659.819095] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 3662.379168] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 3662.389004] brcmfmac: brcmf_cfg80211_get_station: GET STA INFO failed, -110
[ 3664.939200] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 3667.499250] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 3667.507892] brcmfmac: brcmf_cfg80211_get_station: GET STA INFO failed, -110
```

**Describe the bug**
Connect to a 5Ghz network, and these errors appear with dropouts within a matter of seconds during any sustained network activity.

Setting ```iwconfig wlan0 power off``` in an /etc/rc.local fixes the issue

System

Which model of Raspberry Pi? Pi 4B
Which OS and version (cat /etc/rpi-issue)?
```
cat /etc/issue
Ubuntu Eoan Ermine (development branch) \n \l
```
Which firmware version (vcgencmd version)?
```
sudo vcgencmd version
Aug  1 2019 15:14:50 
Copyright (c) 2012 Broadcom
version a49274f4de9406db851802c64ea2e8a13df7e0d3 (clean) (release) (start)
```
Which kernel version (uname -a)?
```
uname -a
Linux rpi4 4.19.63-v8-gdee436110 #1 SMP PREEMPT Sat Aug 3 20:29:00 UTC 2019 aarch64 aarch64 aarch64 GNU/Linux
```
Prior shared pastebin describes issue well: https://paste.ubuntu.com/p/BsTRn6NVHJ/

JamesH65, 2019-08-05:
Does this  go wrong in the release version of Raspbian? We are unlikely to spend any time on bugs in other  people distro's (Ubuntu in your case)

erikschreier, 2019-08-05:
The Problem is known for Ubuntu like Systems on first boot, thats why the official Mate Beta of Ubuntu has "Disabled WiFi Power Management." in Patchnotes...

5 Minutes of Google wont hurt.

satmandu, 2019-08-05:
@erikschreier I'm just noting it in case people search for the issue. I've already worked around it as per my experience with similar issues on older Raspberry PI devices. Also this problem isn't happening just after first boot for me.

@JamesH65 This is on arm64, so no I haven't verified this on Raspbian. This issue does not crop up on the RPI3B+ on ubuntu/arm64 using the ubuntu kernel.

I'm using an almost entirely rpi-4.19.y compiled kernel so I was just wondering if this was a kernel issue. Is power management an issue on the 4B for anyone else (or just on arm64)?

pelwell, 2019-08-05:
The downstream kernel includes a commit which should disable power management - `dmesg | grep brcmfmac` will tell you if it is present or not.

satmandu, 2019-08-05:
Huh. So it appears that it is disabled by default. Must be pebkac...

pelwell, 2019-08-05:
It is still possible that the commit, which will have been cherry-picked for many kernel versions, no longer does what it used to.

pelwell, 2019-08-05:
No - it still looks reasonable.

satmandu, 2019-08-06:
Just noticed some slowness and saw this:

```
 dmesg | grep brcmfmac ; iwconfig
[   10.509836] brcmfmac: F1 signature read @0x18000000=0x15264345
[   10.597779] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   10.614777] usbcore: registered new interface driver brcmfmac
[   11.214024] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   11.238572] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[   75.202241] brcmfmac: power management disabled
docker0   no wireless extensions.

wlan0     IEEE 802.11  ESSID:"Manhattan"  
          Mode:Managed  Frequency:5.54 GHz  Access Point: FE:EC:DA:97:E6:12   
          Bit Rate=13 Mb/s   Tx-Power=31 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
          Link Quality=31/70  Signal level=-79 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:21  Invalid misc:0   Missed beacon:0

eth0      no wireless extensions.
```
so I ran ```iwconfig wlan0 power off```

and now see this:
```
 dmesg | grep brcmfmac ; iwconfig
[   10.509836] brcmfmac: F1 signature read @0x18000000=0x15264345
[   10.597779] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   10.614777] usbcore: registered new interface driver brcmfmac
[   11.214024] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   11.238572] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[   75.202241] brcmfmac: power management disabled
[ 1340.257059] brcmfmac: power management disabled
docker0   no wireless extensions.

wlan0     IEEE 802.11  ESSID:"Manhattan"  
          Mode:Managed  Frequency:5.54 GHz  Access Point: FE:EC:DA:97:E6:12   
          Bit Rate=12 Mb/s   Tx-Power=31 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=31/70  Signal level=-79 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:22  Invalid misc:0   Missed beacon:0

eth0      no wireless extensions.

lo        no wireless extensions.
```
It's possible the driver is reporting power management disabled and something is enabling it again?

In any case I'm now disabling it in rc.local again just to be safe.

pelwell, 2019-08-06:
The function that is called to change the power manege state of the WiFi chip is modified so that all calls - whether to enable or disable - cause it to be disabled _and_ return no error, which could cause the caller to think power management had been enabled. But there's no harm in you making doubly sure.

satmandu, 2019-08-06:
Thanks for clearing that up.

closed 2019-08-06 by satmandu