raspberrypi/linux #3136: RPi4: Purethermal 2 board not working

agjunyent, 2019-08-08:
**Describe the bug**
I've trying to obtain images on a newly baked RPi4 with Raspbian Buster from a PureThermal 2 board (https://groupgets.com/manufacturers/getlab/products/purethermal-2-flir-lepton-smart-i-o-module) that was working perfectly in a RPi3B+, but I'm totally unsuccessful.

I've seen that whenever I try to get a frame from the camera, using any method, I get the following error in dmesg:
```
usb 1-1.3: Not enough bandwidth for new device state.
usb 1-1.3: Not enough bandwidth for altsetting 1
```
I've tested it in both the USB2 and the USB3 ports of the Raspberry, but I always get the same error.

The device is detected correctly: I can see it detected in dmesg, I can see the /dev/video ports being created, and I can even extract information with v4l2-ctl. The problem only appears when I open the port and try to grab a frame

**To reproduce**
Connect the PureThermal2 board to an USB port
Wait for it to be detected and ready
Try to grab a frame with any method: Python, Opencv, FFmpeg, guvcview...
No frame is obtained, and it throws that error

**System**
* Which model of Raspberry Pi?
RPi4 (4GB RAM)
* Which OS and version (`cat /etc/rpi-issue`)? 
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage4
* Which firmware version (`vcgencmd version`)?
version a49274f4de9406db851802c64ea2e8a13df7e0d3 (clean) (release) (start)
* Which kernel version (`uname -a`)?
Linux raspberrypi 4.19.63-v7l+ #1249 SMP Thu Aug 1 16:31:35 BST 2019 armv7l GNU/Linux

**Logs**
`dmesg` log after connecting the camera: [dmesg](https://pastebin.com/UM4Q3Q01)

**Additional context**
I received the Raspberry Pi 4 around 2 weeks ago, installed Raspbian-Buster, updated everything, and copied the python code from a RPi3B+ where the PureThermal 2 board was working to it, installed dependencies (python, opencv, ffmpeg) and tried to grab a frame. I've been unsuccessful until now.

agjunyent, 2019-08-20:
Updated to 4.19.66-v7l+ with rpi-update today. Still same exact issue.

I've read it is because the camera is reserving too much bandwith. It works at full speed (12MB) when connected, not High speed (480MB). I've tried setting some quirks of the uvcvideo module, but still failing. 

I've also tested the RPi4 with other USB cameras and they work fine.

Remember that everything is working fine under a Raspberry Pi 3B+

popcornmix, 2019-08-20:
@P33M any ideas?

P33M, 2019-08-20:
Please post a full `sudo lsusb -v` log.

agjunyent, 2019-08-21:
```
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0002 2.0 root hub
  bcdDevice            4.19
  iManufacturer           3 Linux 4.19.66-v7l+ dwc_otg_hcd
  iProduct                2 DWC OTG Controller
  iSerial                 1 fe980000.usb
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0019
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0008
    Ganged power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood        1 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0000
Device Status:     0x0001
  Self Powered

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
[... 697 lines cut ...]
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x02
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
Device Status:     0x0001
  Self Powered
```

P33M, 2019-08-21:
Ah. The connected port is in full-speed mode not high-speed. The max payload size for an isoc packet is 1023 bytes at full speed, but for some reason the bandwidth reservation is failing.

What happens if you plug the device into a  usb2.0 multi-TT hub and connect that hub to the Pi?

agjunyent, 2019-08-21:
Yes, the camera works in full speed. I've seen that on their repo, they say the camera works at Full Speed (12MB).

The problem is that it works perfectly in a Raspberry Pi 3B+ with exactly the same code being executed (a simple python script that opens a VideoCapture and grabs frames).

Here the output after plugging the camera on a HUB and to the Pi, still same exact error in dmesg:

```
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0002 2.0 root hub
  bcdDevice            4.19
  iManufacturer           3 Linux 4.19.66-v7l+ dwc_otg_hcd
  iProduct                2 DWC OTG Controller
  iSerial                 1 fe980000.usb
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0019
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0008
    Ganged power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood        1 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0000
Device Status:     0x0001
  Self Powered
[... 778 lines cut ...]
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x02
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
Device Status:     0x0001
  Self Powered
```

P33M, 2019-08-21:
From lsusb for the terminus tech hub:
```
  bDeviceProtocol         1 Single TT
```
You should find one that supports multi-TT operation.

agjunyent, 2019-08-21:
Oh, sorry

I'll try to find one asap.

agjunyent, 2019-08-23:
After trying 7 different hubs, all of them appear as "1 Single TT"... Do you have any model that surely is Multi-TT? In Spain at least I find impossible to get one...

albertfaromatics, 2019-08-23:
Applying the patch to the Camera firmware described in https://github.com/groupgets/purethermal1-firmware/issues/24#issuecomment-524177762 solves the issue for some pixel formats.

It reduces the video packet size to half, resulting in the RPi reserving the correct bandwidth then.

I don't know why, but all the other versions of Raspberry (at lest RPi2, and RPi3/RPi3 B+) work without this patch on the camera firmware.

kekiefer, 2019-08-24:
At least at interval 1, the stream bandwidth gets rejected at packet size somewhere between 642 and 802 bytes.

P33M, 2019-08-27:
I have a USB hub chip evaluation board that can be configured as a full-speed hub. If I plug the hub in, and then plug a cheap USB2.0 uvc webcam into the hub, I see the same issue.

Bandwidth calculations are dependent on whether hubs are multi-TT or single-TT, and the hub inside the USB3.0 controller is single-TT. The full-speed hub forces the TT to be used so my test just confirms the issue relates to large isochronous packet sizes rather than the Purethermal 2.

The largest isoc frame size for this camera in this mode is 956 bytes, with a 32-byte isoc endpoint for audio data and a 16-byte interrupt endpoint for events. The sum of all these is 1004 bytes, so it looks like the xhci driver might not be calculating bandwidth properly.

albertfaromatics, 2019-09-02:
Any news on this?
I understand from the last message that the issue is related to Kernel/driver itself, nothing to do with the USB cameras, right?
I can try to take a look to the problem, but I think I will need some guidance to start with it...

albertfaromatics, 2019-09-23:
Sorry for being a bit annoying, but I've been trying to look for where the possible error could be, but I'm totally unexperienced with kernels, and I don't know how to continue. 
Is there any planning on this, or should we work on updating all our cameras and use the patch deployed on them?
I'm thinking if this problem could also appear with other full-speed devices, making the RPi4 useless for us until a patch for this problem is deployed.

Thanks

P33M, 2019-09-23:
The issue does appear with other full-speed devices with large isochronous endpoints. Most devices have fall-back endpoint max-packet sizes that allow for reduced bandwidth reservations - but the IR cam has a single alternate setting.

From the video bitrates in the descriptors, it looks like all but the largest (160x120@24BPP) require less bandwidth than the 482-byte endpoint packet size provides.

I would recommend using the modified firmware, as I currently don't have a timescale for a fix.

albertfaromatics, 2019-09-23:
Understood!
I'll keep an eye here for when the patch is deployed.

Thanks P33M

P33M, 2019-09-27:
I can confirm that using a multi-tt hub doesn't solve the issue. The transfer is failing because the xhci controller is reporting "Secondary Bandwidth Error" when the endpoint gets enabled - implying the controller thinks that there is not enough periodic bandwidth available to add the requested transfer to the internal packet schedule. XHCI is supposed to have knowledge of hub topologies and maintain its own internal bandwidth allocations on a per-TT basis, so it appears that something is going wrong here.

I've noticed something odd about the XHCI controller - it claims to be a rev 1.0 supporting device, yet in the HCCPARAMS1 capabilities field it does not support Secondary Bandwidth Domain reporting - a required feature. 

I can more easily probe the reservation size that causes the controller to barf through the use of the gadget driver g_webcam on a pi zero - it lets you specify arbitrary maxpacket sizes on its isochronous endpoint.

P33M, 2020-03-27:
The xhci controller is reporting secondary bandwidth errors when a full-speed isochronous IN endpoint is configured that has a maxpacket size of >792 bytes.

I'm still investigating various configurations to see if this limit interacts with any other devices (i.e. if it's global or per-device), but I would recommend that, for now at least, you should ask the manufacturers of the camera to provide a firmware that limits the endpoint size to e.g. 768 bytes.

albertfaromatics, 2020-03-31:
Then I understand that is a problem from the xhci controller. The controller is specific for the RPI, because it is not happening on any other Linux device, no? I think that the Purethermal people have already deployed a patch that as you say limits the endpoint size. I'll keep roaming around here just to mantain myself and my company updated

P33M, 2020-04-17:
The issue is specific to the VIA PCIe-USB bridge chipset we are using on Pi 4. There is a firmware-imposed limitation of a total of 792 bytes of available full-speed Isochronous bandwidth shared across all attached devices.

I am in active discussions with the manufacturer about what can be done to remove this limit.

Thors161, 2020-07-01:
Stumbled upon this issue working on something similar, thank you for the multi-tt hub suggestion P33M. 
If of interest, I noticed the PureThermal2 1.3.0 FW has an extra altsetting with wMaxPacketSize 642 instead of 962 bytes, which is lower than the mentioned 792 limit (can be updated using dfu).

```
Bus 001 Device 113: ID 1e4e:0100 Cubeternet WebCam
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x1e4e Cubeternet
  idProduct          0x0100 WebCam
  bcdDevice            2.00
  iManufacturer           1 GroupGets
  iProduct                2 PureThermal (fw:v1.3.0)
  iSerial                 3 XXXXXXXX-5102-3038-3835-393400000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x02bd
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0 
      iFunction               2 PureThermal (fw:v1.3.0)
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      0 
      iInterface              0 
      VideoControl Interface Descriptor:
        bLength                13
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdUVC               1.00
        wTotalLength       0x0103
        dwClockFrequency        6.000000MHz
        bInCollection           1
        baInterfaceNr( 0)       1
      VideoControl Interface Descriptor:
        bLength                18
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0201 Camera Sensor
        bAssocTerminal          0
        iTerminal               0 
        wObjectiveFocalLengthMin      0
        wObjectiveFocalLengthMax      0
        wOcularFocalLength            0
        bControlSize                  3
[... 408 lines cut ...]
          Usage Type               Data
        wMaxPacketSize     0x0282  1x 642 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       2
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      2 Video Streaming
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x03c2  1x 962 bytes
        bInterval               1

```

andrewvaughanj, 2020-07-01:
I had this exact same issue (PT2 + RPi4, 8GB), and there a few similar issues for this in the PT repos:

- https://github.com/groupgets/purethermal1-firmware/issues/24

- https://github.com/groupgets/purethermal1-uvc-capture/issues/15

The solutions here talk about rebuilding/reflashing the latest firmware-

YMMV, but, for me, rebuilding and flashing the updated firmware 100% fixed things. It went from totally not working, to my old code (for an RPi3) working without modification!

If you can, I really suggest you try building the latest firmware and flashing it via DFU, because I think @kekiefer has resolved this 😄

P33M, 2020-07-01:
Oops I forgot to update this issue. There's a VLi firmware release that allows for use of up to 1023-byte isochronous endpoints here: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=273027

parkerlreed, 2025-01-05:
Apparently you can get similar symptoms if the module has become slightly unseated. Just a heads up for anyone else running across this.