raspberrypi/linux #3141: Please add the kernel module w1_ds28e17 into the standard kernel config

Fux2, 2019-08-11:
Please add the kernel module w1_ds28e17 into the standard kernel config.

I second the request from here: [https://www.raspberrypi.org/forums/viewtopic.php?t=248178](url)
As it seems the thread creator from the above link didn't open an issue, I am doing it now.

`modprobe w1_ds28e17 
modprobe: FATAL: Module w1_ds28e17 not found in directory /lib/modules/4.19.58-v7+`

pelwell, 2019-08-11:
Opening an issue is by far the best way to request a change like this, so thank you for doing so. The config change is now in rpi-4.19.y and will be picked up by future builds.

poproar, 2019-08-12:
Thanks for opening this issue.

JamesH65, 2019-12-20:
Closing this issue as questions answered/issue resolved.

closed 2019-12-20 by JamesH65