raspberrypi/linux #3142: (Pi4) Kernel 4.19.57~.65 Causing Some kind of Memory Deadlock

CaelThunderwing, 2019-08-11:
i Discovered this over at the LibreElec Forums working (well testing more or less) with "MagicSeb" on his build for LibreElec with RetroArch.

Something that started to plague it was kernel Panics/instability on a "Crash to Kodi" (like Kodi restarting almost in a loop,cant enter settings w/o a loop, trying to reboot the pi causing a Deadlock,) we tried narrowing it down to Stock "Lakka" Firmware... No dice.. tried Stock FW and kernel. BINGO.. we got somewhere..

the stock image of that point used 4.19.56

trying 4.19.57 n higher w/ or w/o updated Firmware was resulting in the same instability. we had others testing builds, it happened on 1GB models, 2GB Models and 4GB Models.

tried going back to 4.19.56 but updated firmware to latest (to fix Wifi issues) and we got stability back. I dont know how to track down the exact change/ Exact single commit for the Pi Kernel where this Broke.

but the instability issues causing the deadlocks after launching RetroArch would be either idle or Playing a game or launch game then pull up quick menu.. it didnt matter. it'd cause a "Deadlock on Memory" kernel panic after 2 to 2hrs n 30 mins.

Here's the crashes i captured (sorry crappy phone is crappy, and i was using n old pizza box to prop the screen, i dont have stands for this it was a quick 3D Print a friend did for me,)
(this one happened with RetroArch after the 2hr 30 min mark)
[image]

This was from trying later that day trying to reboot after a crash to Kodi
[image]

the Kernel in use at the time of this was 4.19.64

JamesH65, 2019-12-20:
@CaelThunderwing Is this issue still present on the latest kernel/firmware?

CaelThunderwing, 2019-12-27:
sorry i had been busy,  from a brief test  with last release by MagicSeb (he maintains a version of libreElec+Retroarch for the Pi4, and i saw no crashes, infact the only remaining problem is unrelated to this bug. which he has moved to kernel 5.4