raspberrypi/linux #3145: RPi 3B 4.16.63 system freezes with USB printer connected and printing

OliverGruenberg, 2019-08-12:
RPi 3B
4.19.63

Entire system freezes and does not respond after some minutes of using the printer.

dmesg outputs:
[  +0.064324] usb 1-1.3: New USB device found, idVendor=04a9, idProduct=32db, bcdDevice= 0.01
[  +0.000016] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000010] usb 1-1.3: Product: SELPHY CP1300
[  +0.000008] usb 1-1.3: Manufacturer: Canon
[  +0.000009] usb 1-1.3: SerialNumber: B110090201509819
...
[  +0.060121] usblp 1-1.3:1.0: usblp0: USB Bidirectional printer dev 5 if 0 alt 0 proto 2 vid 0x04A9 pid 0x32DB
...
[Aug 8 10:49] usb 1-1.3: usbfs: usb_submit_urb returned -121
[  +0.000174] usb 1-1.3: usbfs: usb_submit_urb returned -121
[  +0.000124] usb 1-1.3: usbfs: usb_submit_urb returned -121
[  +0.000124] usb 1-1.3: usbfs: usb_submit_urb returned -121
[  +0.000130] usb 1-1.3: usbfs: usb_submit_urb returned -121
[  +0.000137] usb 1-1.3: usbfs: usb_submit_urb returned -121
[  +0.000143] usb 1-1.3: usbfs: usb_submit_urb returned -121
[  +0.000125] usb 1-1.3: usbfs: usb_submit_urb returned -121
[  +0.000123] usb 1-1.3: usbfs: usb_submit_urb returned -121
[  +0.000123] usb 1-1.3: usbfs: usb_submit_urb returned -121
[  +0.000127] usb 1-1.3: usbfs: usb_submit_urb returned -121
[  +0.000128] usb 1-1.3: usbfs: usb_submit_urb returned -121
[  +0.000127] usb 1-1.3: usbfs: usb_submit_urb returned -121
[  +0.000123] usb 1-1.3: usbfs: usb_submit_urb returned -121
[  +0.000123] usb 1-1.3: usbfs: usb_submit_urb returned -121
[  +0.000134] usb 1-1.3: usbfs: usb_submit_urb returned -121
...
+0.000157] NYET/NAK/ACK/other in non-error case, 0x00000002
[  +0.000017] NYET/NAK/ACK/other in non-error case, 0x00000002
[  +0.000054] usb 1-1.3: usbfs: usb_submit_urb returned -121
[  +0.000080] NYET/NAK/ACK/other in non-error case, 0x00000002
[  +0.000020] NYET/NAK/ACK/other in non-error case, 0x00000002
[  +0.000018] NYET/NAK/ACK/other in non-error case, 0x00000002
[  +0.000023] usb 1-1.3: usbfs: usb_submit_urb returned -121
...
[  +0.071027] usb 1-1.3: USB disconnect, device number 5

Please see entire dmesg output attached.
[k4-19-63-dmesg-freeze.log](https://github.com/raspberrypi/linux/files/3491920/k4-19-63-dmesg-freeze.log)

P33M, 2019-08-21:
Is the failure specific to the version of kernel (i.e. is there any previous version that worked)? If so, which is the version that causes the failure?

OliverGruenberg, 2019-11-14:
We have tried different kernel versions and different printers.
The problem occurs with some of the printers (and we tried many) but with any kernel version.
I assume that some printers require too much current from the USB bus?
Or timing problems between the devices?

A workaround at the moment is using a passive USB hub, seems like it kind of manages the talk between the Raspberry Pi and the printer.
But shouldn't it be possible to implement this behaviour into software?

JamesH65, 2019-12-20:
@p33m This sounds like a limitation of the 3B USB hardware so not much that can be done?