raspberrypi/linux #3146: Kernel 5.3.0-rc3 will not compile

Dark-Sky, 2019-08-12:
Tried to compile aarch64 using latest pull as of today. Cross compiling with gcc 9.1.0. The compile errors out trying to compile hifiberry_dacplus.
 
`  CC [M]  sound/soc/bcm/hifiberry_dacplus.o
sound/soc/bcm/hifiberry_dacplus.c:276:3: error: 'struct snd_soc_dai_link' has no member named 'cpu_dai_name'
  276 |  .cpu_dai_name = "bcm2708-i2s.0",
      |   ^~~~~~~~~~~~
sound/soc/bcm/hifiberry_dacplus.c:276:18: error: initialization of 'struct snd_soc_dai_link_component *' from incompatible pointer type 'char *' [-Werror=incompatible-pointer-types]
  276 |  .cpu_dai_name = "bcm2708-i2s.0",
      |                  ^~~~~~~~~~~~~~~
sound/soc/bcm/hifiberry_dacplus.c:276:18: note: (near initialization for 'snd_rpi_hifiberry_dacplus_dai[0].cpus')
sound/soc/bcm/hifiberry_dacplus.c:277:3: error: 'struct snd_soc_dai_link' has no member named 'codec_dai_name'
  277 |  .codec_dai_name = "pcm512x-hifi",
      |   ^~~~~~~~~~~~~~
sound/soc/bcm/hifiberry_dacplus.c:277:20: warning: initialization of 'unsigned int' from 'char *' makes integer from pointer without a cast [-Wint-conversion]
  277 |  .codec_dai_name = "pcm512x-hifi",
      |                    ^~~~~~~~~~~~~~
sound/soc/bcm/hifiberry_dacplus.c:277:20: note: (near initialization for 'snd_rpi_hifiberry_dacplus_dai[0].num_cpus')
sound/soc/bcm/hifiberry_dacplus.c:277:20: error: initializer element is not computable at load time
sound/soc/bcm/hifiberry_dacplus.c:277:20: note: (near initialization for 'snd_rpi_hifiberry_dacplus_dai[0].num_cpus')
sound/soc/bcm/hifiberry_dacplus.c:278:3: error: 'struct snd_soc_dai_link' has no member named 'platform_name'; did you mean 'platforms'?
  278 |  .platform_name = "bcm2708-i2s.0",
      |   ^~~~~~~~~~~~~
      |   platforms
sound/soc/bcm/hifiberry_dacplus.c:278:19: error: initialization of 'struct snd_soc_dai_link_component *' from incompatible pointer type 'char *' [-Werror=incompatible-pointer-types]
  278 |  .platform_name = "bcm2708-i2s.0",
      |                   ^~~~~~~~~~~~~~~
sound/soc/bcm/hifiberry_dacplus.c:278:19: note: (near initialization for 'snd_rpi_hifiberry_dacplus_dai[0].codecs')
sound/soc/bcm/hifiberry_dacplus.c:279:3: error: 'struct snd_soc_dai_link' has no member named 'codec_name'
  279 |  .codec_name = "pcm512x.1-004d",
      |   ^~~~~~~~~~
sound/soc/bcm/hifiberry_dacplus.c:279:16: warning: initialization of 'unsigned int' from 'char *' makes integer from pointer without a cast [-Wint-conversion]
  279 |  .codec_name = "pcm512x.1-004d",
      |                ^~~~~~~~~~~~~~~~
sound/soc/bcm/hifiberry_dacplus.c:279:16: note: (near initialization for 'snd_rpi_hifiberry_dacplus_dai[0].num_codecs')
sound/soc/bcm/hifiberry_dacplus.c:279:16: error: initializer element is not computable at load time
sound/soc/bcm/hifiberry_dacplus.c:279:16: note: (near initialization for 'snd_rpi_hifiberry_dacplus_dai[0].num_codecs')
sound/soc/bcm/hifiberry_dacplus.c: In function 'snd_rpi_hifiberry_dacplus_probe':
sound/soc/bcm/hifiberry_dacplus.c:310:7: error: 'struct snd_soc_dai_link' has no member named 'cpu_dai_name'
  310 |    dai->cpu_dai_name = NULL;
      |       ^~
sound/soc/bcm/hifiberry_dacplus.c:310:4: warning: statement with no effect [-Wunused-value]
  310 |    dai->cpu_dai_name = NULL;
      |    ^~~
sound/soc/bcm/hifiberry_dacplus.c:311:7: error: 'struct snd_soc_dai_link' has no member named 'cpu_of_node'
  311 |    dai->cpu_of_node = i2s_node;
      |       ^~
sound/soc/bcm/hifiberry_dacplus.c:311:4: warning: statement with no effect [-Wunused-value]
  311 |    dai->cpu_of_node = i2s_node;
      |    ^~~
sound/soc/bcm/hifiberry_dacplus.c:312:9: error: 'struct snd_soc_dai_link' has no member named 'platform_name'; did you mean 'platforms'?
  312 |    dai->platform_name = NULL;
      |         ^~~~~~~~~~~~~
      |         platforms
sound/soc/bcm/hifiberry_dacplus.c:312:4: warning: statement with no effect [-Wunused-value]
  312 |    dai->platform_name = NULL;
      |    ^~~
sound/soc/bcm/hifiberry_dacplus.c:313:7: error: 'struct snd_soc_dai_link' has no member named 'platform_of_node'
  313 |    dai->platform_of_node = i2s_node;
      |       ^~
sound/soc/bcm/hifiberry_dacplus.c:313:4: warning: statement with no effect [-Wunused-value]
  313 |    dai->platform_of_node = i2s_node;
      |    ^~~
cc1: some warnings being treated as errors
`

popcornmix, 2019-08-12:
The 5.3-rc3 tree is very experimental - major things won't work, and 64-bit is something we haven't tested at all.
If you want it to compile you'll currently need to disable the I2S audio cards which have broken due to a non-trivial API change. See: https://github.com/raspberrypi/linux/commit/64c57f5637186a42af2bee9d42af4ada3a7704c3

You'll probably find fkms won't work (it's not currently for 32-bit kernel).

n00b7, 2019-08-18:
Outdated, rc4 is out now try to recompile

Dark-Sky, 2019-08-19:
Thanks for the update. I do not use those cards and disabled all of them in that section. It was not a big deal to me disabling them but what is a big deal to me is the vkms-v3d not working so until then I will use the 4.19 branch with AARCH64.

https://github.com/raspberrypi/linux/issues/3162

Dark-Sky, 2019-08-29:
> Outdated, rc4 is out now try to recompile

Compiled -rc6 today and still errors out with the I2S audio cards enabled.

JamesH65, 2019-12-20:
@Dark-Sky Is this still an issue with the latest kernel?

Dark-Sky, 2019-12-20:
@JamesH65 

No they i2c cards compile just fine but regarding the title Kernel will not compile yes that was an issue in the latest kernel when I tried to cross compile with gcc 9.2. I can cross compile it with gcc 8.3.1 though if that is what you are refering to. Per popcornmix request I compiled the upstream v5.4 and the latest 5.5-rc trees and gcc 9.2 still does not work. That part looks like an upstream issue to me.

https://github.com/raspberrypi/linux/issues/3377#issuecomment-567568270