raspberrypi/linux #3149: fbtft does not work on kernel 5.2.0 and above

alexandref75, 2019-08-14:
A fbtft, more specifically a pitft 2.8" resistive (uses the fb_ili9340 driver does not work with kernels 5.2.0 and above. Images do not appear on the display.

**Describe the bug**
Changes to internal kernel API of gpiod invalidates that way that fbtft identifies the GPIO pins. Without the dc-gpio pin, the SPI interface to the display does not work.

**To reproduce**
Connect a fbtft display that uses GPIO pins and boot a 5.2.X kernel or above, use the following command to present an image at the display:
fbi -T 2 -d /dev/fb1 -noverbose -a adapiluv320x240.jpg

**Expected behaviour**
An framebuffer that presents the image at the display.

**Actual behaviour**
The backlight and the display remain off and no image is shown.

**System**
pi@raspberrypi1:~ $ raspinfo 
System Information
------------------

Raspberry Pi Model B Rev 2
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2019-06-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 150e25c4f8123a4c9c63e8dca1b4737fa6c1135c, stage2

Linux raspberrypi1 5.2.6+ #23 Tue Aug 13 21:00:43 CDT 2019 armv6l GNU/Linux
Hardware        : BCM2835
Revision        : 000e
Serial          : 000000008ceda0f9
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Jul  9 2019 14:40:53 
Copyright (c) 2012 Broadcom
version 6c3fe3f096a93de3b34252ad98cdccadeb534be2 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root        7429824 1686896   5408336  24% /
devtmpfs          217628       0    217628   0% /dev
tmpfs             221960       0    221960   0% /dev/shm
tmpfs             221960    3092    218868   2% /run
tmpfs               5120       0      5120   0% /run/lock
tmpfs             221960       0    221960   0% /sys/fs/cgroup
/dev/mmcblk0p1    258096   44152    213944  18% /boot
tmpfs              44392       0     44392   0% /run/user/1000

Filename                                Type            Size    Used    Priority
/var/swap                               file            102396  0       -2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: (none)
raspberrypi-sys-mods:
  Installed: 20190730
openbox:
  Installed: (none)
[... 388 lines cut ...]
[   19.000759] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[   19.056368] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
[   27.236682] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[   27.406241] fb_ili9340: module is from the staging directory, the quality is unknown, you have been warned.
[   27.436429] fbtft_of_value: buswidth = 8
[   27.436463] fbtft_of_value: debug = 0
[   27.436485] fbtft_of_value: rotate = 90
[   27.436503] fbtft_of_value: fps = 30
[   27.436951] fb_ili9340 spi0.0: Failed to request dc-gpios GPIO:-2
[   27.437155] fb_ili9340: probe of spi0.0 failed with error -2
[   28.217859] stmpe-ts stmpe-ts: DMA mask not set
[   28.282698] input: stmpe-ts as /devices/platform/soc/20204000.spi/spi_master/spi0/spi0.1/stmpe-ts/input/input0
[   31.164353] random: crng init done
[   31.164387] random: 7 urandom warning(s) missed due to ratelimiting
[   32.153048] 8021q: 802.1Q VLAN Support v1.8
[   33.375783] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   34.329002] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   34.367140] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   34.838188] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   36.331010] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC5E1

**Patch**

The attached patch fixes the problem. It is tested on kernel 5.2.6.
[0001-Fix-fbtft-for-kernels-5.2.0-and-above.patch.txt](https://github.com/raspberrypi/linux/files/3499568/0001-Fix-fbtft-for-kernels-5.2.0-and-above.patch.txt)

pelwell, 2019-08-14:
Thanks - that looks good, apart from one or two cosmetic issues that I can fix. Do you have a Signed-off-by line I may add?

pelwell, 2019-08-14:
Reading through the patch again, I think it should be split in two: a one-line fix that removes the spurious `!` in `if (!par->gpio.dc)`, and a cleanup patch. Have you tried with just the one-line fix?

lategoodbye, 2019-08-14:
Looks like a squash of some upstream commits:
Staging: fbtft: Fix probing of gpio descriptor
Staging: fbtft: Fix reset assertion when using gpio descriptor
Staging: fbtft: Fix GPIO handling

So they might be backported soon.

pelwell, 2019-08-14:
I'll hold off on a push/merge.

alexandref75, 2019-08-14:
Wow. i should improve my searching, it would have saved a lot of debugging. the commits are much better than mine so this issue can be closed. the commits are from jun/11. it is already on 5.3.0-rc4 but i could not find references of backporting.

lategoodbye, 2019-08-14:
The patches has been applied at the end of July, so it will take some time until they backported.

pelwell, 2019-09-02:
The back-ports  have now appeared in 5.2:
* 3558601
* 1dd12a5
* 46af9cb

Please retest, and close if appropriate.

JamesH65, 2019-09-17:
@alexandref75 Can this be closed?

alexandref75, 2019-09-17:
Yes, it can be closed. Sorry for the delay.

closed 2019-09-17 by alexandref75