raspberrypi/linux #3152: RPi4 I2C HDMI interface (/dev/i2c-2) not working from userspace

bangom, 2019-08-14:
Hi,

I'm in the process of migrating a project from RPI3b to RPI4 (4GB) due to a better GPU performance of the new platform.

One of the functions on RPI3 was turning on/of and changing the brightness of HDMI connected LCD via ddcutil command (details see https://www.ddcutil.com/raspberry/).

Unfortunately, on RPI4 I'm struggling with accessing /dev/i2c-2 HDMI1 I2C from userspace:
- "ddcutil detect" command takes long, then outputs "No displays found".
- "i2cdetect -y 2" also takes long to finish, there are no devices on the bus.

After either of those commands"dmesg" shows new errors:
[61839.796098] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[61840.836046] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
[61854.116299] i2c-bcm2835 fe805000.i2c: i2c transfer timed out
(error messages repeat as you execute more commands using /dev/i2c-2)

Seems that there is a problem accessing i2c HDMI bus via /dev/i2c-2 from userspace.

root@mm-rpi4:/home/pi# ls -la /dev/i2c*
crw-rw---- 1 root i2c 89, 0 Aug 13 17:10 /dev/i2c-0
crw-rw---- 1 root i2c 89, 1 Aug 13 17:10 /dev/i2c-1
crw-rw---- 1 root i2c 89, 2 Aug 13 17:10 /dev/i2c-2

uname -a:
Linux mm-rpi4 4.19.65-v7l+ #1252 SMP Fri Aug 9 18:04:13 BST 2019 armv7l GNU/Linux

vcgencmd version: 
Copyright (c) 2012 Broadcom
version 9f8431fb7839c7f00f52b81f5822ddab2b31d0db (clean) (release) (start)

ddcutil -V:
ddcutil 0.9.5

GPU i2c access shoud be disabled according to the /boot/config.txt:
hdmi_ignore_edid=0xa5000080 # (tried also =1)
hdmi_edid_file=1
hdmi_ignore_cec=1
dtparam=i2c2_iknowwhatimdoing
dtoverlay=vc4-fkms-v3d

Seems like something is blocking access via /dev/i2c-2. Either there is a wrong mapping of /dev/i2c-2 to HDMI ports (also there are 2 HDMI ports in new RPI4...) or GPU have "exclusive access" to i2c disallowing userspace access?

Any idea how to fix the problem? Thanks!

timg236, 2019-08-14:
i2c-1 and i2c-2 pointer to registers which no longer exist on 2711, we will remove those.

On 2711 each HDMI block has a dedicated I2C controller with a different interface to the 2835 I2C generic controller. In theory this could be exposed to Linux and a new driver written although it would need to play nicely with firmware DDC reading etc.

However, in practice that might be problematic because when 4Kp60 is enabled the HDMI hardware state machine gets exclusive access to the I2C controller for maintaining the SCDC (Status and Control Data Channel) so the ARM/VPU lose access.

bangom, 2019-08-14:
OK, I don't need 4Kp60 for this, 1080p on one display is sufficient. Would it be possible to access HDMI i2c from userspace as on old RPI3 so we can use ddcutil?

timg236, 2019-08-14:
Possibly, but it's not likely to happen soon.

bangom, 2019-08-14:
That's unfortunate. Could you give me some hints then so I can start making my own patch? Exposing HDMI1 / HDMI2 i2c registers via /dev/ and disabling HDMI DDC readings (like hdmi_ignore_edid option in RPI3 config.txt)? Where can I find some info?

timg236, 2019-08-14:
@lurch Perhaps the HDMI I2C could be included in the datasheet ?

bangom, 2019-08-19:
> @lurch Perhaps the HDMI I2C could be included in the datasheet?

Let me know when we can continue, I want to move forward with my project.

In phase1 I want to make a quick module fix, so:
- /dev/ic2-2 point to correct i2c register of HDMI0 port,
- hdmi_ignore_edid parameter in config.txt disables GPU DDC reading of HDMI0 port.

Could you please share details for RPi4 so I can proceed?

lurch, 2019-08-19:
We've had an internal discussion, and exposing I2C2 to user-code is fairly low priority at the moment (so I won't be adding it to the datasheet).

timg236, 2019-08-28:
Accessing HDMI I2C user from user-space is not going to work without firmware changes and there are other higher priority issues being worked on right now.

6by9, 2019-08-28:
Please reread the page you referred to.
It references using ```dtoverlay=vc4-kms-v3d```, which is where Linux has taken full control of i2c2, HDMI hardware, pixel valves, HVS, and various other things. This driver is not ready for Pi4 - you are running vc4-fkms-v3d (f for firmware) where the gpu firmware is still in control of all those hardware blocks. There is no inter-processor arbitration, therefore any attempt to do that will fail as the gpu will be waiting for those interrupts, and is likely to have lower interrupt latency than the ARM.

vc4-kms-v3d will come for the Pi4, but as timg236 has said, it's not at the top of the priority list yet.

bangom, 2019-08-28:
> We've had an internal discussion, and exposing I2C2 to user-code is fairly low priority at the moment (so I won't be adding it to the datasheet).

Do I understand this correctly that raspberrypi team doesn't want to release specification of HDMI DDC-CI to I2C_2835 bridge publicly so we can implement "RPI4 HDMI I2C user-space device" ourselves?

6by9, 2019-12-29:
vc4-kms-v3d for Pi4 is being worked on, and the basics (including i2c) are up and running. I'd hope we'd be looking at alpha release some time in January.

bangom, 2019-12-29:
If you need to control HDMI connected LCD on RPI4 via CEC commands (for example ddcutil) and can't wait for RPI4 drivers fix, you can try my "HDMI connector split" workaround which I've used for +3 months with no issues. You will need to solder a little bit...

Split HDMI cable from LCD into two branches:

a) CEC branch which consists of HDMI Pins 15-18:
https://de.wikipedia.org/wiki/Datei:HDMI_Connector_Pins1.png

Important: you need to use 5V-to-3.3V I2C level shifter because HDMI uses 5V but RPI 3.3V levels:

[ HDMI monitor port ] _______/----- [ CEC branch ] ----- [ 5to3VlevelShifter ] ----- [ RPI4-I2C-busX ]

b) HDMI branch - rest of the HDMI pins. I've soldered them back to new HDMI A male connector according to HDMI connector pinouts.

Then you can drive LCD HDMI monitor from you RPI4 via ddcutil with --bus <i2c bus number> argument. <i2c bus number> is the I2C RPI4 bus where you've connected CEC branch from split HDMI cable.

6by9, 2020-07-10:
The early version was merged to our rpi-5.4.y kernel branch via #3515 back in March. That was v1 of the patches sent to the relevant kernel mailing lists.
v4 was sent on Wednesday, and I'm about to backport those patches to our branch.

If you have the 5.4 branch (available via rpi-update), then adding ```dtoverlay=vc4-kms-v3d``` will load the driver, and you should get /dev/i2c11 and /dev/i2c12 as the two I2C interfaces on the HDMI ports.

Please note that It is still in beta and being reviewed by mainline. There are known issues with the v1 patchset that can cause issues with the display not coming up and similar, so please be warned.

6by9, 2020-07-10:
Minor observation. i2cdetect uses different mechanisms to "detect" I2C devices based on the address (0x30-0x37 and 0x50-0x5F use READ, otherws use QUICK).

It looks like only READ mode works at present on this controller, so use ```i2cdetect -r -y 11``` to avoid getting incorrect detections.

samuele2723, 2020-08-24:
hello, is there any update on the RPi support for DDC\CI util? or everyone should go for the hardware solution of bangom.

@Everyone. Any tried to replicate the hardware solution? any picture step by step?

samuele2723, 2020-08-24:
Hello, i installed the 5.4 kernel and now i have the `dtoverlay=vc4-kms-v3d` options. Nevertheless i don't get any /dev/i2c11 and /dev/i2c12.
Also, using `ddcutil` command that's my goal, doesn't work still on rasp4

dawidtomczyk, 2020-12-12:
What's the current status of the implementation?

6by9, 2020-12-12:
Vc4-kms-v3d is working on pi4. Almost all changes are merged into the mainline kernel, and those that aren't have been merged to our rpi-5.10.y branch. BRANCH=next rpi-update will get you a build of that 5.10 branch. 

The observation at https://github.com/raspberrypi/linux/issues/3152#issuecomment-656665996 still applies.

dawidtomczyk, 2020-12-16:
I'm able to switch my monitor input source using ddcutil without an issue now. Thanks for your work @6by9!

mayurankv, 2020-12-19:
dawidtomczyk, can I just confirm that ddcutil is working for you on a pi 4 without any extra workarounds?

raomin, 2021-01-27:
Just to make it easier for everyone, a quick step by step: 
1. run `BRANCH=next rpi-update`, this should install rpi firmware 5.10
2. in `/boot/config.txt` look for the `[pi4]` section and set this value `dtoverlay=vc4-kms-v3d`
3. reboot
4. `ddcutil detect` should now detect your screen (is ddc capable of course...)

6by9, 2021-01-27:
@bangom As the OP are you happy to close now that the HDMI I2C interfaces (plural) are now working on Pi4?

AlonsoAlv, 2021-02-02:
> Just to make it easier for everyone, a quick step by step:
>
> 1. run `BRANCH=next rpi-update`, this should install rpi firmware 5.10
> 2. in `/boot/config.txt` look for the `[pi4]` section and set this [...]

I have followed the instructions, but when in  `/boot/config.txt` I change `dtoverlay=vc4-fkms-v3d` (with f) to `dtoverlay=vc4-kms-v3d` (without f) no image is shown after booting up, no matter which monitor I try.
Commenting the rest of parameters on `config.txt` doesn't seem to work either.
I don't know what happens because the raspberry detects the monitor while booting up, but after seeing the rainbow and the "Raspberry Pi OS" images, the monitor says "no signal" and shuts itself down. Any help?

6by9, 2021-02-03:
Do not use ```BRANCH=next``` in ```rpi-update``` now. The 5.10 kernel has been in the standard rpi-update branch since 15th Dec.
Using it will get you an older kernel than using a plain ```sudo rpi-update```

AlonsoAlv, 2021-02-03:
It worked!
Thank you very much for answering!

Nedial, 2021-02-23:
I am trying to get ddcutil to work on my Raspberry 4.
Instead of "rpi-update" I used "sudo apt update" and "sudo apt full-upgrade". I understood it to give me the latest version. Or is that wrong?

The command "uname -a" gives the following result: 
Linux raspberrypi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
Does this give me the version that ddcutil should work on?

I changed the entry under [pi4] in /boot/config.txt to "dtoverlay=vc4-kms-v3d". After booting, I no longer had a picture and could not get to the Raspberry over the network via console. I changed the config.txt on another computer back to the original state and then it worked again.

mattncsu, 2021-03-10:
> 
> 
> Just to make it easier for everyone, a quick step by step:
> 
>     1. run `rpi-update`, this should install rpi firmware 5.10
> 
>     2. in `/boot/config.txt` look for the `[pi4]` section and set this value `dtoverlay=vc4-kms-v3d`
> 
>     3. reboot
> 
>     4. `ddcutil detect` should now detect your screen (is ddc capable of course...)
> 

This worked for me, thanks! And thanks for the developers for adding HDMI I2C support!

6by9, 2021-03-10:
There is no need to run rpi-update. The 5.10 kernel is now shipped through the standard Raspberry Pi OS repos.

6by9, 2021-03-10:
Issue resolved. Closing.

closed 2021-03-10 by 6by9

Nedial, 2021-03-20:
Sorry, I have to ask for help again. I have on my Raspberry 4 the version "Linux raspberrypi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux". If I change the entry in /boot/config.txt in the "[pi4]" section to "dtoverlay=vc4-kms-v3d" and reboot, part of the startup process appears on the screen, then the screen stays dark and I can't reach the Raspberry via WLAN either. The entry in the /boot/config.txt I have edited via "nano", so not in an incompatible editor, that's not the reason. The entry is definitely set correctly. Is there any idea what else could be the problem?
Thank you very much.

pelwell, 2021-03-20:
Open a new issue for a different problem.

Nedial, 2021-03-20:
For my configuration, the instructions seem to be wrong. I could solve the problem. 
I have in config.txt not only in section "[pi4]" but also in section "[all]" the entry "dtoverlay=vc4-fkms-v3d". In both sections the entry must be set to "dtoverlay=vc4-kms-v3d" for it to work. Possibly this affects others, so maybe it is helpful.

robertalexa, 2021-03-26:
@6by9 Hi, sorry for pestering you via tagging, but i am super confused, i was wondering if you could clarify.

To my knowledge, kernel 5.10 should have fixed this issue and ddcutil should now be working. However the [documentation](https://www.ddcutil.com/raspberry/) for ddcutil raspberry still says that it is unavailable for RPI4.

I am to assume that this is actually correct and the documentation above mentioned is outdated?

> Just to make it easier for everyone, a quick step by step:
> 
> 1. run `rpi-update`, this should install rpi firmware 5.10
> 2. in `/boot/config.txt` look for the `[pi4]` section and set this value `dtoverlay=vc4-kms-v3d`
> 3. reboot
> 4. `ddcutil detect` should now detect your screen (is ddc capable of course...)
> 
> edit: removed `BRANCH=next`

Best wishes,
Rob

6by9, 2021-03-26:
Third party documentation - not our problem.

5.10 is now the standard kernel so no need to use rpi-update, just ```sudo apt update``` ```sudo apt dist-upgrade```.

Steps 1-3 will give you /dev/i2c-11 and /dev/i2c-12 that can be used to read the EDID etc.

How a 3rd party app makes use of them again isn't a problem for the kernel to solve. (```ddcutil -b 11``` looks like the appropriate invocation to use /dev/i2c-11).

Note that the nodes can move to /dev/i2c-12 and /dev/i2c-13 if ```dtparam=i2c_vc=on``` is also present in config.txt.

robertalexa, 2021-03-29:
Thanks to the above conversation, I have done some testing, got in touch with the ddcutil people and the above mentioned documentation has been updated.

Hopefully this will help others too.

ybm911, 2022-04-24:
> I have followed the instructions, but when in `/boot/config.txt` I change `dtoverlay=vc4-fkms-v3d` (with f) to `dtoverlay=vc4-kms-v3d` (without f) no image is shown after booting up, no matter which [...]

I have the Same thing

dfanica, 2023-01-20:
Updating `dtoverlay=vc4-fkms-v3d` to `dtoverlay=vc4-kms-v3d` (removing the `f`) from `[pi4]` section under `/boot/config.txt` solved my issue 👍 Thanks @raomin

cagnulein, 2024-06-05:
guys what about the pi5? I can't get it to work

cagnulein, 2024-06-05:
oh yeah fixed, it was a xorg issue after all. thanks for the confirmation