raspberrypi/linux #3153: Compiling kernel for Raspberry Pi (32) breaks camera

agherzan, 2019-08-14:
**Describe the bug**
I am facing a strange behavior trying to take advantage of the camera on Raspberry Pi 3 (32bit). For this test case I used `raspberrypi-kernel_1.20190709-1` but this stands on newer tags as well. Compiled a minimum image with Yocto (latest firmware as well). Now with this, `raspistill` hangs with:

```
openat(AT_FDCWD, "/dev/vcsm", O_RDWR|O_LARGEFILE) = 4
ioctl(3, _IOC(_IOC_NONE, 0xc4, 0xc, 0), 0x2900f) = 0
ioctl(3, _IOC(_IOC_WRITE, 0xc4, 0x4, 0xc), 0x7ee97290) = 0
futex(0x76f89044, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 1, NULL, FUTEX_BITSET_MATCH_ANY
```

Full log for above [here](https://github.com/raspberrypi/linux/files/3502579/raspistill-doesn-t-work.txt).

Now, I took a Raspbian image that I had around and that uses exactly same kernel version. Overwritten the kernel and kernel modules from Raspbian to Yocto image and everything works as expected.

What I've checked is a diff of kernel config which is almost the same:

```
[andrei@fluxx ~/Downloads/cam]$ diff config-works.txt config-doesn-t-work.txt 
7c7
< # Compiler: arm-linux-gnueabihf-gcc (crosstool-NG crosstool-ng-1.22.0-88-g8460611) 4.9.3
---
> # Compiler: arm-poky-linux-gnueabi-gcc (GCC) 9.1.0
10c10
< CONFIG_GCC_VERSION=40903
---
> CONFIG_GCC_VERSION=90100
21c21
< CONFIG_LOCALVERSION="-v7"
---
> CONFIG_LOCALVERSION=""
617c617
< CONFIG_PLUGIN_HOSTCC="g++"
---
> CONFIG_PLUGIN_HOSTCC=""
619d618
< # CONFIG_GCC_PLUGINS is not set
6333a6333,6334
> CONFIG_CC_HAS_SANCOV_TRACE_PC=y
> # CONFIG_KCOV is not set
```

I also diffed dmesg and I can't see anything obvious there either. Also, `lsmod` has the same kernel modules loaded.

JamesH65, 2019-09-17:
I cannot see anything untoward in that strace, and TBH, for a non-Raspbian build we won't be putting any time in to tracking things down. It'll be either a missing library or a device tree problem somewhere I would think.

pelwell, 2019-09-17:
> Now, I took a Raspbian image that I had around and uses exactly same kernel version. Overwritten the kernel and kernel modules (from Raspbian to Yocto image) and everything works as expected.

To be sure I've understood you correctly, can you show the results of all four combinations of kernel (Raspbian, yours) and OS (Raspbian, Yocto)?

agherzan, 2019-09-24:
We have tested the latest kernel and firmware and we couldn't reproduce anymore. Closing this.

closed 2019-09-24 by agherzan