raspberrypi/linux #3154: tcpdump vlan problems with lan78xx driver?

gopstein, 2019-08-15:
I was debugging code which ran on one RPi 3B+, but failed on a different 3B+. Looks like the Pi with the problem has the LAN7800 ethernet chip, but the working one has the SMSC9512/9514 Fast Ethernet Adapter.

The scenario is that we run "tcpdump -i eth0 -e vlan" on a trunk port of a network which has vlans. On the working RPi (and also on an x86 based Ubuntu system), tcpdump displays packets with a vlan tag. On the Pi with the LAN7800, no packets are ever displayed. We also tried booting the failing Pi off the working Pi's sdcard. The failing RPi still would not display packets, hence it appearing to be a HW/driver issue.

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.58-v7+ #1245 SMP Fri Jul 12 17:25:51 BST 2019 armv7l GNU/Linux

pi@raspberrypi:~ $ dmesg | grep 78xx
[ 0.324777] usbcore: registered new interface driver lan78xx
[ 4.280187] lan78xx 1-1.1.1:1.0 (unnamed net_device) (uninitialized): No External EEPROM. Setting MAC Speed
[ 4.299811] libphy: lan78xx-mdiobus: probed
[ 4.338242] lan78xx 1-1.1.1:1.0 (unnamed net_device) (uninitialized): int urb period 64

pi@raspberrypi:~ $ tcpdump -e vlan
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on eth0, link-type EN10MB (Ethernet), capture size 262144 bytes

pi@raspberrypi:~ $ tcpdump --version
tcpdump version 4.9.2
libpcap version 1.8.1
OpenSSL 1.1.1c 28 May 2019

pi@raspberrypi:~ $ lsmod
Module Size Used by
nfnetlink_queue 24576 0
nfnetlink_log 20480 0
nfnetlink 16384 2 nfnetlink_log,nfnetlink_queue
rfcomm 49152 4
bnep 20480 2
hci_uart 40960 1
btbcm 16384 1 hci_uart
serdev 20480 1 hci_uart
bluetooth 389120 29 hci_uart,bnep,btbcm,rfcomm
ecdh_generic 28672 1 bluetooth
fuse 110592 3
8021q 32768 0
garp 16384 1 8021q
stp 16384 1 garp
llc 16384 2 garp,stp
evdev 24576 2
joydev 20480 0
fb_ili9486 16384 2
fbtft 45056 1 fb_ili9486
syscopyarea 16384 1 fbtft
sysfillrect 16384 1 fbtft
sysimgblt 16384 1 fbtft
ads7846 24576 0
fb_sys_fops 16384 1 fbtft
brcmfmac 311296 0
brcmutil 16384 1 brcmfmac
raspberrypi_hwmon 16384 0
sha256_generic 20480 0
hwmon 16384 2 raspberrypi_hwmon,ads7846
cfg80211 614400 1 brcmfmac
snd_bcm2835 24576 2
snd_pcm 102400 1 snd_bcm2835
rfkill 28672 6 bluetooth,cfg80211
snd_timer 32768 1 snd_pcm
bcm2835_codec 36864 0
bcm2835_v4l2 45056 0
snd 73728 7 snd_timer,snd_bcm2835,snd_pcm
v4l2_mem2mem 24576 1 bcm2835_codec
bcm2835_mmal_vchiq 32768 2 bcm2835_codec,bcm2835_v4l2
videobuf2_dma_contig 20480 1 bcm2835_codec
v4l2_common 16384 1 bcm2835_v4l2
videobuf2_vmalloc 16384 1 bcm2835_v4l2
videobuf2_memops 16384 2 videobuf2_dma_contig,videobuf2_vmalloc
videobuf2_v4l2 24576 3 bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem
videobuf2_common 45056 4 bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
i2c_bcm2835 16384 0
videodev 200704 6 bcm2835_codec,v4l2_common,videobuf2_common,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
media 36864 2 videodev,v4l2_mem2mem
vc_sm_cma 36864 1 bcm2835_mmal_vchiq
spi_bcm2835 20480 0
uio_pdrv_genirq 16384 0
uio 20480 1 uio_pdrv_genirq
fixed 16384 0
i2c_dev 20480 0
ip_tables 24576 0
x_tables 32768 1 ip_tables
ipv6 450560 24

```

SudhamaniB, 2019-09-11:
Hi,
Can you provide below info
How you will be sending vlan packets to ethernet port of RPI?
what extra hardware setup needed to test this issue?

Regards,
Sudhamani

gopstein, 2019-09-12:
If you have a switch that supports VLANs, you can configure some VLAN on it (e.g. VLAN 10).  Then configure one port as "Native" on that VLAN and connect that to the rest of your network.  Then you can configure another port as a "Trunk" that contains that VLAN and connect the Pi to that port.  With that configuration, packets coming from your network through the switch and out to the RPi will get tagged with that VLAN number.  
Alternatively, you could probably also use a crossover cable between the Pi and PC/Mac/Linux box that has a VLAN configured on its network port.

SudhamaniB, 2019-09-17:
Hi, 

yes, I am able to reproduce the issue with the above serup, are you trying to resolve the issue?

Regards, 
Sudhamani

gopstein, 2019-09-19:
I'm not trying to resolve it.  I don't know anything about writing or fixing linux drivers.

SudhamaniB, 2019-09-28:
Hi,

Can you tell me what is the vlan ID of the tagged frame sent through trunk port?
How many Vlan ID's the setup is configured with?
Rpi side interface vlan ID ?
RPi is end device or switch?

Regards,
Sudhamani

gopstein, 2019-09-28:
It doesn't matter which VLAN id you use, you can use configure any VLAN ID you want on the switch.
You don't configure the VLAN id on the Pi. Tcpdump should see the VLAN-tagged packets even if you don't configure the Pi's interface with the VLAN id.

The Pi is just a normally configured Pi running Raspbian.  If you take a regular Pi, using a monitor and keyboard (don't try to use SSH), and connect it to a trunk port on a switch, 'tcpdump -e vlan' should display packets that have a VLAN tag.  The Pi will not be able to talk to the network since the IP stack will ignore the frames with VLAN tags, but tcpdump will show them.

If you have a Pi with the SMSC9512/9514 ethernet adapter, you should see the VLAN tagged packets using 'tcpdump -e vlan'

6by9, 2019-09-28:
LAN7800 is using hardware VLAN filtering (and tag stripping), therefore, unless tcpdump has automatically disabled that, you won't receive data tagged for any VLAN that isn't registered on that Pi.
I suspect ```ethtool -k``` is going to return ```rx-vlan-offload: on``` even when you have tcpdump running.

It should be possible to disable that offloading via ```ethtool -K <feature name> off```. Determining the feature name isn't as obvious as using the string printed from the status. I think you want ```ethtool -K rx-vlan-filter off```, but can't verify that easily at the moment. Please try it and check the status then reported from ```ethtool -k```.

SudhamaniB, 2019-09-29:
Hi,

I could able to capture the vlan packets by turning off the  rx-vlan-filter using ethtool.
I have captured output for each command i followed to get the vlan packets captured.
Here is the log from RPI terminal
Hope this resolves the issue.
If this is fine, better to close the issue, so that other people will not try working on the same
/*-------------------------LOG-----------------------------------*/
/*----------------driver being used in RPI 3 B+ --------------------*/
pi@raspberrypi:~$ ethtool -i eth0
driver: lan78xx   ------------------>lan78xx driver
version: 1.0.6
firmware-version: 
expansion-rom-version: 
bus-info: usb-3f980000.usb-1.1.1
supports-statistics: yes
supports-test: no
supports-eeprom-access: yes
supports-register-dump: no
supports-priv-flags: no
pi@raspberrypi:~$ 
/*---------------default ethtool -k eth0 output---------------------*/
pi@raspberrypi:~$ ethtool -k eth0
Features for eth0:
Cannot get device udp-fragmentation-offload settings: Operation not supported
rx-checksumming: on
tx-checksumming: on
        tx-checksum-ipv4: off [fixed]
        tx-checksum-ip-generic: on
        tx-checksum-ipv6: off [fixed]
        tx-checksum-fcoe-crc: off [fixed]
        tx-checksum-sctp: off [fixed]
scatter-gather: on
        tx-scatter-gather: on
        tx-scatter-gather-fraglist: off [fixed]
tcp-segmentation-offload: off
        tx-tcp-segmentation: off [fixed]
        tx-tcp-ecn-segmentation: off [fixed]
        tx-tcp-mangleid-segmentation: off [fixed]
        tx-tcp6-segmentation: off [fixed]
udp-fragmentation-offload: off
generic-segmentation-offload: on
generic-receive-offload: on
large-receive-offload: off [fixed]
rx-vlan-offload: on
tx-vlan-offload: off [fixed]
ntuple-filters: off [fixed]
receive-hashing: off [fixed]
highdma: off [fixed]
rx-vlan-filter: on              ------------------------------->ON
vlan-challenged: off [fixed]
tx-lockless: off [fixed]
netns-local: off [fixed]
tx-gso-robust: off [fixed]
tx-fcoe-segmentation: off [fixed]
tx-gre-segmentation: off [fixed]
tx-gre-csum-segmentation: off [fixed]
tx-ipxip4-segmentation: off [fixed]
tx-ipxip6-segmentation: off [fixed]
tx-udp_tnl-segmentation: off [fixed]
tx-udp_tnl-csum-segmentation: off [fixed]
tx-gso-partial: off [fixed]
tx-sctp-segmentation: off [fixed]
tx-esp-segmentation: off [fixed]
fcoe-mtu: off [fixed]
tx-nocache-copy: off
loopback: off [fixed]
rx-fcs: off [fixed]
rx-all: off [fixed]
tx-vlan-stag-hw-insert: off [fixed]
rx-vlan-stag-hw-parse: off [fixed]
rx-vlan-stag-filter: off [fixed]
l2-fwd-offload: off [fixed]
hw-tc-offload: off [fixed]
esp-hw-offload: off [fixed]
esp-tx-csum-hw-offload: off [fixed]
rx-udp_tunnel-port-offload: off [fixed]
/*--------------changing rx-vlan-filter to off---------------------*/
pi@raspberrypi:~$ sudo ethtool -K eth0 rx-vlan-filter off
Cannot get device udp-fragmentation-offload settings: Operation not supported
Cannot get device udp-fragmentation-offload settings: Operation not supported
/*--------------ethtoll output after changing rx-vlan-filter to off---------------------*/
pi@raspberrypi:~$ ethtool -k eth0
Features for eth0:
Cannot get device udp-fragmentation-offload settings: Operation not supported
rx-checksumming: on
tx-checksumming: on
        tx-checksum-ipv4: off [fixed]
        tx-checksum-ip-generic: on
        tx-checksum-ipv6: off [fixed]
        tx-checksum-fcoe-crc: off [fixed]
        tx-checksum-sctp: off [fixed]
scatter-gather: on
        tx-scatter-gather: on
        tx-scatter-gather-fraglist: off [fixed]
tcp-segmentation-offload: off
        tx-tcp-segmentation: off [fixed]
        tx-tcp-ecn-segmentation: off [fixed]
        tx-tcp-mangleid-segmentation: off [fixed]
        tx-tcp6-segmentation: off [fixed]
udp-fragmentation-offload: off
generic-segmentation-offload: on
generic-receive-offload: on
large-receive-offload: off [fixed]
rx-vlan-offload: on
tx-vlan-offload: off [fixed]
ntuple-filters: off [fixed]
receive-hashing: off [fixed]
highdma: off [fixed]
rx-vlan-filter: off              -------------------->OFF
vlan-challenged: off [fixed]
tx-lockless: off [fixed]
netns-local: off [fixed]
tx-gso-robust: off [fixed]
tx-fcoe-segmentation: off [fixed]
tx-gre-segmentation: off [fixed]
tx-gre-csum-segmentation: off [fixed]
tx-ipxip4-segmentation: off [fixed]
tx-ipxip6-segmentation: off [fixed]

/*---------------tcpdump output at rpi side---------------------*/
pi@raspberrypi:~$ sudo tcpdump -i eth0 -e vlan 
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode                                
listening on eth0, link-type EN10MB (Ethernet), capture size 262144 bytes
09:34:21.687595 20:1a:06:8a:c1:87 (oui Unknown) > Broadcast, ethertype 802.1Q (0x8100), length 60: vlan 5, p 0, ethertype ARP, Request who-has 192.168.1.200 tell 192.168.1.197, length 42
09:34:22.686309 20:1a:06:8a:c1:87 (oui Unknown) > Broadcast, ethertype 802.1Q (0x8100), length 60: vlan 5, p 0, ethertype ARP, Request who-has 192.168.1.200 tell 192.168.1.197, length 42
09:34:23.686287 20:1a:06:8a:c1:87 (oui Unknown) > Broadcast, ethertype 802.1Q (0x8100), length 60: vlan 5, p 0, ethertype ARP, Request who-has 192.168.1.200 tell 192.168.1.197, length 42

Regards,
Sudhamani.

SudhamaniB, 2019-09-29:
Hi,

I could able to capture the vlan packets by turning off the rx-vlan-filter using ethtool.
I have captured output for each command i followed to get the vlan packets captured.
Here is the log from RPI terminal
Hope this resolves the issue.
If this is fine, better to close the issue, so that other people will not try working on the same
/-------------------------LOG-----------------------------------/
/----------------driver being used in RPI 3 B+ --------------------/
pi@raspberrypi:$ ethtool -i eth0
driver: lan78xx ------------------>lan78xx driver
version: 1.0.6
firmware-version:
expansion-rom-version:
bus-info: usb-3f980000.usb-1.1.1
supports-statistics: yes
supports-test: no
supports-eeprom-access: yes
supports-register-dump: no
supports-priv-flags: no
pi@raspberrypi:$
/---------------default ethtool -k eth0 output---------------------/
pi@raspberrypi:$ ethtool -k eth0
Features for eth0:
Cannot get device udp-fragmentation-offload settings: Operation not supported
rx-checksumming: on
tx-checksumming: on
tx-checksum-ipv4: off [fixed]
tx-checksum-ip-generic: on
tx-checksum-ipv6: off [fixed]
tx-checksum-fcoe-crc: off [fixed]
tx-checksum-sctp: off [fixed]
scatter-gather: on
tx-scatter-gather: on
tx-scatter-gather-fraglist: off [fixed]
tcp-segmentation-offload: off
tx-tcp-segmentation: off [fixed]
tx-tcp-ecn-segmentation: off [fixed]
tx-tcp-mangleid-segmentation: off [fixed]
tx-tcp6-segmentation: off [fixed]
udp-fragmentation-offload: off
generic-segmentation-offload: on
generic-receive-offload: on
large-receive-offload: off [fixed]
rx-vlan-offload: on
tx-vlan-offload: off [fixed]
ntuple-filters: off [fixed]
receive-hashing: off [fixed]
highdma: off [fixed]
rx-vlan-filter: on ------------------------------->ON
vlan-challenged: off [fixed]
tx-lockless: off [fixed]
netns-local: off [fixed]
tx-gso-robust: off [fixed]
tx-fcoe-segmentation: off [fixed]
tx-gre-segmentation: off [fixed]
tx-gre-csum-segmentation: off [fixed]
tx-ipxip4-segmentation: off [fixed]
tx-ipxip6-segmentation: off [fixed]
tx-udp_tnl-segmentation: off [fixed]
tx-udp_tnl-csum-segmentation: off [fixed]
tx-gso-partial: off [fixed]
tx-sctp-segmentation: off [fixed]
tx-esp-segmentation: off [fixed]
fcoe-mtu: off [fixed]
tx-nocache-copy: off
loopback: off [fixed]
rx-fcs: off [fixed]
rx-all: off [fixed]
tx-vlan-stag-hw-insert: off [fixed]
rx-vlan-stag-hw-parse: off [fixed]
rx-vlan-stag-filter: off [fixed]
l2-fwd-offload: off [fixed]
hw-tc-offload: off [fixed]
esp-hw-offload: off [fixed]
esp-tx-csum-hw-offload: off [fixed]
rx-udp_tunnel-port-offload: off [fixed]
/--------------changing rx-vlan-filter to off---------------------/
pi@raspberrypi:$ sudo ethtool -K eth0 rx-vlan-filter off
Cannot get device udp-fragmentation-offload settings: Operation not supported
Cannot get device udp-fragmentation-offload settings: Operation not supported
/--------------ethtoll output after changing rx-vlan-filter to off---------------------/
pi@raspberrypi:~$ ethtool -k eth0
Features for eth0:
Cannot get device udp-fragmentation-offload settings: Operation not supported
rx-checksumming: on
tx-checksumming: on
tx-checksum-ipv4: off [fixed]
tx-checksum-ip-generic: on
tx-checksum-ipv6: off [fixed]
tx-checksum-fcoe-crc: off [fixed]
tx-checksum-sctp: off [fixed]
scatter-gather: on
tx-scatter-gather: on
tx-scatter-gather-fraglist: off [fixed]
tcp-segmentation-offload: off
tx-tcp-segmentation: off [fixed]
tx-tcp-ecn-segmentation: off [fixed]
tx-tcp-mangleid-segmentation: off [fixed]
tx-tcp6-segmentation: off [fixed]
udp-fragmentation-offload: off
generic-segmentation-offload: on
generic-receive-offload: on
large-receive-offload: off [fixed]
rx-vlan-offload: on
tx-vlan-offload: off [fixed]
ntuple-filters: off [fixed]
receive-hashing: off [fixed]
highdma: off [fixed]
rx-vlan-filter: off -------------------->OFF
vlan-challenged: off [fixed]
tx-lockless: off [fixed]
netns-local: off [fixed]
tx-gso-robust: off [fixed]
tx-fcoe-segmentation: off [fixed]
tx-gre-segmentation: off [fixed]
tx-gre-csum-segmentation: off [fixed]
tx-ipxip4-segmentation: off [fixed]
tx-ipxip6-segmentation: off [fixed]

/---------------tcpdump output at rpi side---------------------/
pi@raspberrypi:~$ sudo tcpdump -i eth0 -e vlan
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on eth0, link-type EN10MB (Ethernet), capture size 262144 bytes
09:34:21.687595 20:1a:06:8a:c1:87 (oui Unknown) > Broadcast, ethertype 802.1Q (0x8100), length 60: vlan 5, p 0, ethertype ARP, Request who-has 192.168.1.200 tell 192.168.1.197, length 42
09:34:22.686309 20:1a:06:8a:c1:87 (oui Unknown) > Broadcast, ethertype 802.1Q (0x8100), length 60: vlan 5, p 0, ethertype ARP, Request who-has 192.168.1.200 tell 192.168.1.197, length 42
09:34:23.686287 20:1a:06:8a:c1:87 (oui Unknown) > Broadcast, ethertype 802.1Q (0x8100), length 60: vlan 5, p 0, ethertype ARP, Request who-has 192.168.1.200 tell 192.168.1.197, length 42

Regards,
Sudhamani.

6by9, 2019-09-29:
Great.
I'd got the right feature name, but the wrong property in the report (I'm guessing vlan-rx-offload is tag stripping, not filtering). I'm glad it was that way around as I hadn't noticed rx-vlan-filter.

gopstein, 2019-09-29:
Thanks.  disabling rx-vlan-filter fixed the issue.  I'm closing the bug report now.

closed 2019-09-29 by gopstein