raspberrypi/linux #3156: SC16IS752 device tree overlay, bug?

Dloranger, 2019-08-15:
I am also documenting my ongoing debugging effort on the forums in hopes of getting some insights from the larger RPi audience, but so far the thread has been radio silence.  This particular bit looks like a bug hence the report for further investigation by more knowledgeable device tree users.

https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=248736

**Describe the bug**
In the device tree overlay for this device (I2C version only has been checked), There are 2 fields for the clocking frequency "clock-frequency=...", however it appears that only 1 gets updated as a result of the override.  I haven't fully confirmed this is truly a bug, but I can't get the device to on my crystal frequency of 12.288MHz hence the investigation into the possible causes.

**To reproduce**
Instantiate the device in the device tree using the clock-frequency override, in my example
dtoverlay=sc16is752-i2c,int_pin=503,xtal=12288000 (note the gpio 503 is an MCP23017 expander also on the i2c bus).  I have not fully confirmed the gpio503 is correct as I see absolutely no activity on the interrupt signal from the chip so I haven't inspected the gpio in software to confirm the mapping has succeeded as predicted.

Inspect the device tree using the command `dtc -I fs /proc/device-tree` and locate the SC16IS752  device, which in my case yields the following

```
sc16is752@4d {
                                compatible = "nxp,sc16is752";
                                clocks = <0x84>;
                                gpio-controller;
                                status = "okay";
                                i2c-max-frequency = <0x61a80>;
                                interrupt-parent = <0x10>;
                                interrupts = <0x1f1 0x2>;
                                phandle = <0x85>;
                                reg = <0x4d>;
                                #gpio-cells = <0x0>;
                                clock-frequency = <0xbb8000>;

                                sc16is752_clk {
                                        compatible = "fixed-clock";
                                        #clock-cells = <0x0>;
                                        phandle = <0x84>;
                                        clock-frequency = <0xe10000>;
                                };
                        };
```

Here is the full dt if needed

```
dtc -I fs /proc/device-tree
<stdout>: Warning (unit_address_vs_reg): Node /soc has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /soc/vchiq has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /soc/i2c@7e804000/ads1015 has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /soc/i2c@7e804000/ads1015/channel_a has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /soc/gpiomem has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /soc/axiperf has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /soc/reg_1v8@0 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /memory has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /axi/vc_mem has a reg or ranges property, but no unit name
<stdout>: Warning (avoid_default_addr_size): Relying on default #address-cells value for /axi/vc_mem
<stdout>: Warning (avoid_default_addr_size): Relying on default #size-cells value for /axi/vc_mem
/dts-v1/;

/ {
        compatible = "raspberrypi,3-model-b", "brcm,bcm2837";
        serial-number = "000000006086d354";
        model = "Raspberry Pi 3 Model B Rev 1.2";
        memreserve = <0x3b400000 0x4c00000>;
        interrupt-parent = <0x1>;
        #address-cells = <0x1>;
        #size-cells = <0x1>;

        clocks {
                compatible = "simple-bus";
                #address-cells = <0x1>;
                #size-cells = <0x0>;

                clock@3 {
                        compatible = "fixed-clock";
                        #clock-cells = <0x0>;
[... 1377 lines cut ...]
* Which OS and version (`cat /etc/rpi-issue`)? 
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, e910bf55ac92e27c6e77043e8995bcb5e0a2d742, stage2
* Which firmware version (`vcgencmd version`)?
Mar 27 2019 15:45:53
Copyright (c) 2012 Broadcom
version 2e98b31d18547962e564bdf88e57b3df7085c29b (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux openrepeater 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

**Logs**
If applicable, add the relevant output from `dmesg` or similar.
dmesg
note - I can't find where this error code is described so unable to debug it further
[    5.466004] sc16is7xx: probe of 1-004d failed with error -22

vcdbg log msg
002345.680: brfs: File read: /mfs/sd/overlays/sc16is752-i2c.dtbo
002359.741: Loaded overlay 'sc16is752-i2c'
002359.796: dtparam: int_pin=497
002360.238: dtparam: addr=0x4D
002360.746: dtparam: xtal=12288000

**Additional context**
Add any other relevant context for the problem.

Dloranger, 2019-08-17:
Okay I am fairly certain there is 2 bugs in this overlay.  

#1) Prior to making a hard coded change to the clock_frequency value in the DTS I was only getting gibberish, but on the expected cadence I am expecting.  I have now hardcoded the actual frequency of my crystal, recompiled and rebooted.  Now when I monitor the input, the signal is coming through perfectly as desired.

#2) When I use a native GPIO (24) everything works as expected with the change to bug 1, but leaving the fix in place and reverting back to my desired gpio on the expander, the driver fails to load the chipset and the ttySC0/1 are not created.

pelwell, 2019-08-17:
I can see one bug: the `xtal` parameter is pointing at the wrong label - it should say:
```
               xtal = <&sc16is752_clk>,"clock-frequency:0";
```
However, as far as the DT goes, the `int_pin` parameter seems to do the right thing, so I'm puzzled as to why it isn't working.

6by9, 2019-08-17:
@pelwell correct me if I'm wrong, but the ```int_pin``` parameter is changing the GPIO line within the GPIO controller specified by ```interrupt-parent```. That's set as ```&gpio``` and you're not changing it, therefore it is a native GPIO.
https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/overlays/sc16is752-i2c-overlay.dts#L19

503 is the old sysfs style GPIO numbering for the line, whereas the kernel internally uses controller, and index on that controller.

If you want to change to using a GPIO expander on the I2C bus then
(a) that needs to be wired up for interrupts
(b) you need to change ```interrupt-parent``` to point at that GPIO exapnder, and
(c) ```int_pin``` needs to match the line on that controller that you wish to use.

Dloranger, 2019-08-17:
I can switch to a native gpio on my hat board, I was just trying to leave
as many native gpio open as possible for flexibility in adding other hats.

My expectation was that since the expander is being loaded up in the device
tree it would be usable just like the native gpio.  I can understand the
distinction between a native pin and something that needs to be software
driver based though.

I appreciate the confirmation on the clock bug, I will work around until a
fix hits mainline updates.  I am on Stretch for the moment, but will move
to buster before long.

pelwell, 2019-08-17:
It would have helped if I'd read all the words in your post, and not conflated it with a similar sounding interrupt issue on the Forums that is all native GPIO.

As @6by9 said, it is possible to use non-native GPIOs from an expander, but you would need a custom overlay to do so for a number of reasons:

1. Unless you take some special measures, labels in an applied overlay aren't made available to other overlays - they just disappear, leaving the phandles. This is good because it prevents overlays from unintentionally clashing, but it makes it harder to add shared resource via an overlay.

2. It isn't currently possible to pass a label name as a parameter to an overlay, so the labels for any GPIO controllers you might want to use would have to appear in the pre-built overlay.

The combination of those factors means that a custom overlay is probably the best way to go. Note that there is a utility, [ovmerge](https://github.com/raspberrypi/utils/blob/master/ovmerge/ovmerge), that can combine overlays to make a new one, renumbering fragments and allowing you to refer to labels between the constituent overlays. Because ovmerge manipulates the source code rather than compiling and decompiling, the end result is quite readable. It is currently used to maintain the [upstream overlay](https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/overlays/upstream-overlay.dts).

Dloranger, 2019-08-17:
I will have a look at the ovmerge tool to streamline other overlays I am working on. I frequently have to merge overlays to get multiple devices of the same type working together, example mcp23017.

I am okay with using the native gpio pin for now, I only have a few native gpio consumed and I expose several more through the expanders.

I was not 100% clear on which line to make the applicable change to the xtal = .... above.  Will you be updating the source with the correction? If so I will issue a "sed" command to reflect the code change in my scripts to fix the code until it hits mainline.  Right now I have it decompiling, fixing the hardcode and recompiling the dtbo which works, but obviously is not the best way to do it.

pelwell, 2019-08-17:
The patch is in the rpi-4.19.y now.

Dloranger, 2019-08-31:
I think something may still be broken.  I copied the entire file from the updated dts and recompiled it using dtc

```
pi-star@pi-star(rw):~$ sudo nano /boot/overlays/sc16is752-i2c.dts
pi-star@pi-star(rw):~$ sudo dtc -I dts -O dtb -o /boot/overlays/sc16is752-i2c.dtbo /boot/overlays/sc16is752-i2c.dts
Warning (unit_address_vs_reg): Node /fragment@0 has a unit name, but no reg property
```
and use the following to instantiate the device (note I updated the clock frequency from before to one more suitable to UART @ 115200)

`dtoverlay=sc16is752-i2c,int_pin=24,addr=0x4D,xtal=3686400`

when rebooting I get the following (which appears to be ignoring the address update since it says its looking at 0x48

```
[    4.261837] i2c i2c-1: Failed to register i2c client sc16is752 at 0x48 (-16)
[    4.261854] i2c i2c-1: of_i2c: Failure registering /soc/i2c@7e804000/sc16is752@48
[    4.261866] i2c i2c-1: Failed to create I2C device for /soc/i2c@7e804000/sc16is752@48
```

any ideas?

Dloranger, 2019-08-31:
just to be sure, I also moved my crystal input pin back to the 12.288MHz and it still responds the same way. its possible its a soldering issue as I had to lift the pin, but I see the crystal making it into the pin at least and no obvious soldering defects when looking under magnification.

I will try a second board when I get a chance just to confirm I haven't damaged the chip somehow.

6by9, 2019-08-31:
Your kernel log says
```
[    4.261837] i2c i2c-1: Failed to register i2c client sc16is752 at 0x48 (-16)
```
0x48 there, but you're dtoverlay line supposedly changed the address to 0x4D. I think there's something still wrong in your overlay.
The change is merged into the kernel image, therefore please use rpi-update to update a non-critical Pi to use this updated image, and report back. Sorry, but it's nearly impossible to predict what has gone wrong with your rebuilt dt files.

Dloranger, 2019-08-31:
Sure thing, I will start fresh and see how it goes.

Dloranger, 2019-09-05:
Everything is working when I got an image up to buster and used the precompiled DTBO files

closed 2019-09-05 by Dloranger