raspberrypi/linux #3157: More DMA issues on RPI4b in 64-bit mode with >1G RAM

yaroslavros, 2019-08-17:
It seems that there are still problems with DMA operation on RPI4 in 64-bit mode with > 1G RAM. This time on bcm2835-mmc.

When I try downloading large file via WiFi using scp, I run into reproducible swiotlb buffer issue at any points between 700Mb to 2Gb of downloaded data. Kernel logs are:

`bcm2835-dma fe00700.dma: swiotlb buffer is full (sz: 1536 bytes)`
`bcm2835-dma fe00700.dma: DMA: Out of SW-IOMMU space for 1536 bytes`
`mmc-bcm2835 fe300000.mmcnr: dma_map_sg returned zero length`
`mmc1: Timeout waiting for hardware interrupt.`

I have confirmed this at 2, 3 and 4Gb RAM. It does not occur on 1Gb RAM - I have downloaded 6Gb file via Wifi without any trouble.

This happens on the latest 4.19.y kernel

I'll dig more into this in upcoming days. I suspect there is a leak somewhere in the relevant driver stack that prevents swiotlb bounce buffers from being freed. This would explain normal behaviour on 1G RAM where bounce buffers are not used.

lategoodbye, 2019-08-17:
Please provide the exact kernel version

yaroslavros, 2019-08-17:
I see it on the latest 4.19.y 
`# uname -a`
`Linux rpi4a 4.19.64-v8 #34 SMP Sat Aug 17 13:51:51 BST 2019 aarch64 GNU/Linux`

I strongly suspect that this issue is related to the way bcm2835-mmc works with swiotlb and affects all kernel versions.

I am pretty sure root cause is the same as in #3114

yaroslavros, 2019-09-09:
Closing as the fix is committed

closed 2019-09-09 by yaroslavros