raspberrypi/linux #3161: Pi3B+ wrong revision

Rydekull, 2019-08-18:
**Describe the bug**
Installing Fedora 30 and trying to use the python pinout libraries I discovered that it failed because /proc/cpuinfo wasn't correctly reporting the revision of the board. Instead it is reporting:
https://paste.fedoraproject.org/paste/rC-ToLoHOPU2fo-dJRD6vw

Digging in to this a bit further, it seems like this is related to Device Tree not identifying the board correctly, possibly?
Then again, looking into device-tree it seems like its okay there, its just /proc/cpuinfo giving out wrong information.

It is the RPI 3B+ v1.3 from 2017, the board should be using a Cortex-A53 CPU on the BCM2837 board 

```
[root@localhost device-tree]# cat /proc/cpuinfo | tail -n 12
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 51.20
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: 0000
Serial		: 000000007cb25365
[root@localhost device-tree]# cat /proc/device-tree/model /proc/device-tree/compatible 
Raspberry Pi 3 Model B+
raspberrypi,3-model-b-plusbrcm,bcm2837
[root@localhost device-tree]#
```

We can tell that the board is wrong, and the revision is 0000. While it should be a020d3 according to:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md

**To reproduce**
Install https://download.fedoraproject.org/pub/fedora/linux/releases/30/Spins/armhfp/images/Fedora-Minimal-armhfp-30-1.2-sda.raw.xz

Boot it up, check:
cat /proc/cpuinfo
cat /proc/device-tree/model 
cat /proc/device-tree/compatible

Rydekull, 2019-08-18:
oh, and, yes, i've seen the note on the revision-code page, that it seems known that they are reporting a different board and revision, but it seems like a very breaking change.

pelwell, 2019-08-18:
What output do you get from the following command?:
```
$ od -tx1 /proc/device-tree/system/linux,revision
```
It should return:
```
0000000 00 a0 20 d3
0000004
```

Rydekull, 2019-08-19:
No such file exists, /proc/device-tree/system

pelwell, 2019-08-19:
Is Fedora using U-boot? That is not the Device Tree provided by the firmware, so this is not our bug/limitation.

JamesH65, 2019-09-17:
It appears this issue is not related to our kernel and should be reported to Fedora.

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

closed 2019-12-20 by JamesH65