raspberrypi/linux #3162: [PI 4B] [RPI-5.3Y] X server does not work with the VC4 FKMS MESA V3D

n00b7, 2019-08-18:
on version rpi-4.19y everything is fine but on  version 5.3 the X-server wont start, glitches out, and causes the entire Pi to freeze.

pelwell, 2019-08-18:
There are some major changes in 5.3 and the rpi-5.3.y branch is a work-in-progress.

nkichukov, 2019-09-25:
This affects me too, I wish I found this bug report earlier.
GNU/Gentoo Linux here, arm64 kernel and raspberrypi 4B with 4GB memory.

Xorg server starts up, lightdm runs fine for very long while, however once a DE session is started (xfce4-session) it takes a few seconds to operate and the whole system crashes, not only X, but the whole rpi4 The only option is to power-cycle it.

rpi-5.3.y -> 5.3.0 with the latest commit to date with bcm2711_defconfig kernel configuration. Mesa 19.2.0_rc4 with v3d support in.

Once I compiled kernel rpi-4.19.y, with bcm2711_defconfig on the same system, I got:
'vlank_mode=0 glxgears' to output at ~1200fps.

Let me know if I can help with something else.
Thanks,
-Nikolay

pelwell, 2019-09-25:
Although progress has been made, we are aware of (and can easily reproduce) some remaining issue(s) in the v3d driver on 5.3.

nkichukov, 2019-09-26:
Sounds good. Thank you.
Happy to test the 5.3.y series again, once ready, so keep us posted.
Cheers,
-N

JamesH65, 2019-12-20:
@6by9 @pelwell Any progress on this?

pelwell, 2019-12-20:
I found and fixed an issue with MMU exception handling, and as far as I know it's all good now. You should also grab the latest Mesa library package - `sudo apt update; sudo apt install libgl1-mesa-dri`, I think.

JamesH65, 2019-12-20:
@nkichukov If you could test this and close if fixed, that would be splendid!

nkichukov, 2019-12-20:
I just switched to 5.4.3 kernel recently and the system was stable. However I have not verified the performance numbers as reported by glxgears. I will do that and report.

nkichukov, 2019-12-20:
Confirmed, no more crashes with 5.4.3 kernel.
compiz enabled DE produces around 250 FPS, where same DE without compositor is back to the 1200 FPS vicinities of glxgears.

Any chance we can get better FPS rates with compositors enabled? XFCE4 built-in compositor is also suffering poor performance on 4.19.83 and 5.4.3 kernels... 
Thanks,
-N