raspberrypi/linux #3163: brcm,bcm7211-pcie in ./arch/arm/boot/dts/bcm2838.dtsi ?

satmandu, 2019-08-18:
I noticed this bit of [code](https://github.com/raspberrypi/linux/blob/73a743bcf8618982afab993a25cc36df0d9d9ae4/arch/arm/boot/dts/bcm2838.dtsi#L330) in in arch/arm/boot/dts/bcm2838.dtsi
```
pcie_0: pcie@7d500000 {
			reg = <0x0 0x7d500000 0x9310>,
			      <0x0 0x7e00f300 0x20>;
			msi-controller;
			msi-parent = <&pcie_0>;
			#address-cells = <3>;
			#interrupt-cells = <1>;
			#size-cells = <2>;
			bus-range = <0x0 0x01>;
			compatible = "brcm,bcm7211-pcie", "brcm,bcm7445-pcie",
				     "brcm,pci-plat-dev";
```
There's no bcm7211, right? That should be bcm2711?

pelwell, 2019-08-18:
Nope. BCM7211 is a sister-chip to the BCM2711 aimed at set-top boxes, leading to commits like this: https://www.mail-archive.com/linux-kernel@vger.kernel.org/msg1995440.html

satmandu, 2019-08-18:
Ah my mistake.

closed 2019-08-18 by satmandu