raspberrypi/linux #3168: Pi4 i2c pullup resistors interfere with boot process, 3.3V regulator shutdown during reboot.

noafterglow, 2019-08-20:
**Is this the right place for my bug report?  That's a really great question....
Not sure how this happens, it may be a kernel bug, or it may be a hardware bug, or it may be a firmware bug.

**Describe the bug**
If the i2c resistors on the peripheral i2c port are connected to an alternate source of 3.3V, (like from a regulator driven from the 5V pin on a hat), the system will not boot.

Additionally, the 3.3V regulator on the pi is turned off during reboot.  This may be REALLY annoying if a hat uses it for something which takes a long time to stabilize, (like GPS, Gyro, etc.)  This could be really annoying for robotics or any controls application that's not aware of it. 

The reboot process appears to properly finish, then the 3.3V regulator is shut off???, and the system will not actually reboot.  It appears to wait forever for something with no hdmi output.  The 3.3V regulator seems to have maybe 0.5V on that pin, presumably the leakage from the i2c pullup resistors.

Note that this does NOT happen when PINN "reboots" to start Buster, only when Buster tries to sudo reboot.

lategoodbye, 2019-08-29:
Could you please name the pins on the GPIO header which are affected?

6by9, 2019-08-29:
@jimbojr Do I recall that the PMIC won't power up if any of the rails are held above a certain threshold? Backpowered USB data rails rings a bell from alpha testing.

The Pi 3V3 regulator being disabled during reboot has been fixed with a bootloader change. See https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=246027 and #3065.

jimbojr, 2019-09-23:
Yes if you leak current back into the unpowered 3V3 of the Pi4 it will likely fail to start (as the PMIC will only start if all rails are ~< 250mV so it can be sure of a 'controlled' start).

JamesH65, 2019-12-20:
Closing this issue as questions answered/issue resolved.

closed 2019-12-20 by JamesH65