raspberrypi/linux #3169: External sdcard stops working after 5-6 days

dev-0x7C6, 2019-08-20:
**Describe the bug**
After 5-6 days of writing random files (at low bandwidth), external sdcard stop working. Kernel report this:

> (mmcblk1: error -110 sending stop command, original cmd response 0x900, card status 0x400e00)

After that the only option to get external sdcard working again is to restart system.

**System**
* Raspberry Pi CM3
* Used sdcard: XMORE 3.71 GiB (industrial) FAT32
* Controller: 3f300000.mmc
* Driver: sdhci-iproc
* Yocto (sumo) custom distro 
* Firmware: version 31e0613622dc2f2463bf3dd74e6c897d91201a4d (clean) (release)
* Linux mpi540-cm3 4.14.114 #1 SMP Thu Jul 25 12:03:40 UTC 2019 armv7l GNU/Linux

**Logs**
Jun 15 00:12:40 mpi540-cm3 kernel: mmcblk1: error -110 sending stop command, original cmd response 0x900, card status 0x400e00
Jun 15 00:12:40 mpi540-cm3 kernel: ttyAMA ttyAMA0: 1 input overrun(s)
Jun 15 00:14:48 mpi540-cm3 kernel: INFO: task kworker/1:3:30433 blocked for more than 120 seconds.
Jun 15 00:14:48 mpi540-cm3 kernel:       Tainted: G           O    4.14.114 #1
Jun 15 00:14:48 mpi540-cm3 kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Jun 15 00:14:48 mpi540-cm3 kernel: kworker/1:3     D    0 30433      2 0x00000000
Jun 15 00:14:48 mpi540-cm3 kernel: Workqueue: events_freezable mmc_rescan
Jun 15 00:14:48 mpi540-cm3 kernel: [<806d2c30>] (__schedule) from [<806d3374>] (schedule+0x4c/0xac)
Jun 15 00:14:48 mpi540-cm3 kernel: [<806d3374>] (schedule) from [<80576c28>] (__mmc_claim_host+0xa0/0x1d4)
Jun 15 00:14:48 mpi540-cm3 kernel: [<80576c28>] (__mmc_claim_host) from [<80576d8c>] (mmc_get_card+0x30/0x34)
Jun 15 00:14:48 mpi540-cm3 kernel: [<80576d8c>] (mmc_get_card) from [<8057fa28>] (mmc_sd_detect+0x20/0x74)
Jun 15 00:14:48 mpi540-cm3 kernel: [<8057fa28>] (mmc_sd_detect) from [<805796e8>] (mmc_rescan+0xd8/0x3fc)
Jun 15 00:14:48 mpi540-cm3 kernel: [<805796e8>] (mmc_rescan) from [<80137ce0>] (process_one_work+0x224/0x518)
Jun 15 00:14:48 mpi540-cm3 kernel: [<80137ce0>] (process_one_work) from [<80138d9c>] (worker_thread+0x68/0x60c)
Jun 15 00:14:48 mpi540-cm3 kernel: [<80138d9c>] (worker_thread) from [<8013e4c4>] (kthread+0x170/0x178)
Jun 15 00:14:48 mpi540-cm3 kernel: [<8013e4c4>] (kthread) from [<8010826c>] (ret_from_fork+0x14/0x28)
Jun 15 00:16:51 mpi540-cm3 kernel: INFO: task elp-recorder:492 blocked for more than 120 seconds.
Jun 15 00:16:51 mpi540-cm3 kernel:       Tainted: G           O    4.14.114 #1
Jun 15 00:16:51 mpi540-cm3 kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Jun 15 00:16:51 mpi540-cm3 kernel: elp-recorder    D    0   492    483 0x00000000
Jun 15 00:16:51 mpi540-cm3 kernel: [<806d2c30>] (__schedule) from [<806d3374>] (schedule+0x4c/0xac)
Jun 15 00:16:51 mpi540-cm3 kernel: [<806d3374>] (schedule) from [<8014be94>] (io_schedule+0x20/0x40)
Jun 15 00:16:51 mpi540-cm3 kernel: [<8014be94>] (io_schedule) from [<8021fef8>] (__lock_page+0x114/0x13c)
Jun 15 00:16:51 mpi540-cm3 kernel: [<8021fef8>] (__lock_page) from [<8022069c>] (pagecache_get_page+0x21c/0x2e4)
Jun 15 00:16:51 mpi540-cm3 kernel: [<8022069c>] (pagecache_get_page) from [<80221374>] (grab_cache_page_write_begin+0x28/0x3c)
Jun 15 00:16:51 mpi540-cm3 kernel: [<80221374>] (grab_cache_page_write_begin) from [<802ca8a8>] (block_write_begin+0x30/0xd4)
Jun 15 00:16:51 mpi540-cm3 kernel: [<802ca8a8>] (block_write_begin) from [<802cabb0>] (cont_write_begin+0x264/0x3b4)
Jun 15 00:16:51 mpi540-cm3 kernel: [<802cabb0>] (cont_write_begin) from [<80391298>] (fat_write_begin+0x68/0x98)
Jun 15 00:16:51 mpi540-cm3 kernel: [<80391298>] (fat_write_begin) from [<802214f0>] (generic_perform_write+0xb0/0x1ac)
Jun 15 00:16:51 mpi540-cm3 kernel: [<802214f0>] (generic_perform_write) from [<80223d50>] (__generic_file_write_iter+0x11c/0x1d0)
Jun 15 00:16:51 mpi540-cm3 kernel: [<80223d50>] (__generic_file_write_iter) from [<80223f98>] (generic_file_write_iter+0x194/0x274)
Jun 15 00:16:51 mpi540-cm3 kernel: [<80223f98>] (generic_file_write_iter) from [<8028d900>] (__vfs_write+0xe0/0x130)
Jun 15 00:16:51 mpi540-cm3 kernel: [<8028d900>] (__vfs_write) from [<8028db34>] (vfs_write+0xb4/0x1c0)
Jun 15 00:16:51 mpi540-cm3 kernel: [<8028db34>] (vfs_write) from [<8028dd80>] (SyS_write+0x4c/0xa0)
Jun 15 00:16:51 mpi540-cm3 kernel: [<8028dd80>] (SyS_write) from [<801081c0>] (ret_fast_syscall+0x0/0x28)
Jun 15 00:16:51 mpi540-cm3 kernel: INFO: task elp-pqm540ui:511 blocked for more than 120 seconds.
Jun 15 00:16:51 mpi540-cm3 kernel:       Tainted: G           O    4.14.114 #1
Jun 15 00:16:51 mpi540-cm3 kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Jun 15 00:16:51 mpi540-cm3 kernel: elp-pqm540ui    D    0   511    394 0x00000000
Jun 15 00:16:51 mpi540-cm3 kernel: [<806d2c30>] (__schedule) from [<806d3374>] (schedule+0x4c/0xac)
Jun 15 00:16:51 mpi540-cm3 kernel: [<806d3374>] (schedule) from [<8014be94>] (io_schedule+0x20/0x40)
Jun 15 00:16:51 mpi540-cm3 kernel: [<8014be94>] (io_schedule) from [<8021fbe0>] (wait_on_page_bit+0x124/0x144)
Jun 15 00:16:51 mpi540-cm3 kernel: [<8021fbe0>] (wait_on_page_bit) from [<8027f11c>] (migrate_pages+0x6a8/0x860)
Jun 15 00:16:51 mpi540-cm3 kernel: [<8027f11c>] (migrate_pages) from [<8022d8bc>] (alloc_contig_range+0x18c/0x378)
Jun 15 00:16:51 mpi540-cm3 kernel: [<8022d8bc>] (alloc_contig_range) from [<80288b14>] (cma_alloc+0xfc/0x288)
Jun 15 00:16:51 mpi540-cm3 kernel: [<80288b14>] (cma_alloc) from [<804a7270>] (dma_alloc_from_contiguous+0x48/0x4c)
Jun 15 00:16:51 mpi540-cm3 kernel: [<804a7270>] (dma_alloc_from_contiguous) from [<80115d90>] (__alloc_from_contiguous.constprop.5+0x40/0xbc)
Jun 15 00:16:51 mpi540-cm3 kernel: [<80115d90>] (__alloc_from_contiguous.constprop.5) from [<80115e50>] (cma_allocator_alloc+0x44/0x4c)
Jun 15 00:16:51 mpi540-cm3 kernel: [<80115e50>] (cma_allocator_alloc) from [<8011545c>] (__dma_alloc+0x1ec/0x2fc)
Jun 15 00:16:51 mpi540-cm3 kernel: [<8011545c>] (__dma_alloc) from [<80115604>] (arm_dma_alloc+0x50/0x58)
Jun 15 00:16:51 mpi540-cm3 kernel: [<80115604>] (arm_dma_alloc) from [<805c7828>] (vchiq_prepare_bulk_data+0x308/0x6f8)
Jun 15 00:16:51 mpi540-cm3 kernel: [<805c7828>] (vchiq_prepare_bulk_data) from [<805bf7d8>] (vchiq_bulk_transfer+0x2e4/0x564)
Jun 15 00:16:51 mpi540-cm3 kernel: [<805bf7d8>] (vchiq_bulk_transfer) from [<805c4350>] (vchiq_ioctl+0x450/0x1a64)
[... 180 lines cut ...]
Jun 15 00:20:57 mpi540-cm3 kernel: [<802a10a8>] (do_vfs_ioctl) from [<802a17ec>] (SyS_ioctl+0x44/0x68)
Jun 15 00:20:57 mpi540-cm3 kernel: [<802a17ec>] (SyS_ioctl) from [<801081c0>] (ret_fast_syscall+0x0/0x28)
Jun 15 00:20:57 mpi540-cm3 kernel: INFO: task kworker/1:3:30433 blocked for more than 120 seconds.
Jun 15 00:20:57 mpi540-cm3 kernel:       Tainted: G           O    4.14.114 #1
Jun 15 00:20:57 mpi540-cm3 kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Jun 15 00:20:57 mpi540-cm3 kernel: kworker/1:3     D    0 30433      2 0x00000000
Jun 15 00:20:57 mpi540-cm3 kernel: Workqueue: events_freezable mmc_rescan
Jun 15 00:20:57 mpi540-cm3 kernel: [<806d2c30>] (__schedule) from [<806d3374>] (schedule+0x4c/0xac)
Jun 15 00:20:57 mpi540-cm3 kernel: [<806d3374>] (schedule) from [<80576c28>] (__mmc_claim_host+0xa0/0x1d4)
Jun 15 00:20:57 mpi540-cm3 kernel: [<80576c28>] (__mmc_claim_host) from [<80576d8c>] (mmc_get_card+0x30/0x34)
Jun 15 00:20:57 mpi540-cm3 kernel: [<80576d8c>] (mmc_get_card) from [<8057fa28>] (mmc_sd_detect+0x20/0x74)
Jun 15 00:20:57 mpi540-cm3 kernel: [<8057fa28>] (mmc_sd_detect) from [<805796e8>] (mmc_rescan+0xd8/0x3fc)
Jun 15 00:20:57 mpi540-cm3 kernel: [<805796e8>] (mmc_rescan) from [<80137ce0>] (process_one_work+0x224/0x518)
Jun 15 00:20:57 mpi540-cm3 kernel: [<80137ce0>] (process_one_work) from [<80138d9c>] (worker_thread+0x68/0x60c)
Jun 15 00:20:57 mpi540-cm3 kernel: [<80138d9c>] (worker_thread) from [<8013e4c4>] (kthread+0x170/0x178)
Jun 15 00:20:57 mpi540-cm3 kernel: [<8013e4c4>] (kthread) from [<8010826c>] (ret_from_fork+0x14/0x28)
Jun 15 00:22:40 mpi540-cm3 kernel: mmc1: Card stuck in programming state! mmcblk1 card_busy_detect
Jun 15 00:22:40 mpi540-cm3 kernel: ttyAMA ttyAMA0: 1 input overrun(s)
Jun 15 00:22:43 mpi540-cm3 kernel: mmc1: card never left busy state
Jun 15 00:22:43 mpi540-cm3 kernel: mmc1: tried to reset card, got error -110
Jun 15 00:22:43 mpi540-cm3 kernel: ttyAMA ttyAMA0: 1 input overrun(s)
Jun 15 00:22:43 mpi540-cm3 kernel: print_req_error: I/O error, dev mmcblk1, sector 5455870
Jun 15 00:22:43 mpi540-cm3 kernel: mmcblk1: error -110 sending status command, retrying
Jun 15 00:22:43 mpi540-cm3 kernel: mmcblk1: error -110 sending status command, retrying
Jun 15 00:22:43 mpi540-cm3 kernel: mmcblk1: error -110 sending status command, aborting

pelwell, 2019-08-20:
Please try a different SD card.

dev-0x7C6, 2019-08-20:
Tested with:
SanDisk Ultra microSDHC 32GB Android 48 MB/s (Working)
SanDisk Ultra microSDHC 16GB Android 80MB/s UHS-I (Working)
Kingston microSDHC Canvas Select 16GB 80R/10W Class 10 UHS-I (Working)
Kingston microSDHC Canvas Select 32GB 80R/10W Class 10 UHS-I (Working)
Goodram SDU4GDPGRB - 4GB; pSLC (Working)
XMORE 3.71 GiB; pSLC (**Not working**) // working fine with iMX.6 on different devices

We're using XMORE pSLC industrial sdcards for many devices in our company, so we would like to stick with it. Any solution or ideas how to solve a problem would be much appreciated.

lategoodbye, 2019-08-20:
Is your kernel based on this repository?
Did you tried rpi-4.19?
Is there a swap file?

dev-0x7C6, 2019-08-22:
> Is your kernel based on this repository?

Yes (rpi-4.14.y)

> Did you tried rpi-4.19?

Not yet, upgrade of kernel is postponed for now.

> Is there a swap file?

Disabled (no swap partition nor swap service), about 600MiB of ram free (cache)

> Out of curiosity, what is the exact vendor naming of the SD card (should start with SDU...)?

SDU004GXPI8C016Z Xmore - uSD 4GB 

> Are you able to reproduce this issue with the same kernel but a Raspberry Pi board?

I didn't try yet to isolate from our production hardware board/distro. Point taken.

#### Side question
Is there any value in the logs that I attached? or it's to generic to pinpoint exact problem? 

#### Side notes
We're using constantly ttyAMA0 to communicate with external MCU, sometimes we have a random repetition and dmesg from times to times generate this
> ttyAMA ttyAMA0: 1 input overrun(s)

Maybe there is a chance that is correlated somehow? since "input overrun(s)" happens when card is about to broke.

dev-0x7C6, 2019-09-23:
We got official response from XMore:

> thanks for the log file.
> I suspect that you encountered a very specific use case that could cause the card to hang in programming status.
> In this case, three conditions must be met.
> 1) Small data is written from 8 sectors or smaller
> 2) A previously written LBA address is written again.
> 3) The last data packet exceeds the size of a controller internal buffer.

> If this is the case, the board remains in busy status and can only be reset by a power cycle.
> To avoid this behavior, the FW must be modified.
> The new FW is available by the new P/N:
> SDU004GXAIMS001E
> microSD Card, 4GB, pSLC, -40°C to 85°C
> This card will fix our problem.

I think this is enough proof to close this bug. And sdcard firmware is to blame. Thank you for the questions and suggestions.

**Off topic**

As this faulty sdcard was shipped with many our devices, we're working on seamless  workaround to this problem. Any advice will be appreciated. We're using sdcard with single 4GiB fat32 partition w 4k clusters. Currently we're testing with 8k clusters formatted fat32.

JamesH65, 2019-09-23:
@dev-0x7C6 Ouch, so the fix is to buy new cards? I'd be inclined to ask for a refund! Cannot comment on a workaround though, I would have thought that would have to come from XMore as well.

dev-0x7C6, 2019-10-09:
@JamesH65 Yes, pretty much. We're switching to other vendor. Not sure how company will progress with refund.

As for workaround 8K formatted cards working fine (about 2 weeks without hangs). Probably our software update will propose to format those cards with 8K clusters.

closed 2019-10-09 by dev-0x7C6