raspberrypi/linux #3170: raspi-config sets wifi country based on wlan0 only

pateljj04, 2019-08-20:
**Describe the bug**
If using Pi as AP configured on wlan0 and using wlan1 (second wifi dongle) to connect to wifi as a client, it shows as Wifi Country not set. No way from the gui using the Wireless & Network Settings in the menu panel to select a network to join. You can't set the wifi country by going to localization in raspi-config. 

**To reproduce**
Followed - https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md stopped at sharing internet connection. Reboot

**Expected behaviour**
Select wifi network to join on wlan1 using the gui. 

**Actual behaviour**
Wireless & Network settings panel on the gui shows wifi country not set, nothing happens when you try to set the country. 

* Which model of Raspberry Pi? Pi3B+

From digging a little it looks like when you have nohook wpa_supplicant raspi-config takes that as wlan0 doesn't exist, therefore it shows an error when trying to get the country for the second wifi dongle.

pelwell, 2019-08-20:
raspi-config is designed to make it easier for users to perform simple, common tasks. Running an AP and using a dongle as a client it well outside what I would call the set of simple, common tasks.

pateljj04, 2019-08-20:
Got it! Sorry then I will close this.

closed 2019-08-20 by pateljj04